Barracuda09/SATPI #147: Signal streng is sometimes wrong (2 instead of 240)

SaulGoodman1337, 2022-01-23:
I am not sure what you need to understand my problem. But my signal strength is sometimes not read correctly. instead of 240 a 2 is transmitted. then again the 240. this leads in the TVheadend to jumping signal strengths from 1% to 100%.
I have a Gigablue UE4K in conjunction with a Gigablue SCR LNB

Barracuda09, 2022-01-23:
Hello @SaulGoodman1337 

Does it jump more then 1 time (only in the begin) or just all the time? is it regular intervals?

SaulGoodman1337, 2022-01-23:
Hi,
All the time and there is no Intervall. I think its a parsing issue. 
Sometimes the Signal strenght ist 240 then 2, then 4 and then 240.

Barracuda09, 2022-01-23:
Which version of TvH are you using?

SaulGoodman1337, 2022-01-23:
4.3
But I think that is irrelevant. The values already come out wrong from SATPI. 
But I can not exclude whether the values already come wrong from the tuner ?

SaulGoodman1337, 2022-01-23:
[image]
[image]

It seems that 239 is Display correct 😅

Barracuda09, 2022-01-23:
Is the value jumping in the also in the SatPI web interface?

SaulGoodman1337, 2022-01-23:
> Is the value jumping in the also in the SatPI web interface?

Yes, exactly. However, I don't know if this already happens on Enigma level. 
Wait, I just look for the remote control, configure a tuner and go into the satfinder.

Barracuda09, 2022-01-23:
Ok... Strange I have not seen this issue here before.

Barracuda09, 2022-01-23:
But the streams seem OK..

SaulGoodman1337, 2022-01-23:
> But the streams seem OK..
[image]

The stream is perfect... 
I think something strange is happening here when parsing. but I must confess that I don't quite understand your code here. can you possibly tell me at which point you do this exactly?

Barracuda09, 2022-01-23:
I use an android phone, it has SAT>IP Alignment app. if you use an android phone you could try that?

Barracuda09, 2022-01-23:
https://github.com/Barracuda09/SATPI/blob/6b90b138ec2903254be982793a485dd43d7aa05e/src/input/dvb/Frontend.cpp#L292-L300

https://github.com/Barracuda09/SATPI/blob/6b90b138ec2903254be982793a485dd43d7aa05e/src/input/dvb/FrontendData.cpp#L334-L350

https://github.com/Barracuda09/SATPI/blob/6b90b138ec2903254be982793a485dd43d7aa05e/src/output/StreamThreadRtcpBase.cpp#L105-L130

Barracuda09, 2022-01-23:
Do you have some TCP date where it shows there is an parsing error?

SaulGoodman1337, 2022-01-23:
> Do you have some TCP data where it shows there is an parsing error?

No, that was a guess. But I think that I was on the wrong track. 
I have now played again with the signal finder in Enigma and then found that the error also occurs there. However, if I activate the following in the tuner option: 
"force legacy signal stats" in the tuner option it works and the signal strength is constant at 99%..... The description says that the data is then calculated from a V3 API . maybe that tells you something?
I can live with the error but it would be nice if it could be solved somehow. but now I'm not sure if this is a satpi problem at all.

Barracuda09, 2022-01-23:
Yes, In SatPI there is something similar. Only it will switch to old API when it can not read the new one.

https://github.com/Barracuda09/SATPI/blob/6b90b138ec2903254be982793a485dd43d7aa05e/src/input/dvb/Frontend.cpp#L307-L310

SaulGoodman1337, 2022-01-23:
ah, we're getting closer.
I think it would make sense to offer the following setting for the option in the web interface:
- Automatic
- V5? Api
- V3 API (deprecated)

how does that sound to you?

Barracuda09, 2022-01-23:
Yes perfect!!

Here is an test version:
[satpi_armv7ve_force_oldapi_V1.6.2-11-g6b90b13.tar.gz](https://github.com/Barracuda09/SATPI/files/7920692/satpi_armv7ve_force_oldapi_V1.6.2-11-g6b90b13.tar.gz)
t

SaulGoodman1337, 2022-01-23:
> Yes perfect!!
> 
> Here is an test version: [satpi_armv7ve_force_oldapi_V1.6.2-11-g6b90b13.tar.gz](https://github.com/Barracuda09/SATPI/files/7920692/satpi_armv7ve_force_oldapi_V1.6.2-11-g6b90b13.tar.gz)

it does not work,sorry. i have checked several times to make sure that i have really pushed the correct binary onto the box.

Barracuda09, 2022-01-23:
Mmmm... this one then?
[satpi_armv7ve_force_oldapi_V1.6.2-11-g6b90b13.tar.gz](https://github.com/Barracuda09/SATPI/files/7920716/satpi_armv7ve_force_oldapi_V1.6.2-11-g6b90b13.tar.gz)

SaulGoodman1337, 2022-01-23:
> Mmmm... this one then? [satpi_armv7ve_force_oldapi_V1.6.2-11-g6b90b13.tar.gz](https://github.com/Barracuda09/SATPI/files/7920716/satpi_armv7ve_force_oldapi_V1.6.2-11-g6b90b13.tar.gz)

nope...

Barracuda09, 2022-01-23:
Strange.. So it is the same outcome??

SaulGoodman1337, 2022-01-23:
> Strange.. So it is the same outcome??

yes exactly the same behavior as before too. i'll check the satfinder again...

SaulGoodman1337, 2022-01-23:
[image]

99% on all transponder with no problems... hmm

Saentist, 2022-01-23:
Change stb image and problem will disappear.
This is common problem.

SaulGoodman1337, 2022-01-25:
> Change stb image and problem will disappear. This is common problem.

I will try this out today and report back.

SaulGoodman1337, 2022-01-27:
> Change stb image and problem will disappear. This is common problem.

Any recomendation?

Barracuda09, 2022-01-27:
@SaulGoodman1337 

What are you using now?

SaulGoodman1337, 2022-01-27:
> @SaulGoodman1337
> 
> What are you using now?

Openatv 6.5

Barracuda09, 2022-01-27:
Ok, I did not know of this problem (or if this is a driver issue). But I do not use openatv on my Gigablue and Vu+ FBC receivers.

Do you use Enigma?

I use on Gigablue the TeamBlue image and Vu+ just stock image. But I do not use Enigma
I use openatv on my Vu+ zero 4k and have Enigma runing. but do not see this problem there.

SaulGoodman1337, 2022-01-27:
I use the box completely headless. Tomorrow I will try it with the teamblue image.

SaulGoodman1337, 2022-01-31:
> Ok, but did you disable or prevent Enigma from starting then?

Nope. How ?
With teamblue Image i have the same behavior

Barracuda09, 2022-01-31:
In `usr/bin/enigma2.sh` there are lines that call the bin `/usr/bin/enigma2` just put those line in comment with #

SaulGoodman1337, 2022-02-04:
> In `usr/bin/enigma2.sh` there are lines that call the bin `/usr/bin/enigma2` just put those line in comment with #

Make No difference... How can i help you to debug this?
I want wo Check If the value is coming wrong from libdvb or Driver or If this is a parsing issue inside satpi

Barracuda09, 2022-02-04:
I don't think it is a pasring issue, because it is already faulty in the Web interface and no parsing is in that code path.

Here is the function that gets the status from linux/driver:
https://github.com/Barracuda09/SATPI/blob/6b90b138ec2903254be982793a485dd43d7aa05e/src/input/dvb/Frontend.cpp#L292-L397

SaulGoodman1337, 2022-02-04:
so....
`strength = (strength * 240) / 0xffff;
SI_LOG_INFO("test @#1", DIGIT(strength, 3));
strength = 240;`

`root@gbue4k:~# tail -f /var/log/messages|grep "test "
Feb  4 09:55:58 gbue4k local1.info SatPI[23035]: test 237
Feb  4 09:55:58 gbue4k local1.info SatPI[23035]: test 240
Feb  4 09:55:58 gbue4k local1.info SatPI[23035]: test 004
Feb  4 09:55:58 gbue4k local1.info SatPI[23035]: test 002
Feb  4 09:55:59 gbue4k local1.info SatPI[23035]: test 232
Feb  4 09:55:59 gbue4k local1.info SatPI[23035]: test 237
Feb  4 09:55:59 gbue4k local1.info SatPI[23035]: test 240
Feb  4 09:55:59 gbue4k local1.info SatPI[23035]: test 004
Feb  4 09:55:59 gbue4k local1.info SatPI[23035]: test 002
Feb  4 09:55:59 gbue4k local1.info SatPI[23035]: test 232
Feb  4 09:55:59 gbue4k local1.info SatPI[23035]: test 237
Feb  4 09:55:59 gbue4k local1.info SatPI[23035]: test 240
Feb  4 09:55:59 gbue4k local1.info SatPI[23035]: test 004
Feb  4 09:56:00 gbue4k local1.info SatPI[23035]: test 002
Feb  4 09:56:00 gbue4k local1.info SatPI[23035]: test 232
Feb  4 09:56:00 gbue4k local1.info SatPI[23035]: test 237
Feb  4 09:56:00 gbue4k local1.info SatPI[23035]: test 240
Feb  4 09:56:00 gbue4k local1.info SatPI[23035]: test 004
Feb  4 09:56:00 gbue4k local1.info SatPI[23035]: test 002
Feb  4 09:56:00 gbue4k local1.info SatPI[23035]: test 232
Feb  4 09:56:00 gbue4k local1.info SatPI[23035]: test 237
Feb  4 09:56:00 gbue4k local1.info SatPI[23035]: test 240
Feb  4 09:56:00 gbue4k local1.info SatPI[23035]: test 004`

so what is now certain is that the values already come incorrectly from the driver (?).
This is bad. But if you look at the values that come out of there, I still do not get rid of the feeling that here is parsed wrong. But just on driver level.... Ne idea what I can do here? The setting of the signal strength to 240 is also nice, but not the point of the thing.

I hope you have another idea ....

Barracuda09, 2022-02-04:
Does it use the old style read?

SaulGoodman1337, 2022-02-04:
> Does it use the old style read?

yep and i dont know why. have you made changes to this code? if FULL_DVB_API_VERSION >= 0x050A is comment out on my local branch. 
[image]

Barracuda09, 2022-02-04:
Could you try to find out why? Put some SI_LOG_INFO(..) in the switch statement?

SaulGoodman1337, 2022-02-04:
SatPI[30284]: Debug: FULL_DVB_API_VERSION= 1290
SatPI[30284]: Debug: Do Old Api

Barracuda09, 2022-02-04:
Please try to add SI_LOG_INFO in these 3 places:
```
		case FE_SCALE_NOT_AVAILABLE:
		default:
			_oldApiCallStats = true;
			SI_LOG_INFO("strengthScale error -> use old api");
		break;

```

SaulGoodman1337, 2022-02-04:
> Please try to add SI_LOG_INFO in these 3 places:
> 
> ```
> 		case FE_SCALE_NOT_AVAILABLE:
> 		default:
> 			_oldApiCallStats = true;
> 			SI_LOG_INFO("strengthScale error -> use old api");
> 		break;
> ```

there a no output if i use SU_LOG_INFO() inside the default. 
the cmdseq before is already empty.

`                                const auto strengthScale = cmdseq.props[0].u.st.stat[0].scale;
                                SI_LOG_INFO("Debug: strengthScale= @#1", strengthScale);`

Feb  4 12:23:41 gbue4k local1.info SatPI[32346]: Debug: FULL_DVB_API_VERSION= 1290
Feb  4 12:23:41 gbue4k local1.info SatPI[32346]: Debug: Do New Api
Feb  4 12:23:41 gbue4k local1.info SatPI[32346]: Debug: strengthScale=
Feb  4 12:23:41 gbue4k local1.info SatPI[32346]: Debug: cnrScale=
Feb  4 12:23:41 gbue4k local1.info SatPI[32346]: Debug: berScale=
Feb  4 12:23:41 gbue4k local1.info SatPI[32346]: Debug: Do Old Api
Feb  4 12:23:41 gbue4k local1.info SatPI[32346]: Debug: FULL_DVB_API_VERSION= 1290
Feb  4 12:23:41 gbue4k local1.info SatPI[32346]: Debug: Do Old Api
Feb  4 12:23:42 gbue4k local1.info SatPI[32346]: Debug: FULL_DVB_API_VERSION= 1290
Feb  4 12:23:42 gbue4k local1.info SatPI[32346]: Debug: Do Old Api
Feb  4 12:23:42 gbue4k local1.info SatPI[32346]: Debug: FULL_DVB_API_VERSION= 1290
.......
.......

SaulGoodman1337, 2022-02-04:
sorry my fault. here the correct output
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: FULL_DVB_API_VERSION= 1290
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: Do New Api
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: ioctl(_fd_fe, FE_GET_PROPERTY, &cmdseq) = 0
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: strengthScale=
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: strengthScale error -> use old api
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: cnrScale=
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: cnrScale error -> use old api
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: berScale=
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: berScale error -> use old api
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: Do Old Api
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: FULL_DVB_API_VERSION= 1290
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: Do Old Api
Feb  4 13:02:53 gbue4k local1.info SatPI[2425]: Debug: FULL_DVB_API_VERSION= 1290
....
.....

Barracuda09, 2022-02-04:
You could build with `make debug` to have line numbers and file-names

Is it the same on Frontent 2?

SaulGoodman1337, 2022-02-04:
> You could build with `make debug` to have line numbers and file-names
> 

> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: RTSP Connection from 192.168.178.2 Port 38506 with fd: 9
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: RTSP Stream data from client None with IP 192.168.178.2 on Port 38506: SETUP rtsp://192.168.178.4/?src=1&fe=1&freq=11052.75&sr=22000&msys=dvbs2&mtype=8psk&pol=h&fec=23&ro=0.35&pids=0 RTSP/1.0
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Transport: RTP/AVP/TCP;interleaved=0-1
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: CSeq: 1
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: <CRLF>
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Found FrondtendID: 0 (fe=1)  StreamID: -1  SessionID: 0280456878
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, StreamClient[0] with SessionID 0280456878 for dvbs2
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Parsing transport parameters...
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, New frequency requested, clearing old channel data...
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Frontend: 0, Parsing transport parameters (Finished)
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Frontend: 0, Found Streaming type: RTP/TCP
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Updating frontend...
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Closing all active PID filters...
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 10 (0 ms)
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Start tuning process for DVB-S(2)...
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Opened /dev/dvb/adapter0/frontend1 for Writing DiSEqC command with fd: 11
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Sending DiSEqC: [0x70] [0x04] [0xB2] [0x02] - DiSEqC Src: 0 - UB: 0
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Closing /dev/dvb/adapter0/frontend1 with fd: 11
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Frontend: 0, Set Properties: Frequency 1210000
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Tuned, waiting on lock...
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Not locked yet   (FE status 0x01)...
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Tuned and locked (FE status 0x1F)
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Updating PID filters...
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Opened /dev/dvb/adapter0/demux0 using fd: 11
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Set DMX buffer size to 18874368 Bytes
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Set DMX_SET_SOURCE with (Src: 0 - Offset: 18)
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Frontend: 0, Set filter PID: 0000
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Frontend: 0, Set filter PID: 0001
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Frontend: 0, Set filter PID: 0016
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Frontend: 0, Set filter PID: 0017
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Frontend: 0, Set filter PID: 0018
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Updating frontend (Finished in 640 ms)
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, RTP/TCP set network buffer size: 437 KBytes
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Start RTCP/TCP stream to 192.168.178.2:38506
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Start RTP/TCP stream to 192.168.178.2:38506
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Send reply in 641 ms
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: RTSP/1.0 200 OK
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: CSeq: 1
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Session: 0280456878;timeout=60
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Transport: RTP/AVP/TCP;unicast;client_ip=192.168.178.2;interleaved=0-1
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: com.ses.streamID: 100
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: <CRLF>
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Debug: FULL_DVB_API_VERSION= 1290
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Debug: Do Old Api
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: RTSP Stream data from client None with IP 192.168.178.2 on Port 38506: PLAY rtsp://192.168.178.4/stream=100?addpids=1,16,17,18,20,5420,5421,5422,5423,5425,5426 RTSP/1.0
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Session: 0280456878
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: CSeq: 2
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: <CRLF>
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Found FrondtendID: 0 (fe=1)  StreamID: 100  SessionID: 0280456878
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, StreamClient[0] with SessionID 0280456878
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Parsing transport parameters...
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Frontend: 0, Parsing transport parameters (Finished)
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Updating frontend...
> Feb  4 13:14:38 gbue4k local1.info SatPI[3029]: Frontend: 0, Updating PID filters...
> Feb  4 13:14:38 gbue4k local1.debug SatPI[3029]: Frontend: 0, Set filter PID: 0020
[... cut ...]
> Feb  4 13:14:45 gbue4k local1.info SatPI[3029]: Frontend: 0, Closing /dev/dvb/adapter0/frontend0 fd: 10
> Feb  4 13:14:45 gbue4k local1.debug SatPI[3029]: Send reply in 405 ms
> Feb  4 13:14:45 gbue4k local1.debug SatPI[3029]: RTSP/1.0 200 OK
> Feb  4 13:14:45 gbue4k local1.debug SatPI[3029]: CSeq: 4
> Feb  4 13:14:45 gbue4k local1.debug SatPI[3029]: Session: 0280456878
> Feb  4 13:14:45 gbue4k local1.debug SatPI[3029]: <CRLF>
> Feb  4 13:14:45 gbue4k local1.info SatPI[3029]: RTSP Client 192.168.178.2:38506 Connection closed with fd: 9

> Is it the same on Frontent 2?

yep.

Barracuda09, 2022-02-04:
Ok, I hook-up my Gigablue quad 4k this evening (or this weekend).

I currently have the Vu+ Zero 4K for testing, but I do not see this behavior on this receiver.

SaulGoodman1337, 2022-02-04:
very nice. thank you.

Barracuda09, 2022-02-04:
If you switch off the receiver and on again is the problem then directly present?

Barracuda09, 2022-02-04:
I hooked up the Gigablue receiver on a universal LNB, but unfortunately do not see the jumping in the signal strength.

SaulGoodman1337, 2022-02-06:
hmmm. minisatip
Feb  6 07:54:12 gbue4k daemon.notice minisatip[11314]: get_signal took 48 ms for adapter 0 handle 6 (status: 31, ber: 0, strength:23, snr: 217, force scan 0)

strength is 23 instead of 23[0..9]...
the values already come out of the driver wrong. i still can't get rid of the feeling that something is going wrong with the parsing. in the driver itself!
the values do not look like fantasy values, but they are chopped off. 
Do you have an idea or maybe a contact to which I can turn?

Barracuda09, 2022-02-06:
What does this show:
`modinfo /usr/share/platform/dvb.ko`

Saentist, 2022-02-06:
What happen if in place of enigma2 used neutrino?

SaulGoodman1337, 2022-02-06:
> What does this show: `modinfo /usr/share/platform/dvb.ko`

minisatip: unrecognized option '--cleanpsi'
root@gbue4k:~# modinfo /usr/share/platform/dvb.ko
filename:       /usr/share/platform/dvb.ko
license:        GPL
description:    BCM7xxx dvb adapter driver
version:        20200723104807-u171
depends:
vermagic:       4.1.20-1.9 SMP mod_unload ARMv7 p2v8

Barracuda09, 2022-02-06:
Ok, I have to same driver data on my Quad 4K. 
Strange I do not see this problem here

Barracuda09, 2022-02-07:
Have to say that (to me) the signal strength on Gigablue is somewhat buggy. 
I do not see the jumping but now I have connected to the JESS LNB it seems somehow different a much(much) lower value.

SaulGoodman1337, 2022-02-07:
Which values ? 
For me ist seems that the 3 Digit values are chopped. Instead of 240 i have 22,23 or 24. Somethimes 38... Somethimes 4.. 
I think this is a Driver issue and No Problem that you can fix. But... In minisatip which i Test at the moment(sorry for that) is the Option to force the Signal strenght to a value.

I think this is a good Feature Request. Easy to implement i think.

Barracuda09, 2022-02-07:
Yes does values I have seen as well, as well just half the value.