Barracuda09/SATPI #148: make LIBDVBCSA=yes on main Branch failed (getFrontendDecryptInterface(FeID))

SaulGoodman1337, 2022-01-31:
`root@astra:/home/astra/satpi# make LIBDVBCSA=yes
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS src/Version.cpp -o obj/Version.o
g++ obj/Version.o obj/InterfaceAttr.o obj/HeaderVector.o obj/HttpServer.o obj/HttpcServer.o obj/Log.o obj/Properties.o obj/RtspServer.o obj/main.o obj/Satpi.o obj/Stream.o obj/StreamClient.o obj/StreamManager.o obj/StringConverter.o obj/TransportParamVector.o obj/Utils.o obj/base/M3UParser.o obj/base/Thread.o obj/base/ThreadBase.o obj/base/TimeCounter.o obj/base/XMLSaveSupport.o obj/base/XMLSupport.o obj/input/DeviceData.o obj/input/Transformation.o obj/input/dvb/Frontend.o obj/input/dvb/FrontendData.o obj/input/dvb/delivery/DiSEqc.o obj/input/dvb/delivery/DiSEqcEN50494.o obj/input/dvb/delivery/DiSEqcEN50607.o obj/input/dvb/delivery/DiSEqcLnb.o obj/input/dvb/delivery/DiSEqcSwitch.o obj/input/dvb/delivery/DVBC.o obj/input/dvb/delivery/DVBS.o obj/input/dvb/delivery/DVBT.o obj/input/dvb/delivery/Lnb.o obj/input/file/TSReader.o obj/input/file/TSReaderData.o obj/input/childpipe/TSReader.o obj/input/childpipe/TSReaderData.o obj/input/stream/Streamer.o obj/input/stream/StreamerData.o obj/mpegts/Filter.o obj/mpegts/PacketBuffer.o obj/mpegts/PAT.o obj/mpegts/PCR.o obj/mpegts/PidTable.o obj/mpegts/PMT.o obj/mpegts/SDT.o obj/mpegts/TableData.o obj/output/StreamThreadBase.o obj/output/StreamThreadHttp.o obj/output/StreamThreadRtcpBase.o obj/output/StreamThreadRtcp.o obj/output/StreamThreadRtcpTcp.o obj/output/StreamThreadRtp.o obj/output/StreamThreadRtpTcp.o obj/output/StreamThreadTSWriter.o obj/socket/HttpcSocket.o obj/socket/TcpSocket.o obj/socket/SocketAttr.o obj/socket/UdpSocket.o obj/upnp/ssdp/Server.o obj/decrypt/dvbapi/Client.o obj/decrypt/dvbapi/ClientProperties.o obj/decrypt/dvbapi/Keys.o obj/input/dvb/Frontend_DecryptInterface.o -o satpi  -pthread -lrt -ldvbcsa
/usr/bin/ld: obj/decrypt/dvbapi/Client.o: in function `decrypt::dvbapi::Client::decrypt(FeID, mpegts::PacketBuffer&)':
/home/astra/satpi/src/decrypt/dvbapi/Client.cpp:97: undefined reference to `StreamManager::getFrontendDecryptInterface(FeID)'
/usr/bin/ld: obj/decrypt/dvbapi/Client.o: in function `decrypt::dvbapi::Client::stopDecrypt(FeID)':
/home/astra/satpi/src/decrypt/dvbapi/Client.cpp:211: undefined reference to `StreamManager::getFrontendDecryptInterface(FeID)'
/usr/bin/ld: obj/decrypt/dvbapi/Client.o: in function `decrypt::dvbapi::Client::threadEntry()':
/home/astra/satpi/src/decrypt/dvbapi/Client.cpp:468: undefined reference to `StreamManager::getFrontendDecryptInterface(FeID)'
/usr/bin/ld: /home/astra/satpi/src/decrypt/dvbapi/Client.cpp:481: undefined reference to `StreamManager::getFrontendDecryptInterface(FeID)'
/usr/bin/ld: /home/astra/satpi/src/decrypt/dvbapi/Client.cpp:496: undefined reference to `StreamManager::getFrontendDecryptInterface(FeID)'
/usr/bin/ld: obj/decrypt/dvbapi/Client.o:/home/astra/satpi/src/decrypt/dvbapi/Client.cpp:535: more undefined references to `StreamManager::getFrontendDecryptInterface(FeID)' follow
collect2: error: ld returned 1 exit status
make: *** [Makefile:197: satpi] Error 1

Building Main Branch with  LIBDVBCSA=yes. maybe releated to #145 ?

Barracuda09, 2022-01-31:
Hi @SaulGoodman1337 

I have to check this, I most of the time build it with  `LIBDVBCSA=yes` and see no build issues

SaulGoodman1337, 2022-01-31:
with commit-id 70bffd9255a0d1a152fe0583e194fe40e416033d compiling is working.with latest commit it does not work.

Barracuda09, 2022-01-31:
You can download comiler from and unpack it in `~/vu/opt/`
https://github.com/Broadcom/stbgcc-8.3/releases

then run:
source env.sh
make

See:
https://github.com/Barracuda09/SATPI/wiki/Build-SatPI

SaulGoodman1337, 2022-01-31:
thanks! its working with ENIGMA=yes. but with libdvbcsa i get following error:
`root@astra:/home/astra/satpi# make clean
Clearing project...
...Done
root@astra:/home/astra/satpi# make LIBDVBCSA=yes ENIGMA=yes
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/Version.cpp -o obj/Version.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/InterfaceAttr.cpp -o obj/InterfaceAttr.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/HttpServer.cpp -o obj/HttpServer.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/HttpcServer.cpp -o obj/HttpcServer.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/Log.cpp -o obj/Log.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/Properties.cpp -o obj/Properties.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/RtspServer.cpp -o obj/RtspServer.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/main.cpp -o obj/main.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/Satpi.cpp -o obj/Satpi.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/Stream.cpp -o obj/Stream.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/StreamClient.cpp -o obj/StreamClient.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/StreamManager.cpp -o obj/StreamManager.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/StringConverter.cpp -o obj/StringConverter.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/Utils.cpp -o obj/Utils.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/base/M3UParser.cpp -o obj/base/M3UParser.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/base/Thread.cpp -o obj/base/Thread.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/base/ThreadBase.cpp -o obj/base/ThreadBase.o
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/base/TimeCounter.cpp -o obj/base/TimeCounter.o
[... cut ...]
/root/vu/opt/toolchains/stbgcc-8.3-0.4/bin/arm-linux-c++ -c -march=armv7ve -mtune=cortex-a15 -mfpu=vfpv4 -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DUSE_DEPRECATED_DVBAPI -DHAS_NP_FUNCTIONS -DENIGMA src/decrypt/dvbapi/Client.cpp -o obj/decrypt/dvbapi/Client.o
src/decrypt/dvbapi/Client.cpp:37:11: fatal error: dvbcsa/dvbcsa.h: No such file or directory
  #include <dvbcsa/dvbcsa.h>
           ^~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:202: obj/decrypt/dvbapi/Client.o] Error 1
root@astra:/home/astra/satpi#`

how to handle this in this build environment? without ENIGMA=yes building is working.

Barracuda09, 2022-01-31:
You need to copy Lib and Header file to the build env

Barracuda09, 2022-01-31:
Copy the lib from the receiver

Barracuda09, 2022-01-31:
You need to copy lib and header somewere in `vu/opt/toolchains/stbgcc-8.3-0.4` (I do not know at the moment were. Have to look when I am at home again)

SaulGoodman1337, 2022-01-31:
sorry i can't quite follow you. i would be grateful if you could get back to me when you are at home.

Barracuda09, 2022-01-31:
The header file `dvbcsa.h` needs to go here:
`~/vu/opt/toolchains/stbgcc-8.3-0.4/arm-unknown-linux-gnueabihf/include/c++/8.3.0/dvbcsa/`

The lib file `libdvbcsa.so` needs to go here:
`~/vu/opt/toolchains/stbgcc-8.3-0.4/arm-unknown-linux-gnueabihf/lib/libdvbcsa.so`

 `libdvbcsa.so` can be found on the receiver in `/usr/lib/libdvbcsa.so.1.0.1` (just rename it to libdvbcsa.so)

SaulGoodman1337, 2022-01-31:
`root@gbue4k:/usr/lib# find / -name "libdvb*"
/var/lib/opkg/info/libdvbsi++1.postinst
/var/lib/opkg/info/libdvbsi++1.control
/var/lib/opkg/info/libdvbsi++1.list
/usr/lib/libdvbsi++.so.1
/usr/lib/libdvbsi++.so.1.0.0`

`root@gbue4k:/usr/lib# find / -name "*dvbcsa*"
root@gbue4k:/usr/lib#`

there is none /usr/lib/libdvbcsa.so.1.0.1 

iam on teamblue image now

Barracuda09, 2022-01-31:
You need to install the packages

SaulGoodman1337, 2022-01-31:
> Sorry, You need to install the packages on the receivers with something like: `opkg install libdvbcsa`

nice ! up and running ! thanks

closed 2022-01-31 by SaulGoodman1337