Barracuda09/SATPI #149: Only 2 streams works on my vuduo4kse with 2xdvbs2x

ddarek, 2022-02-06:
Hi, 
I have Vu+ Duo 4K SE with 2 x FBC DVB-S2X.

In web interface of satpi I see that only frontend 0 and 1 are  "FBC-Root".
I think that also frontend 8 and 9 should be "FBC-Root".

When I connect to SATPI with Tvheadend only two frontends can be used.

I have 2xFBC DVB-S2X phisical tuners. 
I see Frontends 0,1,2,3,4,5,6,7,8,9,10,11,12,14,15
0,1 - Root of first DVB-S2X
8,9 - Root of second DVB-S2x
2,3,4,5,6,7 - virtuals
10,11,12,13,14,15 - virtuals

In Tvheadend only 2 stream at the same time works ok, if more then streams are interrupted, no tuned.

Do you know the reason why it don't work?
Maybe my configuration is wrong?
Should I see 4 root frontends in case of my device ?

Best regards
Dariusz

Barracuda09, 2022-02-06:
Hello @ddarek ,

I only own receivers with only 1 FBC tuner DVB-S2 (So only 0 - 7). 
I have a GB quad which can fit one extra, but currently empty.
So sorry, I have not tested this situation myself.

A few questions:
- Is enigma2 still runing
- Which tuners are used for the streams?

ddarek, 2022-02-06:
Hi,
Thanks for reply,
I have tested without running enigma2.
In enigma2:
-  I have 4 frontends configured: a,b, i, j (simple, diseqcx4 with a and b set for astra and hb).
- The rest of frontends (c,d,e,f,g,,k,l,m,n,o,p) are configured automatically.

I have legacy Quad LNB connected with 4 cables to those 2fbc modules.

Satpi web int. says that only frontend1 and frontend2 are Fbc-root.

Only 2 streams works stable (a and b). 
If I try to use more than 2, then they are unstable.

It looks like frontend 8 and 9 are treated like virtual for first dvbs2x card. 
In enigma2 frontend 8 and 9 are treated like a roots( can be configured) for virtual 10,11,12,13,14,15.

ddarek, 2022-02-06:
In my case I have:
in /proc/stb/frontend i have 15 entries.
first 8 (0-7) has /proc/stb/frontend/x/fbc_set_id with value 0 (I think it means: dvbs2x card 1)
second 8 (8-15)  has  /proc/stb/frontend/x/fbc_set_id with value 1 (I think it means: dvbs2x card 2)

Barracuda09, 2022-02-06:
Yes, I did not foresee this, with 2 FBC tuners.

ddarek, 2022-02-06:
Could you add support for that case ?

Barracuda09, 2022-02-06:
Yes, I could do that.
But first I need to purchase a second tuner for my GB, to test it myself.

Barracuda09, 2022-02-07:
@ddarek 

Could you please provide me the contents of dir (recursive):
- /proc/stb/frontend

Could you give me the contents of these files:
- /proc/stb/frontend/@#1/fbc_connect_choices
- /proc/stb/frontend/@#1/fbc_set_id
- /proc/stb/frontend/@#1/fbc_link
- /proc/stb/frontend/@#1/fbc_connect

ddarek, 2022-02-07:
Sure, 
Here's the full, recursive content of dir /proc/stb/frontent
[frontend.tar.gz](https://github.com/Barracuda09/SATPI/files/8015129/frontend.tar.gz)

Barracuda09, 2022-02-07:
Did you change manually contents of these files:
- `/frontend/8/fbc_connect` to 8
- `/frontend/9/fbc_connect` to 9

ddarek, 2022-02-07:
No, It's axactly as it is.
Adapters are in use by enigma2 and that's all.

Barracuda09, 2022-02-07:
Ok, that nice. So there is just on offset for the second FBC tuner.

ddarek, 2022-02-07:
Yes, I think so.

Barracuda09, 2022-02-07:
Did you build from source or from image repo

ddarek, 2022-02-07:
image repo

Barracuda09, 2022-02-07:
Ok, possible thats quite an old version. Could you try to build it from source?

ddarek, 2022-02-07:
Ok, but later this evening. 
I will try it this evening. I have to refresh my memory about toolchain... :)

Barracuda09, 2022-02-07:
Try to look here:
https://github.com/Barracuda09/SATPI/wiki/Cross-Compiling-SatPI

ddarek, 2022-02-08:
Hi, 
Sorry for a delay but I had no time yesterday.
I have :
- killed enigma2 process.
- started satpi with enabled 0,1,8,9 frontends
- set diseqc for all those frontends
- set channel a frontend connections : 0=A,1=B,8=A,9=B
- restarted satpi process
- made copy of /proc/stb/frontend
- saved log file.

[frontend_and_log.tar.gz](https://github.com/Barracuda09/SATPI/files/8026579/frontend_and_log.tar.gz)

Barracuda09, 2022-02-08:
Did you change in the webUI the frontend 8 `FBC Connection`

ddarek, 2022-02-08:
yes
Frontend 8 : FBC Connection=A
Frontend 9 : FBC Connection=B

ddarek, 2022-02-08:
I have found that frontends 0-7 have fbc_set_id=0 (/proc/stb/frontend/X/fbc_set_id)
frontends 8-15 have fbc_set_id=1

Barracuda09, 2022-02-08:
Yes, you mentioned that frontend 8,9 seem to be connected to 0,1?

ddarek, 2022-02-08:
Yes, 
Frontends 0-7 are for FBC-1
Frontends 8-15 are for FBC-2

I managed to work 4 streams on different server app. 
4 streams use 0,1,8,9 frontends (because I use in legacy lnb configuration with diseqc).
Frontend : 
0 is a FBC-0-A
1 is a FBC-1-B
8 is a FBC-8-A
9 is a FBC-9-B

I'm testing it on tvheadend connected to this app and it seems to work stable ( I managed to do this 1h ago ) :)
So I'm almost sure that problem is that SATPI don't see that 8 and 9 are FBC-Roots.

Barracuda09, 2022-02-08:
Yes the problem is that SatPI writes to 
- /proc/stb/frontend/8/fbc_connect a 0 that should be 8
- /proc/stb/frontend/9/fbc_connect a 1 that should be 9

If you do pobably (wuthout changing anything in the webUI:
- echo 8 > /proc/stb/frontend/8/fbc_connect
- echo 9 > /proc/stb/frontend/9/fbc_connect

ddarek, 2022-02-08:
Hmm, it looks that currently on other app only 3 streams are working, if I add 4th then it's getting unstable and last to are disrupting them selves :)
With : 
echo 8 > /proc/stb/frontend/8/fbc_connect
echo 9 > /proc/stb/frontend/9/fbc_connect
and with 
echo 0 > /proc/stb/frontend/8/fbc_connect
echo 1 > /proc/stb/frontend/9/fbc_connect
it  works the same without any change.

ddarek, 2022-02-08:
It started to be unstable, but after restarting tvheadend it's ok now, so it was a tvheadend problem now (on different server app)

Barracuda09, 2022-02-08:
I will have a look at this, to support the second FBC tuner (NOTES).
It's probably the offset that needs to be added to selection A/B of `FBC Connection` when fbc_set_id = 1 (`variable _fbcSetID`).
https://github.com/Barracuda09/SATPI/blob/6b90b138ec2903254be982793a485dd43d7aa05e/src/input/dvb/delivery/DVBS.cpp#L52-L65

Adapt this to include the offset depending on _fbcSetID
https://github.com/Barracuda09/SATPI/blob/6b90b138ec2903254be982793a485dd43d7aa05e/src/input/dvb/delivery/DVBS.cpp#L291-L300

Barracuda09, 2022-02-09:
I have implemented the support for the second(Slot B) DVB-S2X FBC tuner

ddarek, 2022-02-09:
Hi, great!!
Change looks promising. 
I will compile it tomorrow, test it and give you a feedback.
Thanks

ddarek, 2022-02-09:
I have compiled and run it.
Unfortunately fix don't work :(

Frontend 0 = DVB-S(2) FBC-Root (Slot-A)
Frontend 1 = DVB-S(2) FBC-Root (Slot-A)
Frontend 2 - 15 = DVB-S(2) FBC-Root (Slot-A)

I didn't analyze the change - I'm java dev, I didn't work with c++ since long time.
I think slot-b root0 should be the first frontend with fbc_set_id=1.

I attach the log SATPI log file and content of /proc/stb/frontend dir.
[proc.tar.gz](https://github.com/Barracuda09/SATPI/files/8035867/proc.tar.gz)

Barracuda09, 2022-02-09:
Sorry, forgot some offset adjustments. Now it should make more sense.

Barracuda09, 2022-02-09:
Recheck your settings in SatPI, it could be corrupted because of previous version.

ddarek, 2022-02-10:
Now It's ok :)
Fronteds 8 and 9 are visible as roots.
Works stable with tvheadend. Thanks!!

Barracuda09, 2022-02-10:
Thank you for reporting and info.

Barracuda09, 2023-09-05:
Closing this issue.
It is reported in https://github.com/Barracuda09/SATPI/issues/149#issuecomment-1034556047 that this is working.

closed 2023-09-05 by Barracuda09