Barracuda09/SATPI #151: Request: Unicable/SCR static and dynamic support

lars18th, 2022-02-11:
Hi @Barracuda09 ,

I feel you're doing some new development to improve the wire management (based on #150). So I want to add this request:

- Use case: The target environment is like this. Over 1 coax wire only, four different SCR LNBs are connected. They're configured to work as one multisystem, so each LNB is A, B, C or D only, for four different sat positions. And to this wire are connected 24 tuners (6xFBC). And all is working. However, to the same wire are connected a 5th LNB configured in static mode that is injecting some transponders in free slots from a _different_ satellite position. The problem is how to tune these static transponders in SATPI.
- Request: The idea is to add to the UNICABLE configuration a list of _static frequencies_. Then when some of these frequencies are requested, the tuner, instead of send the Unicable command (or after sending a teardown/close for the assigned slot), will change to this frequency as when it works with an universal LNB (instead of use the slot frequency assigned), but not sending any voltage/tone change (only powering the output).

I feel this could be done with the current unicable API. As the static frequencies could correspond to regular (free) SCR slots. And futhermore, if you really need to send the unicable commands, then you could send to a "fake" position (i.e.64?). And then all the real SCR LNBs will not process the command and don't use the slot assigned to the static frequency requested. Futhermore, from the tuner client side this will be like changing from one slot to another. Therefore, I think this could be feasible and it will work fast without troubles.

You could try to implement this, please?
Thank you, and please continue with this great project.

Barracuda09, 2022-02-11:
So as I understand it now, it is only the frontend that tunes to a requested freq. so now DiSEqC or 'real' LNB connected

lars18th, 2022-02-11:
Hi @Barracuda09 ,

I'm not sure about your comment. The 5th LNB is an another programable SCR/Unicable LNB that is configured to not listen for any JESS/SCR command. Just to pass a fixed transponder/frequency to a fixed slot all the time. That's corresponds to the "static" frequencies that I want to tune with the unicable tuners. Then for each unicable tuner, not only the slot frequency will be in use, but also the static ones (plus some small difference, to round the frequency). This list of static frequencies will be similar to the remapping frequency list. But in this case they are not remapping to anything else. They're only changing the method to "tune" to the target frequency: instead of sending a command to the LNB to center it in the listening frequency, the tuner will change the listening frequency from the current slot to a "new" slot (the static one).

You could implement this, please?

Barracuda09, 2022-02-11:
So the 'static' frequency should be a input in the Unicable settings, and if the frontend receives this frequency it should not send DiSEqC command and/or setup the LNB only tune to this frequency

lars18th, 2022-02-11:
> So the 'static' frequency should be a input in the Unicable settings, and if the frontend receives this frequency it should not send DiSEqC command and/or setup the LNB only tune to this frequency

YES!

Example of the configuration:

- Tuner 1: SCR, slot 1680 MHz
- Tuner 2: SCR, slot 1280 MHz
- Tuner 3: SCR, slot 1380 MHz
- Tuner 4: SCR, slot 1480 MHz
- Tuner 5: SCR, slot 980 MHz
- Tuner 6: SCR, slot 1030 MHz
- Tuner 7: SCR, slot 1080 MHz
- Tuner 8: SCR, slot 1130 MHz
- Tuner 9: SCR, slot 1530 MHz
- Tuner 10: SCR, slot 1580 MHz
- Tuner 11: SCR, slot 1630 MHz
- Tuner 12: SCR, slot 1730 MHz
- Tuner 13: SCR, slot 1780 MHz
- Tuner 14: SCR, slot 1830 MHz
- Tuner 15: SCR, slot 1880 MHz
- Tuner 16: SCR, slot 1930 MHz
- Static frequency list: 2025, 2075, 2125 MHz

So if the client request this:
- 1975MHz, then select one free tuner, and send JESS for 1975MHz.
- 2025MHz, this is a static frequency. So, select a free LNB, disable the slot (if it's in use), and change the tuner freq. to 2025.
- 1825MHz, if the previous tuner goes free, then reconfigure it for its slot, and send the JESS command to request 1825MHz.

I feel it's simple to implement, right?
Regards.

Barracuda09, 2022-02-11:
> I feel it's simple to implement, right?

Yes, I feel it should not be to much trouble.