Barracuda09/SATPI #157: Problems starting Sapti with Zerotier

patacalas, 2022-04-14:
I'm using Satpi with Zerotier.  
I have modified init.d script with --iface-name forcing Zerotier interface use.
It works almost fine but there is a problem when booting as Zerotier uses a software interface and it doesn't exists when system starts. 
If I restart Satip service when system just booted it works like a charm.
I have added "sleep 5" on init.d script and now it works fine, even when I reboot the system.  

Is there any "elegant" solution to avoid service restart or the sleep command?

Barracuda09, 2022-04-15:
Hi @patacalas 

You could look into `LSB Headers` SysV init scripts, here is an example:

```
### BEGIN INIT INFO
# Provides: satpi
# Required-Start: $network
# Required-Stop: $network
# Default-Start: 2 3 4 5
# Default-Stop: 0 1 6
# Short-Description: Starts the SatPI Server
# Description: SatPI is a SAT>IP Server currently supporting DVB-S/S2/T/C and transform DVB-S requests to DVB-C
### END INIT INFO
```

If you have found a working example, could you please post it here for reference ?

Barracuda09, 2024-04-10:
This should be fixed by this commit [Change: Ability to bind to 0.0.0.0 when no adapter param is given](f2de19b)