Barracuda09/SATPI #158: What's the correct setup for using frontends connected to different LNBs?

patacalas, 2022-05-10:
Hi again.
I can't find the way to configure my Sat>IP server on a Gigablue UHD Quad receiver with an extra PnP tuner.
FBC tuners (frontends 0-7) are using Unicable LNBs:
Astra - Position A
Hotbird - Position B
Hispasat - Position C

I've just added an extra twin tuner (frontends 8-9), using a twin legacy LNB for Eutelsat 7B/C, and getting signal from that satellite via Sat>IP is impossible for me. 

Frontends 8 and 9 are set as LNB (no diseqc), but when I try to get an Eutelsat 7B/C channel Sat>IP serves the first frontend available and not the correct one, even using the "fe" query attribute. 
The only way I can get Eutelsat 7B/C is disabling frontends 0 to 7.
 
How can I get it working?  

Thanks in advance for your response.

Barracuda09, 2022-05-10:
Hi @patacalas 

So `fe=8` does not select frontend 8?
Could you make an example request with VLC (with `fe=8`)and give the log of SatPI?

Thanks for testing.

Barracuda09, 2022-05-10:
Hi @patacalas 

Yes you are correct, this is not intuitive. I have to maybe number the Frontends in the WebIF accordingly.
according to the specs `fe=x` should be between 1 and 255

Thanks for testing

closed 2023-05-09 by Barracuda09