Barracuda09/SATPI #160: Problems with multistream channels

patacalas, 2022-07-20:
Hi! I'm trying to tune in Italian multistream channels from Eutelsat 9ºE via SatPI and I've just encountered some problems.
My SatPI server is installed on a Gigablue UHD Quad 4k (with PnP DVB-S2x multistream tuner) and my client is a Vu+ Solo SE.

As you can see in the following pictures, I can tune some channels.

[image]
[image]

The first issue is about modulation. Some transponders use 16APSK modulation, but SatPI seems to ignore it. I checked the log when tuning 16APSK channels and it shows QPSK on log file. So, I can only tune 8PSK transponders like 12466V and 12341V. 

***LOG***
rtsp://192.168.194.39:554/stream=109?fe=10&src=1&freq=12380&pol=v&msys=dvbs2&sr=31400&fec=34&ro=0.35&mtype=qpsk&plts=off&isi=25&plsc=262140

The second issue is about stream number param. SatPI seems to ignore stream number, it always returns the same channels (channels from first stream) even with different Stream ID. As you can see in lines above, the stream "isi" param seems to be parsed.

I would be grateful if you could fix these problems. Nowadays, there are more and more satellite TV channels using multistream technology.

Thanks in advance.

Barracuda09, 2022-07-20:
Hi @patacalas 

Yes the parameters **isi** and **plsc** are not parsed at the moment
The **16APSK** and **32APSK** etc is not parsed correctly as well at the moment

Barracuda09, 2022-07-21:
Hi @patacalas 

Were are these parameters **isi** and **plsc** defined (or from which program is this rtsp uri come from) because they are not part of the SAT>IP specification.

patacalas, 2022-07-21:
Hi @Barracuda09 
This rtsp uri comes from Enigma2 Sat-IP client when trying to tune in a multistream transponder.
These params are not from SAT>IP specifications, but I guess it should be improved to meet new standarts from DVB-S2.
[https://www.etsi.org/deliver/etsi_en/302300_302399/30230701/01.04.01_60/en_30230701v010401p.pdf](url)
Check chapter 5.5.4 from PDF file

I did a Google search and it seems that minisatip added multistream support a few years ago
[https://github.com/catalinii/minisatip/pull/464/commits/a964608d62063169dc50480c911071ad847377c1](url)

It would be great if SatPI could handle multistream transponders as there are more and more channels using this technology.

Barracuda09, 2022-07-21:
Hi @patacalas 

Could you try this version if you have the time? 
[satpi_arm_1.6.2.30~gc701753 Enigma_Multistream.zip](https://github.com/Barracuda09/SATPI/files/9162863/satpi_arm_1.6.2.30.gc701753.Enigma_Multistream.zip)

Many thanks,
Marc

patacalas, 2022-07-22:
Hi @Barracuda09 
I've tested your new version but multistream channels doesn't work anymore.

[image]
The screenshot is about tuning 12466V, this TP worked in the previous version of SatPI as it uses 8PSK modulation.

****LOG****
  Fri Jul 22 03:44:54.0111 2022  7  Range: npt=0.000-
  Fri Jul 22 03:44:54.0111 2022  7  <CRLF>
  Fri Jul 22 03:44:59.0184 2022  7  RTSP Stream data from client None with IP 192.168.194.185 on Port 36576: PLAY rtsp://192.168.194.39:554/stream=109?fe=10&src=1&freq=12466&pol=v&msys=dvbs2&sr=41950&fec=56&ro=0.35&mtype=8psk&plts=off&isi=33 RTSP/1.0
  Fri Jul 22 03:44:59.0184 2022  7  CSeq: 63
  Fri Jul 22 03:44:59.0184 2022  7  Session: 0272645804
  Fri Jul 22 03:44:59.0184 2022  7  <CRLF>
  Fri Jul 22 03:44:59.0186 2022  6  Found FrondtendID: 9 (fe=10)  StreamID: 109  SessionID: 0272645804
  Fri Jul 22 03:44:59.0186 2022  6  Frontend: 9, StreamClient[0] with SessionID 0272645804 for dvbs2
  Fri Jul 22 03:44:59.0187 2022  6  Frontend: 9, Parsing transport parameters...
  Fri Jul 22 03:44:59.0187 2022  6  Frontend: 9, PLAY method with query and no pids=, clearing old pids...
  Fri Jul 22 03:44:59.0189 2022  7  Frontend: 9, Parsing transport parameters (Finished)
  Fri Jul 22 03:44:59.0189 2022  6  Frontend: 9, Updating frontend...
  Fri Jul 22 03:44:59.0190 2022  6  Frontend: 9, Updating frontend (Finished in 0 ms)
  Fri Jul 22 03:44:59.0190 2022  7  Send reply in 0 ms
  Fri Jul 22 03:44:59.0190 2022  7  RTSP/1.0 200 OK
  Fri Jul 22 03:44:59.0190 2022  7  RTP-Info: url=rtsp://192.168.194.39/stream=109
  Fri Jul 22 03:44:59.0190 2022  7  CSeq: 63
  Fri Jul 22 03:44:59.0190 2022  7  Session: 0272645804
  Fri Jul 22 03:44:59.0190 2022  7  Range: npt=0.000-
  Fri Jul 22 03:44:59.0190 2022  7  <CRLF>

Thanks for your prompt response.

patacalas, 2022-07-22:
I've tested another transponder and it worked, and *sometimes* it returns the correct stream (in previous version of SatPI was impossible to tune in different streams on the same transponder). 

[image]
TV8HD is on stream 35, the second stream of this transponder.

Later, I've tried to tune stream 36 but it returned the wrong channels (channels from stream 35 again).

***LOG***
 <CRLF>
  Fri Jul 22 04:23:33.1600 2022  7  RTSP Stream data from client None with IP 192.168.194.185 on Port 36576: PLAY rtsp://192.168.194.39:554/stream=109?fe=10&src=1&freq=12341&pol=v&msys=dvbs2&sr=31400&fec=23&ro=0.35&mtype=8psk&plts=off&isi=36&plsc=262140 RTSP/1.0
  Fri Jul 22 04:23:33.1600 2022  7  CSeq: 618
  Fri Jul 22 04:23:33.1600 2022  7  Session: 0272645804
  Fri Jul 22 04:23:33.1600 2022  7  <CRLF>
  Fri Jul 22 04:23:33.1602 2022  6  Found FrondtendID: 9 (fe=10)  StreamID: 109  SessionID: 0272645804
  Fri Jul 22 04:23:33.1602 2022  6  Frontend: 9, StreamClient[0] with SessionID 0272645804 for dvbs2
  Fri Jul 22 04:23:33.1603 2022  6  Frontend: 9, Parsing transport parameters...
  Fri Jul 22 04:23:33.1603 2022  6  Frontend: 9, PLAY method with query and no pids=, clearing old pids...
  Fri Jul 22 04:23:33.1605 2022  7  Frontend: 9, Parsing transport parameters (Finished)
  Fri Jul 22 04:23:33.1606 2022  6  Frontend: 9, Updating frontend...
  Fri Jul 22 04:23:33.1606 2022  6  Frontend: 9, Updating PID filters...
  Fri Jul 22 04:23:33.1808 2022  7  Frontend: 9, Remove filter PID: 0000 - Packet Count: 000000187:000000
  Fri Jul 22 04:23:33.2010 2022  7  Frontend: 9, Remove filter PID: 0001 - Packet Count: 000000000:000000
  Fri Jul 22 04:23:33.2212 2022  7  Frontend: 9, Remove filter PID: 0016 - Packet Count: 000000004:000000
  Fri Jul 22 04:23:33.2413 2022  7  Frontend: 9, Remove filter PID: 0017 - Packet Count: 000000076:000000
  Fri Jul 22 04:23:33.2620 2022  7  Frontend: 9, Remove filter PID: 0018 - Packet Count: 000001773:000000
  Fri Jul 22 04:23:33.2622 2022  6  Frontend: 9, Updating frontend (Finished in 101 ms)
  Fri Jul 22 04:23:33.2622 2022  7  Send reply in 102 ms
  Fri Jul 22 04:23:33.2622 2022  7  RTSP/1.0 200 OK
  Fri Jul 22 04:23:33.2622 2022  7  RTP-Info: url=rtsp://192.168.194.39/stream=109
  Fri Jul 22 04:23:33.2622 2022  7  CSeq: 618
  Fri Jul 22 04:23:33.2622 2022  7  Session: 0272645804
  Fri Jul 22 04:23:33.2622 2022  7  Range: npt=0.000-
  Fri Jul 22 04:23:33.2622 2022  7  <CRLF>
  Fri Jul 22 04:23:43.8344 2022  6  Frontend: 9, PAT: Section Length: 93  TID: 512  Version: 245  secNr: 0 lastSecNr: 0  CRC: 0x0000
  Fri Jul 22 04:23:43.8345 2022  6  Frontend: 9, PAT: Prog NR: 0x000D - 00013  PMT PID: 0521
  Fri Jul 22 04:23:43.8345 2022  6  Frontend: 9, PAT: Prog NR: 0x0028 - 00040  PMT PID: 0100
  Fri Jul 22 04:23:43.8346 2022  6  Frontend: 9, PAT: Prog NR: 0x0037 - 00055  PMT PID: 0550
  Fri Jul 22 04:23:43.8346 2022  6  Frontend: 9, PAT: Prog NR: 0x003C - 00060  PMT PID: 0224
  Fri Jul 22 04:23:43.8347 2022  6  Frontend: 9, PAT: Prog NR: 0x0041 - 00065  PMT PID: 0225
  Fri Jul 22 04:23:43.8347 2022  6  Frontend: 9, PAT: Prog NR: 0x0046 - 00070  PMT PID: 0700
  Fri Jul 22 04:23:43.8347 2022  6  Frontend: 9, PAT: Prog NR: 0x0047 - 00071  PMT PID: 0710
  Fri Jul 22 04:23:43.8348 2022  6  Frontend: 9, PAT: Prog NR: 0x0048 - 00072  PMT PID: 0720
  Fri Jul 22 04:23:43.8348 2022  6  Frontend: 9, PAT: Prog NR: 0x0050 - 00080  PMT PID: 0800
  Fri Jul 22 04:23:43.8348 2022  6  Frontend: 9, PAT: Prog NR: 0x005A - 00090  PMT PID: 0900
  Fri Jul 22 04:23:43.8349 2022  6  Frontend: 9, PAT: Prog NR: 0x0064 - 00100  PMT PID: 1100
  Fri Jul 22 04:23:43.8349 2022  6  Frontend: 9, PAT: Prog NR: 0x0065 - 00101  PMT PID: 1110
  Fri Jul 22 04:23:43.8350 2022  6  Frontend: 9, PAT: Prog NR: 0x0069 - 00105  PMT PID: 1150
  Fri Jul 22 04:23:43.8350 2022  6  Frontend: 9, PAT: Prog NR: 0x006E - 00110  PMT PID: 0500
  Fri Jul 22 04:23:43.8351 2022  6  Frontend: 9, PAT: Prog NR: 0x0078 - 00120  PMT PID: 6131
  Fri Jul 22 04:23:43.8351 2022  6  Frontend: 9, PAT: Prog NR: 0x0079 - 00121  PMT PID: 6141
  Fri Jul 22 04:23:43.8351 2022  6  Frontend: 9, PAT: Prog NR: 0x0092 - 00146  PMT PID: 1456
  Fri Jul 22 04:23:43.8352 2022  6  Frontend: 9, PAT: Prog NR: 0x0097 - 00151  PMT PID: 0980
  Fri Jul 22 04:23:43.8352 2022  6  Frontend: 9, PAT: Prog NR: 0x0274 - 00628  PMT PID: 1528
  Fri Jul 22 04:23:43.8352 2022  6  Frontend: 9, PAT: Prog NR: 0x0275 - 00629  PMT PID: 1536
  Fri Jul 22 04:23:43.8353 2022  6  Frontend: 9, PAT: Prog NR: 0x0276 - 00630  PMT PID: 1541
  Fri Jul 22 04:23:43.8355 2022  6  Frontend: 9, SDT - Section Length: 607  Transport Stream ID: 512  Version: 205  secNr: 0  lastSecNr: 0  NetworkID: 0029  CRC: 0x83BB44F4
  Fri Jul 22 04:23:43.8355 2022  6  Frontend: 9,  serviceID: 0x000D - 00013  EIT: 0xFC  NetworkName: Persidera  ChannelName: R Italia SMI
  Fri Jul 22 04:23:43.8356 2022  6  Frontend: 9,  serviceID: 0x0028 - 00040  EIT: 0xFC  NetworkName: Persidera  ChannelName: SUPERTENNIS
  Fri Jul 22 04:23:43.8357 2022  6  Frontend: 9,  serviceID: 0x0037 - 00055  EIT: 0xFC  NetworkName: Persidera  ChannelName: Super!
  Fri Jul 22 04:23:43.8358 2022  6  Frontend: 9,  serviceID: 0x003C - 00060  EIT: 0xFC  NetworkName: Persidera  ChannelName: RadioItaliaTV
  Fri Jul 22 04:23:43.8358 2022  6  Frontend: 9,  serviceID: 0x0041 - 00065  EIT: 0xFC  NetworkName: Persidera  ChannelName: RadioItaliaTV
[... cut ...]
  Fri Jul 22 04:23:43.8365 2022  6  Frontend: 9,  serviceID: 0x0274 - 00628  EIT: 0xFC  NetworkName: Persidera  ChannelName: TV8 News On Demand
  Fri Jul 22 04:23:43.8366 2022  6  Frontend: 9,  serviceID: 0x0275 - 00629  EIT: 0xFC  NetworkName: Persidera  ChannelName: VH1 On Demand
  Fri Jul 22 04:23:43.8366 2022  6  Frontend: 9,  serviceID: 0x0276 - 00630  EIT: 0xFC  NetworkName: Persidera  ChannelName: VH1 On Demand

 

I think the problem with transponder 12466V is PLS Mode Gold and PLS Code = 0, so the plsc param is not forwarded.

In addition, I still can't tune in transponder with modulation 16APSK.

Greetings.

Barracuda09, 2022-07-22:
Hi @patacalas 

Could you give me a RTSP setup cmd from 16APSK request?

patacalas, 2022-07-22:
Hi @barracuda09
I've captured traffic from net interface and discovered a problem on Enigma2 SatIp client. It requests QPSK when you are trying to tune in a transponder with 16APSK modulation. 

For example, when you tune 12111V DVB-S2 16APSK 31400 2/3 PLS GOLD 262140 StreamID 1 it requests
rtsp://192.168.194.39:554/stream=109?fe=10&src=1&freq=12111&pol=v&msys=dvbs2&sr=31400&fec=23&ro=0.35&mtype=qpsk&plts=off&isi=1&plsc=262140

So, I guess the problem should be also fixed in Enigma2 satip client plugin. I will Google search for satip plugin support.
I think the mtype param in rtsp request should be something like mtype=16apsk

I did some additional tests and discovered some bugs that I hope could be easily fixed.
To start with, if you zap between two channels on the same transponder but with different stream ID, you get tuning error because SatPI still returns channels from first streamID. You need to zap to a different transponder and, after that, you will be able to receive channels from the second streamID.
In addition, there is a similar problem on channel search. If you modify the stream ID to tune different channels from the same transponder you still receive the channels from previous stream ID. I've checked the captured traffic and got the correct rstp commands, but SatPI seems to ignore this change on isi param. You need to tune in a different transponder and back again to channel search to get the desired stream ID channels. 

Thanks for your support.
Greetings.

Barracuda09, 2022-07-22:
Hi @patacalas 

I think I still need to look into the multistream... as this is most likely not correct. The 16apsk should be working but forgot something.

I will have a look tonight.

Thanks for your feedback and testing

patacalas, 2022-07-22:
Hi @Barracuda09 

I've forgotten something in last message.
Finally, I could tune 12466V transponder with PLS Mode Gold and PLS Code = 0. 
Checking captured traffic, the rtsp request with PLS Code = 0 is rtsp://192.168.194.39:554/stream=109?fe=10&src=1&freq=12466&pol=v&msys=dvbs2&sr=41950&fec=56&ro=0.35&mtype=8psk&plts=off&isi=33 
There is no plsc param but it works. I did the trick mentioned on the previous message, I need to tune in a different transponder and back again to the desired transponder and channel search works. 

I think SatPI need to restart the tuner o something like that to apply the correct multistream tuning params.

Thanks

Barracuda09, 2022-07-22:
Hi @patacalas

Could you please try this if you have the time:
[1.6.2.32~g643fc5b_Enigma_Multistream.zip](https://github.com/Barracuda09/SATPI/files/9169890/1.6.2.32.g643fc5b_Enigma_Multistream.zip)
 
Many thanks!

patacalas, 2022-07-24:
Hi @Barracuda09 
I've just tried the latest version and I can finally tune in to channels with APSK modulation.

[image]
The request is still using mtype=qpsk param, but it works.

I've also tested the issue while zapping between channels with different stream ID on the same transponder, and I still need to tune in to a different transponder to work.

Thank you very much for your excellent work.

Barracuda09, 2022-07-25:
So that means nothing of the tuning parameters (freq. pol ect) change only the stream ID?

Barracuda09, 2022-07-25:
> The request is still using mtype=qpsk param, but it works.
Were is this showing, in SatPI or what do you mean precise? Is this still not good in SatPI?

patacalas, 2022-07-25:
> So that means nothing of the tuning parameters (freq. pol ect) change only the stream ID?

Multistream transponder example. Same frequency, polarization, FEC,... but different stream ID (check red arrows) to tune in different channels.
[image]
The problem when zapping (or doing channel search) is about the stream ID. For example, if the previous channel is on stream 1 (Rete 4, Canale 5, Italia 1,...) and you are zapping to a channel on stream 2 (R101 TV, La 5, Iris,...) you will get tuning error because SatPI still returns channels from stream 1 and not channels from stream 2 (as seen on SatPI log). You need to tune in to another transponder and then zap to a channel from stream 2 to get successful tuning.

patacalas, 2022-07-25:
> > The request is still using mtype=qpsk param, but it works.
> 
> Were is this showing, in SatPI or what do you mean precise? Is this still not good in SatPI?

Captured traffic from client receiver shows incorrect rtsp requests with mtype=qpsk param, but SatPI is working fine and tuning 16ASPK transponders even with incorrect modulation request. I think Enigma2 Sat>IP client should be updated to handle new modulation params.

Barracuda09, 2022-07-25:
Hi @patacalas 

Could you please try this version:
[1.6.2.32~g643fc5b_Enigma_Multistream.zip](https://github.com/Barracuda09/SATPI/files/9183347/1.6.2.32.g643fc5b_Enigma_Multistream.zip)

patacalas, 2022-07-25:
Hi @Barracuda09 
I've tested the last version, it's working fine on previous scanned channels (no problem with stream ID param), but now I can't do channel search successfully.

The stream id param is working as expected but I think that there is a problem with delay while searching for new channels. The client receiver says "no channel founds" and, a little bit later, services name and ids are displayed on SatPI log. Maybe my client is too laggy and limited by processor specs as SatPI is working under VPN encryption. 

Can I set a higher timeout in Enigma2 configuration when searching for channels? Probably the last SatPI version works fine for Sat>IP in local network.

Barracuda09, 2022-07-26:
Again thanks for testing.

Did scaning work in previous versions?

I know scaning with my Vu+ Uno DVB-S2 is problematic, because the 'low' bitrate during scaning takes a long time. So the SAT>IP client time-out before receiving the data.

patacalas, 2022-07-26:
> 
> Did scaning work in previous versions?
> 

Hi @Barracuda09. Channel search worked almost fine on previous versions. About 90% success in normal transponders and 40-50% on multistream transponders (sometimes I need to repeat scanning again and again to get successfully channel search).
Last version is about 70% success on normal transponders and it's impossible to scan multistream transponders. 
I've also detected worse channel zapping. I get a lot of "tuning errors" on both normal and multistream transponders and channel surfing is also slowly.

Tomorrow I will compare tuning times between SatPI versions and post results here.

Barracuda09, 2022-07-27:
It could be because it is an debug version I added here.

patacalas, 2022-07-28:
Hi @Barracuda09 
I've done new tests. I've tried again all the version you shared here to compare tuning times and procedures. All versions take about 16-17s between RTSP request and services data response from SatPI.
As I told before, doing channel search on multistream transponders is impossible with last version. So, I compared the last two versions to try to find differences.

The signal finder works as follows on my receiver.

[image]

A couple of seconds after entering the tuning parameters TSID and ONID are displayed and *service list* button appears. When tuning multistream transponders via SatPI it takes about 16s to appears (as expected because of the timings showed on logfile). 
I don't need to wait until I press the green button to get a successfully channel search, except for SatPI searches. If I wait until TSID, ONID and *service list* button appears the channel search works fine with SatPI.

[image]

At least, that's what happens in all versions except the last one.
I think that's because the channel search takes too long (about 16 seconds) and when I wait on the previous screen (where services lists are displayed) the channel search procedure retrieves those channels from a kind of cache (it only takes about 3 or 4 seconds to complete, even before of SatPI response). 
Next, I'm going to paste the log from two last versions and comment some timestamps.

**Important timestamps from previous version**

Thu Jul 28 10:01:04.8044 2022 > I typed tuning params 
Thu Jul 28 10:01:19.7560 2022 > Services list were available on signal finder
Thu Jul 28 10:01:28.1259 2022 > I pressed the channel search button
--- about 4-5s later > Channel search procedure completed
Thu Jul 28 10:01:44.6364 2022 > SatPI response with channel list

**Important timestamps from last version**

Thu Jul 28 09:55:49.5685 2022 > I typed tuning params 
Thu Jul 28 09:56:05.2230 2022 > Services list were available on signal finder
Thu Jul 28 09:56:10.8931 2022 > I pressed the channel search button
--- about 10-12s later > No channels found error
Thu Jul 28 09:56:11.3728 2022 > SatPI ends without channel list response

### **LOG from previous version--------------**

 Thu Jul 28 10:01:04.8044 2022  7  RTSP Stream data from client None with IP 192.168.194.185 on Port 38984: PLAY rtsp://192.168.194.39:554/stream=109?fe=10&src=1&freq=12341&pol=v&msys=dvbs2&sr=31400&fec=23&ro=0.35&mtype=8psk&plts=off&isi=33&plsc=262140 RTSP/1.0
  Thu Jul 28 10:01:04.8044 2022  7  CSeq: 7
  Thu Jul 28 10:01:04.8044 2022  7  Session: 0261938357
  Thu Jul 28 10:01:04.8044 2022  7  <CRLF>
  Thu Jul 28 10:01:04.8045 2022  6  Found FrondtendID: 9 (fe=10)  StreamID: 109  SessionID: 0261938357
  Thu Jul 28 10:01:04.8046 2022  6  Frontend: 9, StreamClient[0] with SessionID 0261938357 for dvbs2
  Thu Jul 28 10:01:04.8047 2022  6  Frontend: 9, Parsing transport parameters...
  Thu Jul 28 10:01:04.8047 2022  6  Frontend: 9, New frequency requested, clearing old channel data...
  Thu Jul 28 10:01:04.8049 2022  7  Frontend: 9, Parsing transport parameters (Finished)
  Thu Jul 28 10:01:04.8049 2022  6  Frontend: 9, Pause RTCP/UDP stream to 192.168.194.185:46832
  Thu Jul 28 10:01:04.9051 2022  6  Frontend: 9, Pause RTP/UDP stream to 192.168.194.185:46831 (Streamed 0.0038 MBytes)
  Thu Jul 28 10:01:04.9052 2022  6  Frontend: 9, Clearing OSCam filters and Keys...
  Thu Jul 28 10:01:04.9053 2022  6  Frontend: 9, Updating frontend...
  Thu Jul 28 10:01:04.9053 2022  6  Frontend: 9, Closing all active PID filters...
  Thu Jul 28 10:01:04.9054 2022  7  Frontend: 9, Remove filter PID: 0000 - Packet Count: 000000000:000000
  Thu Jul 28 10:01:04.9055 2022  7  Frontend: 9, Remove filter PID: 0001 - Packet Count: 000000000:000000
  Thu Jul 28 10:01:04.9056 2022  7  Frontend: 9, Remove filter PID: 0016 - Packet Count: 000000000:000000
  Thu Jul 28 10:01:04.9056 2022  7  Frontend: 9, Remove filter PID: 0017 - Packet Count: 000000000:000000
  Thu Jul 28 10:01:04.9063 2022  7  Frontend: 9, Remove filter PID: 0018 - Packet Count: 000000000:000000
  Thu Jul 28 10:01:04.9065 2022  6  Frontend: 9, Closing /dev/dvb/adapter0/demux9 fd: 18
  Thu Jul 28 10:01:04.9082 2022  6  Frontend: 9, Closing /dev/dvb/adapter0/frontend9 fd: 15
  Thu Jul 28 10:01:04.9588 2022  6  Frontend: 9, Opened /dev/dvb/adapter0/frontend9 for Read/Write with fd: 15 (0 ms)
  Thu Jul 28 10:01:04.9589 2022  6  Frontend: 9, Start tuning process for DVB-S(2)...
  Thu Jul 28 10:01:04.9589 2022  6  Frontend: 9, Opened /dev/dvb/adapter0/frontend9 for Writing DiSEqC command with fd: 15
  Thu Jul 28 10:01:04.9590 2022  6  Frontend: 9, Sending LNB: Mini-Switch Src: 0
  Thu Jul 28 10:01:05.0735 2022  7  Frontend: 9, Set Properties: Frequency 1741000
  Thu Jul 28 10:01:05.0736 2022  6  Frontend: 9, Tuned, waiting on lock...
  Thu Jul 28 10:01:05.2335 2022  6  Frontend: 9, Tuned and locked (FE status 0x10)
  Thu Jul 28 10:01:05.2336 2022  6  Frontend: 9, Updating PID filters...
  Thu Jul 28 10:01:05.2337 2022  6  Frontend: 9, Updating frontend (Finished in 328 ms)
  Thu Jul 28 10:01:05.2338 2022  6  Frontend: 9, Restart RTCP/UDP stream to 192.168.194.185:46832
  Thu Jul 28 10:01:05.2338 2022  6  Frontend: 9, Restart RTP/UDP stream to 192.168.194.185:46831
  Thu Jul 28 10:01:05.2338 2022  7  Send reply in 429 ms
  Thu Jul 28 10:01:05.2338 2022  7  RTSP/1.0 200 OK
  Thu Jul 28 10:01:05.2338 2022  7  RTP-Info: url=rtsp://192.168.194.39/stream=109
[... 663 lines cut ...]
[... cut ...]
  Thu Jul 28 09:56:11.3728 2022  6  Frontend: 9, Restart RTCP/UDP stream to 192.168.194.185:46832
  Thu Jul 28 09:56:11.3728 2022  6  Frontend: 9, Restart RTP/UDP stream to 192.168.194.185:46831
  Thu Jul 28 09:56:11.3728 2022  7  Send reply in 479 ms
  Thu Jul 28 09:56:11.3728 2022  7  RTSP/1.0 200 OK
  Thu Jul 28 09:56:11.3728 2022  7  RTP-Info: url=rtsp://192.168.194.39/stream=109
  Thu Jul 28 09:56:11.3728 2022  7  CSeq: 67
  Thu Jul 28 09:56:11.3728 2022  7  Session: 0273478590
  Thu Jul 28 09:56:11.3728 2022  7  Range: npt=0.000-
  Thu Jul 28 09:56:11.3728 2022  7  <CRLF>

Barracuda09, 2022-07-28:
Hi

Does it make any difference to lower the DVR buffer size in SatPI (In frontend tab)?

patacalas, 2022-07-28:
I've tried different values of DVR buffer, but changing that value doesn't make any difference.

Barracuda09, 2022-07-28:
What if you tune to a freq without **isi** and **plsc** does that work faster?

patacalas, 2022-07-28:
Yes, normal transponders work faster.

Barracuda09, 2022-07-28:
Ok, so that makes me think there is no logic error in reading the data.
But that it is probably, for some reason, taking this (long) time to retrieve/collect the data from the tuner/driver.

I have to try to check this myself. But my setup is only Astra 19.2 and 23.5 at the moment.

Barracuda09, 2022-07-30:
Hi @patacalas 

I have committed the changes. This is a different version then the files attached here.

patacalas, 2022-07-31:
Hi @Barracuda09 

I was wondering if you can share the latest version compiled for ARM here. It's hard for me to compile the code on my Linux receiver.

I would be happy to test the latest version with 9ºE multistream channels.

Thanks in advance

Barracuda09, 2022-07-31:
Hi @patacalas 

Sure, here is the version:
[satpi_arm_1.6.2.36~g7814e4a_Enigma.zip](https://github.com/Barracuda09/SATPI/files/9229386/satpi_arm_1.6.2.36.g7814e4a_Enigma.zip)

Thanks for testing!