Barracuda09/SATPI #161: Astra cesbo agent

daveraver, 2022-09-03:
Hi all, I am trying to up streams from vuplus uno 4k to astra cesbo. Is it posible? I am unable to config it in astra options to get the stream working. Is not a softcam problem, I dont think so. I get the TP scanned and get all channels data, select 1 channel and give to it a softcam line connection just for this channel, line connects to oscam but it dont ask for nothing to oscam because the stream doesnt start.

I've tested it on tvheadend and it runs with no problem, initially, I havent checked stability yet.

I wonder if satip give some better working than m3u list, and that's why I am testing it too on astra cesbo... Maybe using dengo un uso deeco sat tuner is not the better idea to get the best results, but I havent got anything else to test..

If I could give you some info about it, now I get this message on astra cesbo that it could indicate that I am doing wrong config...
I've copied the stream link from SATPI, it comes from what I put in astra cesbo, almost all, except PID that I added the right ones manually on stream link of SATPI but I got this bad request message:

[sat>ip http://192.168.1.205:8875/?freq=10729&sr=22000&msys=dvbs2&mtype=8psk&fec=23&pol=v&src=2&pids=0,1,16,17,18,1027,1028,1029,1030,1031,1033,1039,1043/2] request failed [request failed. code:503]

I have in use a LNB switch diseqc 1.0 with 2 ports, port 2 is the astra 19.2E sat one.

I am sorry if this bad config is because of my noobs status...

Thanks in advance!

Barracuda09, 2022-09-03:
> [sat>ip http://192.168.1.205:8875/?freq=10729&sr=22000&msys=dvbs2&mtype=8psk&fec=23&pol=v&src=2&pids=0,1,16,17,18,1027,1028,1029,1030,1031,1033,1039,1043/2] request failed [request failed. code:503]
 
In this request at the end there is a `/2` i am not sure were this comes from. 
Could you please give the log from SatPI?

Thanks for testing!

daveraver, 2022-09-03:
Hello, thanks for you attention... here is attached my SATPI log. My lnb is double sat, Hotbird-Astra19.2, and a switch diseqc1.0 with Astra19.2 on port #2. The default PMT of stream link on frontend#1 in SATPI appeared this pmt : 0,1,16,17,18. I added this other pmt of TP 10729: 1027,1028,1029,1030,1031,1033,1039,1043. Without these last PMT the astra cesbo TP scan only found the channel (SID) with no video and audio PID's. The stream doesnt start, when I test the stream it doesnt show more than 21kb/s so the test shows no stream is running.[
[log_satpi.txt](https://github.com/Barracuda09/SATPI/files/9482872/log_satpi.txt)
](url)
[log_satpi.txt](https://github.com/Barracuda09/SATPI/files/9482873/log_satpi.txt)

daveraver, 2022-09-03:
I see that deleting 1027,....,1043 PMT the log seems better ...
[log2_satpi.txt](https://github.com/Barracuda09/SATPI/files/9482879/log2_satpi.txt)

Barracuda09, 2022-09-04:
How do you use Oscam, via SatPI or some other method.

 I still see some strange requests here (Were is this request coming from?)
```
/?freq=10729&sr=22000&msys=dvbs2&mtype=8psk&fec=23&pol=v&src=2&pids=0,1,16,17,18/?src=2&fe=2&msys=dvbs2&freq=10729&pol=v&sr=22000&mtype=8psk&fec=23&pids=1027,1028,1029,1030,1031,1033,1039,1043
```

Probably it will only parse this first part, up until the second /?

daveraver, 2022-09-05:
hi Marc, thanks for your time.

I use a Nline to on astra cesbo softcam to just one channel by Nline. I am testing several ways to get streams, and I dont get all stream video and audio with only 0,1,16,17,18 pid and then I flag the channel pid on tp 10729 to get all the channel info video and audio pids. I am very sure that satpi on a vu+ uno 4k to send streams to astra cesbo is not the best way. Stb is not powerful enough to get all I wanted to stream, cpu is very limited on uno 4k. using a deco is better to send stream by engima2 by using astra cesbo softcam, it is better dont overload stb cpu. but I would like to test anyway satpi, on stb or linux PC, then I will se how useful it could be, not good on stb so...

Barracuda09, 2022-09-05:
You could try to use VLC with this example from Astra 19.2 (Tagesschau 24)
```
http://192.168.1.205:8875/?freq=10743&sr=22000&msys=dvbs&mtype=qpsk&fec=56&pol=h&src=2&pids=0,1,16,17,18,100,100,102,104
```
Just open it in _Network Stream_ under _Media_

Saentist, 2022-09-05:
How is Astra configured to use remote tuner?

Select **New Adapter**
[image]

daveraver, 2022-09-05:
> You could try to use VLC with this example from Astra 19.2 (Tagesschau 24)
> 
> ```
> http://192.168.1.205:8875/?freq=10743&sr=22000&msys=dvbs&mtype=qpsk&fec=56&pol=h&src=2&pids=0,1,16,17,18,100,100,102,104
> ```
> 
> Just open it in _Network Stream_ under _Media_

I could heard german audio, no video ..  What is wrong on my adapter of tp 10729 then?

daveraver, 2022-09-05:
> to

[image]

daveraver, 2022-09-05:
> How is Astra configured to use remote tuner?
> 
> Select **New Adapter** [image]

[image]
this one, it is one of tested, I added pmt pid too, to get all channel video and audio pids

regards,

Saentist, 2022-09-06:
> > How is Astra configured to use remote tuner?
> > 
> > Select **New Adapter** [image]
> 
> [image]
> this one, it is one of tested, I added pmt pid too, to get all channel video and audio pids
> 
> regards,

Wrong
You need to set up and port only
Scan is next step on bottom
Else you need to add is as stream

daveraver, 2022-09-06:
> > > How is Astra configured to use remote tuner?
> > > Select **New Adapter** [image]
> > 
> > 
> > [image]
> > this one, it is one of tested, I added pmt pid too, to get all channel video and audio pids
> > regards,
> 
> Wrong You need to set ip and port only Scan is next step on bottom Else you need to add is as stream

I tryied with only ip:port and satpi hangs on uno 4k. I will try to scan as stream, thank you

Barracuda09, 2022-09-06:
> How do you use Oscam, via SatPI or some other method.
>
> I still see some strange requests here (Were is this request coming from?)
>
> ``` [...]

Thanks @Saentist 

Then this makes sense now

Saentist, 2022-09-06:
> Thanks @Saentist
> 
> Then this makes sense now

There must be a some logic explanation in WIKI what is SAT>IP Server and Client
or more exactly, user need to have basic knowledge how work "network tuner protocol", 
before anything else.

From where come url
'''
/?freq=10729&sr=22000&msys=dvbs2&mtype=8psk&fec=23&pol=v&src=2&pids=0,1,16,17,18/?src=2&fe=2&msys=dvbs2&freq=10729&pol=v&sr=22000&mtype=8psk&fec=23&pids=1027,1028,1029,1030,1031,1033,1039,1043
'''
how is constructed 
how work **SETUP, PLAY, OPTIONS, DESCRIBE, TEARDOWN etc** in SAT>IP protocol
how and why is used  HTTP, IGMP and RTP in sat>ip

p.s. Astra dont work with RTP SAT>IP servers without HTTP support yet ;)

daveraver, 2022-09-06:
then it is not possible to comunicate satpi with astra... if I put just ip:port satpi hangs up... tvheadend works but I wanted try astracesbo, it is more stable...

Saentist, 2022-09-08:
> then it is not possible to comunicate satpi with astra?? if I put just ip:port satpi hangs up... tvheadend works but I wanted to try astracesbo, it is more stable...

This is a moment where you must attach logs from SATPI.
so developer to know with part of code make problem and crash

daveraver, 2022-09-08:
ok thank you, I will send them, along next weeks,
regards,

daveraver, 2022-09-10:
I start astra cesbo and put the satpi device http://192.168.1.205:8875 and the box uno 4k with satpi hangs up... I've had to do a reboot. Log attached, I dont see any error... I just can snam de TP putting the satpi stream url on astra cesbo device... not only ip:port of satpi.
[log_conected to astra cesbo_stb uno 4k hangs up.odt](https://github.com/Barracuda09/SATPI/files/9540070/log_conected.to.astra.cesbo_stb.uno.4k.hangs.up.odt)

Barracuda09, 2022-09-12:
Hi @daveraver 

Thanks for you feedback!.

- What kind of Diseqc switch do you use: Committed or Uncommitted.
  In the log is probably Uncommitted selected
  > [log_satpi.txt](https://github.com/Barracuda09/SATPI/files/9482873/log_satpi.txt)
  ```
   Sat Sep  3 15:56:21.9780 2022  6  Frontend: 1, Sending DiSEqC: [0xE0] [0x10] [0x39] [0xF4] - DiSEqC Src: 1
  ```
  The code for _Uncommitted Switch_ is currently not working correctly.

- The Hang ups you describe is not showing in the log. Could you describe what hangs is the box not responding?
  The Vu+ Uno 4K is powerfull enough to run SatPI (I use two of them without problems DVB-C and S2)

daveraver, 2022-09-12:
Thanky you for your time and for your attention. I think the box hangs I am unable to see any channel when satpi app is blocked, then maybe the tuner get some kind of blockness.

I've decided to use streams as engima2 serve them with no management by satpi.

Regards,
David

closed 2024-04-10 by Barracuda09