Barracuda09/SATPI #162: AlpinLinux build problem

Saentist, 2022-10-02:
I'm try to build SATPI on AlpinLinux 32bit

 # make debug LIBDVBCSA=yes -d
```make
...
  Pruning file 'src/TransportParamVector.h'.
  Pruning file 'src/FileDescriptor.h'.
  Pruning file 'src/RtspServer.h'.
  Pruning file 'src/upnp/ssdp/Server.h'.
  Pruning file 'src/StreamInterface.h'.
  Pruning file 'src/InterfaceAttr.h'.
  Pruning file 'src/Satpi.h'.
  Pruning file 'src/Stream.h'.
  Pruning file 'src/StringConverter.h'.
  Pruning file 'src/Properties.h'.
 Finished prerequisites of target file 'satpi'.
Must remake target 'satpi'.
g++ obj/Version.o obj/InterfaceAttr.o obj/HeaderVector.o obj/HttpServer.o obj/HttpcServer.o obj/Log.o obj/Properties.o obj/RtspServer.o obj/main.o obj/Satpi.o obj/Stream.o obj/StreamClient.o obj/StreamManager.o obj/StringConverter.o obj/TransportParamVector.o obj/Utils.o obj/base/M3UParser.o obj/base/Thread.o obj/base/ThreadBase.o obj/base/TimeCounter.o obj/base/XMLSaveSupport.o obj/base/XMLSupport.o obj/input/DeviceData.o obj/input/Transformation.o obj/input/dvb/Frontend.o obj/input/dvb/FrontendData.o obj/input/dvb/delivery/DiSEqc.o obj/input/dvb/delivery/DiSEqcEN50494.o obj/input/dvb/delivery/DiSEqcEN50607.o obj/input/dvb/delivery/DiSEqcLnb.o obj/input/dvb/delivery/DiSEqcSwitch.o obj/input/dvb/delivery/DVBC.o obj/input/dvb/delivery/DVBS.o obj/input/dvb/delivery/DVBT.o obj/input/dvb/delivery/Lnb.o obj/input/file/TSReader.o obj/input/file/TSReaderData.o obj/input/childpipe/TSReader.o obj/input/childpipe/TSReaderData.o obj/input/stream/Streamer.o obj/input/stream/StreamerData.o obj/mpegts/Filter.o obj/mpegts/PacketBuffer.o obj/mpegts/PAT.o obj/mpegts/PCR.o obj/mpegts/PidTable.o obj/mpegts/PMT.o obj/mpegts/SDT.o obj/mpegts/TableData.o obj/output/StreamThreadBase.o obj/output/StreamThreadHttp.o obj/output/StreamThreadRtcpBase.o obj/output/StreamThreadRtcp.o obj/output/StreamThreadRtcpTcp.o obj/output/StreamThreadRtp.o obj/output/StreamThreadRtpTcp.o obj/output/StreamThreadTSWriter.o obj/socket/HttpcSocket.o obj/socket/TcpSocket.o obj/socket/SocketAttr.o obj/socket/UdpSocket.o obj/upnp/ssdp/Server.o obj/decrypt/dvbapi/Client.o obj/decrypt/dvbapi/ClientProperties.o obj/decrypt/dvbapi/Keys.o obj/input/dvb/Frontend_DecryptInterface.o -o satpi  -pthread -lrt -rdynamic -ldvbcsa
Putting child 0xb7e81c40 (satpi) PID 6084 on the chain.
Live child 0xb7e81c40 (satpi) PID 6084
/usr/lib/gcc/i586-alpine-linux-musl/11.2.1/../../../../i586-alpine-linux-musl/bin/ld: obj/Utils.o: in function `Utils::createBackTrace(char const*)':
/opt/SATPI/src/Utils.cpp:35: undefined reference to `backtrace'
/usr/lib/gcc/i586-alpine-linux-musl/11.2.1/../../../../i586-alpine-linux-musl/bin/ld: /opt/SATPI/src/Utils.cpp:42: undefined reference to `backtrace_symbols_fd'
/usr/lib/gcc/i586-alpine-linux-musl/11.2.1/../../../../i586-alpine-linux-musl/bin/ld: /opt/SATPI/src/Utils.cpp:45: undefined reference to `backtrace_symbols_fd'
collect2: error: ld returned 1 exit status
Reaping losing child 0xb7e81c40 PID 6084
make[1]: *** [Makefile:198: satpi] Error 1
Removing child 0xb7e81c40 PID 6084 from chain.
make[1]: Leaving directory '/opt/SATPI'
Reaping losing child 0xb7f2a6f0 PID 5704
make: *** [Makefile:209: debug] Error 2
Removing child 0xb7f2a6f0 PID 5704 from chain.

```
What dependency do I have to search, or is something other?

Barracuda09, 2022-10-02:
Hi @Saentist 

Thanks for giving this a try!

You could change the **createBackTrac**e function(in **Utils.cpp**) for now like:
```
void Utils::createBackTrace(const char *file) {
/*
	// DO NOT alloc memory on heap!!
	void *array[25];
	const size_t size = backtrace(array, 25);

	// Log all the frames to Backtrace File
	char path[256];
	snprintf(path, sizeof(path), "/tmp/%s.bt", file);
	int backtraceFile = ::open(path, O_CREAT|O_WRONLY|O_TRUNC, 0664);
	if (backtraceFile > 0) {
		backtrace_symbols_fd(array, size, backtraceFile);
		::close(backtraceFile);
	} else {
		backtrace_symbols_fd(array, size, STDOUT_FILENO);
	}
*/
}
```

Barracuda09, 2022-10-02:
Hi @Saentist 

Here is a similar problem, with maybe a solution?:
https://github.com/alpinelinux/docker-alpine/issues/26

Saentist, 2022-10-02:
there is new error
```make
Putting child 0xb7e5fc80 (obj/Utils.o) PID 8514 on the chain.
Live child 0xb7e5fc80 (obj/Utils.o) PID 8514
Reaping winning child 0xb7e5fc80 PID 8514
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Utils.cpp -o obj/Utils.o
Live child 0xb7e5fc80 (obj/Utils.o) PID 8515
src/Utils.cpp:49:1: error: expected declaration before '}' token
   49 | }
      | ^
Reaping losing child 0xb7e5fc80 PID 8515
make[1]: *** [Makefile:205: obj/Utils.o] Error 1
Removing child 0xb7e5fc80 PID 8515 from chain.
make[1]: Leaving directory '/opt/SATPI'
Reaping losing child 0xb7eca6f0 PID 8479
make: *** [Makefile:209: debug] Error 2
Removing child 0xb7eca6f0 PID 8479 from chain.

```

Barracuda09, 2022-10-02:
Did you copy-paste the complete code and replaced it in Utils.cpp

Saentist, 2022-10-02:
just last lines need to be switched
from
```cpp
	}
*/
}
```
to
```cpp
	}
}
*/
```
I guess

Barracuda09, 2022-10-02:
No the last ***/** is to low Just 1 line up before **}**

Saentist, 2022-10-02:
so is correctly copyed then

Barracuda09, 2022-10-02:
Yes, the function **Utils::createBackTrace** should be empty, that is what **/*** and ***/** should do.

Saentist, 2022-10-02:
it comment code section ;)
I see that Alpine use some modified compilation stuff, 
but what else do I can report to help?

Barracuda09, 2022-10-02:
Does is still not build correctly?

Saentist, 2022-10-02:
Build speed is not so fast on 450Mhz single core with 512Mb ram ;)
switched lines still compiling

as you propose crash very fast with upper error.

Barracuda09, 2022-10-02:
Just post when there is a compile error the output as before

Saentist, 2022-10-02:
Is there a way to use **[ccache](https://ccache.dev)** during build?

```make
make debug LIBDVBCSA=yes
make "BUILD=debug"
make[1]: Entering directory '/opt/SATPI'
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Version.cpp -o obj/Version.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/InterfaceAttr.cpp -o obj/InterfaceAttr.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HeaderVector.cpp -o obj/HeaderVector.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HttpServer.cpp -o obj/HttpServer.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HttpcServer.cpp -o obj/HttpcServer.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Log.cpp -o obj/Log.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Properties.cpp -o obj/Properties.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/RtspServer.cpp -o obj/RtspServer.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/main.cpp -o obj/main.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Satpi.cpp -o obj/Satpi.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Stream.cpp -o obj/Stream.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StreamClient.cpp -o obj/StreamClient.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StreamManager.cpp -o obj/StreamManager.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StringConverter.cpp -o obj/StringConverter.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/TransportParamVector.cpp -o obj/TransportParamVector.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O0 -g3 -DDEBUG -fstack-protector-all -Wswitch-default -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Utils.cpp -o obj/Utils.o
src/Utils.cpp: In static member function 'static void Utils::createBackTrace(const char*)':
src/Utils.cpp:53:30: error: qualified-id in declaration before '(' token
   53 | void Utils::annotateBackTrace(const char *app, const char *file) {
      |                              ^
src/Utils.cpp:78:2: error: expected '}' at end of input
   78 | }
      |  ^
src/Utils.cpp:33:47: note: to match this '{'
   33 | void Utils::createBackTrace(const char *file) {
      |                                               ^
src/Utils.cpp:33:41: warning: unused parameter 'file' [-Wunused-parameter]
   33 | void Utils::createBackTrace(const char *file) {
      |                             ~~~~~~~~~~~~^~~~
make[1]: *** [Makefile:205: obj/Utils.o] Error 1
make[1]: Leaving directory '/opt/SATPI'
make: *** [Makefile:209: debug] Error 2

```

Saentist, 2022-10-02:
without DEBUG
```make
make CC='ccache gcc' LIBDVBCSA=yes
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Version.cpp -o obj/Version.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/InterfaceAttr.cpp -o obj/InterfaceAttr.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HeaderVector.cpp -o obj/HeaderVector.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HttpServer.cpp -o obj/HttpServer.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HttpcServer.cpp -o obj/HttpcServer.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Log.cpp -o obj/Log.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Properties.cpp -o obj/Properties.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/RtspServer.cpp -o obj/RtspServer.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/main.cpp -o obj/main.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Satpi.cpp -o obj/Satpi.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Stream.cpp -o obj/Stream.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StreamClient.cpp -o obj/StreamClient.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StreamManager.cpp -o obj/StreamManager.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StringConverter.cpp -o obj/StringConverter.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/TransportParamVector.cpp -o obj/TransportParamVector.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Utils.cpp -o obj/Utils.o
src/Utils.cpp: In static member function 'static void Utils::createBackTrace(const char*)':
src/Utils.cpp:53:30: error: qualified-id in declaration before '(' token
   53 | void Utils::annotateBackTrace(const char *app, const char *file) {
      |                              ^
src/Utils.cpp:78:2: error: expected '}' at end of input
   78 | }
      |  ^
src/Utils.cpp:33:47: note: to match this '{'
   33 | void Utils::createBackTrace(const char *file) {
      |                                               ^
src/Utils.cpp:33:41: warning: unused parameter 'file' [-Wunused-parameter]
   33 | void Utils::createBackTrace(const char *file) {
      |                             ~~~~~~~~~~~~^~~~
make: *** [Makefile:205: obj/Utils.o] Error 1

```

Barracuda09, 2022-10-02:
Here is the Utils.cpp as it should be. Just replace it with this code:
```
/* Utils.cpp

   Copyright (C) 2014 - 2021 Marc Postema (mpostema09 -at- gmail.com)

   This program is free software; you can redistribute it and/or
   modify it under the terms of the GNU General Public License
   as published by the Free Software Foundation; either version 2
   of the License, or (at your option) any later version.

   This program is distributed in the hope that it will be useful,
   but WITHOUT ANY WARRANTY; without even the implied warranty of
   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
   GNU General Public License for more details.

   You should have received a copy of the GNU General Public License
   along with this program; if not, write to the Free Software
   Foundation, Inc., 59 Temple Place - Suite 330, Boston, MA 02111-1307, USA.
   Or, point your browser to http://www.gnu.org/copyleft/gpl.html
 */
#include <Utils.h>
#include <base/ChildPIPEReader.h>

#include <iostream>
#include <fstream>
#include <algorithm>

#include <unistd.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <execinfo.h>

void Utils::createBackTrace(const char */*file*/) {
/*
	// DO NOT alloc memory on heap!!
	void *array[25];
	const size_t size = backtrace(array, 25);

	// Log all the frames to Backtrace File
	char path[256];
	snprintf(path, sizeof(path), "/tmp/%s.bt", file);
	int backtraceFile = ::open(path, O_CREAT|O_WRONLY|O_TRUNC, 0664);
	if (backtraceFile > 0) {
		backtrace_symbols_fd(array, size, backtraceFile);
		::close(backtraceFile);
	} else {
		backtrace_symbols_fd(array, size, STDOUT_FILENO);
	}
*/
}

void Utils::annotateBackTrace(const char *app, const char *file) {
	std::ifstream bt(file, std::ios::in);
	int i = 0;
	for (std::string line; std::getline(bt, line); ++i) {
		const auto begin = line.find('[');
		const auto end = line.find(']');
		if (begin != std::string::npos && end != std::string::npos) {
			const std::string addr = line.substr(begin + 1, end - begin - 1);

			char addr2line[256];
			snprintf(addr2line, sizeof(addr2line),"addr2line %s -e %s", addr.c_str(), app);

			base::ChildPIPEReader exec;
			exec.open(addr2line);
			std::string code;
			char buffer[256];
			std::size_t s;
			while ((s = exec.read(reinterpret_cast<unsigned char *>(buffer), 255)) > 0) {
				code.append(buffer, s);
				code.erase(std::find(code.begin(), code.end(), '\n'));
			}

			std::cout << "[bt] #" << i << " " << code << " -- " << line << std::endl;
		}
	}
}
```

Saentist, 2022-10-02:
Git clone latest source
added commenting to code from lines 33-46
and SUCCESS with expected error for commented code
```make
make CC='ccache gcc' LIBDVBCSA=yes |tee build4.txt
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Version.cpp -o obj/Version.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/InterfaceAttr.cpp -o obj/InterfaceAttr.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HeaderVector.cpp -o obj/HeaderVector.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HttpServer.cpp -o obj/HttpServer.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HttpcServer.cpp -o obj/HttpcServer.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Log.cpp -o obj/Log.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Properties.cpp -o obj/Properties.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/RtspServer.cpp -o obj/RtspServer.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/main.cpp -o obj/main.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Satpi.cpp -o obj/Satpi.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Stream.cpp -o obj/Stream.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StreamClient.cpp -o obj/StreamClient.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StreamManager.cpp -o obj/StreamManager.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StringConverter.cpp -o obj/StringConverter.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/TransportParamVector.cpp -o obj/TransportParamVector.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Utils.cpp -o obj/Utils.o
src/Utils.cpp: In static member function 'static void Utils::createBackTrace(const char*)':
src/Utils.cpp:33:41: warning: unused parameter 'file' [-Wunused-parameter]
   33 | void Utils::createBackTrace(const char *file) {
      |                             ~~~~~~~~~~~~^~~~
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/M3UParser.cpp -o obj/base/M3UParser.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/Thread.cpp -o obj/base/Thread.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/ThreadBase.cpp -o obj/base/ThreadBase.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/TimeCounter.cpp -o obj/base/TimeCounter.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/XMLSaveSupport.cpp -o obj/base/XMLSaveSupport.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/XMLSupport.cpp -o obj/base/XMLSupport.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/input/DeviceData.cpp -o obj/input/DeviceData.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/input/Transformation.cpp -o obj/input/Transformation.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/input/dvb/Frontend.cpp -o obj/input/dvb/Frontend.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/input/dvb/FrontendData.cpp -o obj/input/dvb/FrontendData.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/input/dvb/delivery/DiSEqc.cpp -o obj/input/dvb/delivery/DiSEqc.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/input/dvb/delivery/DiSEqcEN50494.cpp -o obj/input/dvb/delivery/DiSEqcEN50494.o
g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/input/dvb/delivery/DiSEqcEN50607.cpp -o obj/input/dvb/delivery/DiSEqcEN50607.o
[... cut ...]
  Sun Oct  2 13:21:58.8238 2022  6  Root Device Client 192.168.1.100 : tries to discover the network, sending reply back
  Sun Oct  2 13:21:59.1118 2022  6  HTTP Connection from 192.168.1.100 Port 56173 with fd: 12
  Sun Oct  2 13:21:59.1416 2022  6  HTTP Connection from 192.168.1.100 Port 56174 with fd: 13
  Sun Oct  2 13:21:59.5005 2022  6  HTTP Connection from 192.168.1.100 Port 56176 with fd: 14
  Sun Oct  2 13:21:59.5011 2022  6  HTTP Connection from 192.168.1.100 Port 56177 with fd: 15
  Sun Oct  2 13:21:59.5031 2022  6  HTTP Connection from 192.168.1.100 Port 56178 with fd: 16
```

Barracuda09, 2022-10-02:
Thanks @Saentist

So building SatPI works with this. Are there no tuners in this machine?

Saentist, 2022-10-02:
I attach usb DVB-T already

Saentist, 2023-03-05:
Is there a way to ignore this error, without patching file each time when get latest source?
Something as:
`make -ignore-BackTrace`
or autodetect AlpineLinux and ignore BackTrace
I dot use docker (native run is better no matter), but it use AlpinLinux

Barracuda09, 2023-03-06:
Hi @Saentist 

Could you please checkout commit 2a0061b578608234b8d424a9830955fe1867abe1

Saentist, 2023-03-07:
This work, 
there is a error 
src/Utils.cpp:34:2
but build pass

<details><summary>expand details</summary>

```make
wafer:/opt/SATPI# git stash
Saved working directory and index state WIP on master: ecfa3e4 Fix: No need to check nullptr when using new, it will give an std::bad_alloc exception on fail
wafer:/opt/SATPI# git pull
Updating ecfa3e4..2a0061b
Fast-forward
 Makefile                          |  7 +++++++
 README.md                         | 32 +++++++++++++++++---------------
 checks/backtrace.cpp              |  8 ++++++++
 src/TransportParamVector.cpp      |  2 +-
 src/Utils.cpp                     |  5 +++++
 src/input/DeviceData.cpp          |  1 +
 src/input/InputSystem.h           |  1 +
 src/input/dvb/delivery/DVBS.cpp   | 15 +++++++++++----
 src/input/dvb/delivery/DVBS.h     |  8 +++++---
 src/input/dvb/delivery/DiSEqc.cpp | 16 +++++++++++-----
 src/input/dvb/delivery/DiSEqc.h   |  8 +++++++-
 web/frontend.html                 |  3 ++-
 12 files changed, 76 insertions(+), 30 deletions(-)
 create mode 100644 checks/backtrace.cpp
wafer:/opt/SATPI# make CC='ccache gcc' LIBDVBCSA=yes |tee build6.txt
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Version.cpp -o obj/Version.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/InterfaceAttr.cpp -o obj/InterfaceAttr.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HeaderVector.cpp -o obj/HeaderVector.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HttpServer.cpp -o obj/HttpServer.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/HttpcServer.cpp -o obj/HttpcServer.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Log.cpp -o obj/Log.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Properties.cpp -o obj/Properties.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/RtspServer.cpp -o obj/RtspServer.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/main.cpp -o obj/main.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Satpi.cpp -o obj/Satpi.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Stream.cpp -o obj/Stream.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StreamClient.cpp -o obj/StreamClient.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StreamManager.cpp -o obj/StreamManager.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/StringConverter.cpp -o obj/StringConverter.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/TransportParamVector.cpp -o obj/TransportParamVector.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/Utils.cpp -o obj/Utils.o
src/Utils.cpp:34:2: warning: #warning Compiling with no backtrace! [-Wcpp]
   34 | #warning Compiling with no backtrace!
      |  ^~~~~~~
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/M3UParser.cpp -o obj/base/M3UParser.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/Thread.cpp -o obj/base/Thread.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/ThreadBase.cpp -o obj/base/ThreadBase.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/TimeCounter.cpp -o obj/base/TimeCounter.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/XMLSaveSupport.cpp -o obj/base/XMLSaveSupport.o
g++ -c  -DHAS_NO_BACKTRACE_FUNCTIONS -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DLIBDVBCSA -DHAS_NP_FUNCTIONS src/base/XMLSupport.cpp -o obj/base/XMLSupport.o
[... cut ...] t/StreamThreadRtcpBase.o obj/output/StreamThreadRtcp.o obj/output/StreamThreadRtcpTcp.o obj/output/StreamThreadRtp.o obj/output/StreamThreadRtpTcp.o obj/output/StreamThreadTSWriter.o obj/socket/HttpcSocket.o obj/socket/TcpSocket.o obj/socket/SocketAttr.o obj/socket/UdpSocket.o obj/upnp/ssdp/Server.o obj/decrypt/dvbapi/Client.o obj/decrypt/dvbapi/ClientProperties.o obj/decrypt/dvbapi/Keys.o obj/input/dvb/Frontend_DecryptInterface.o -o satpi  -pthread -lrt -ldvbcsa
wafer:/opt/SATPI#
```
</details>

Barracuda09, 2023-03-07:
Hi @Saentist 

Yes, I added the `#warning` the warning on purpose. To indicate that backtrace is not functional.