Barracuda09/SATPI #163: Suggestion: Child PIPE pid filtering

lars18th, 2022-10-03:
Hi @Barracuda09 ,

I discovered today that when using the Child PIPE to ingest a MPTS to the SatPI, when a client request to receive _some_ pids, it receives the full transport stream. Since this is rather inefficient, could you add a simple pid filtering when using the child PIPE input?

Thank you.

lars18th, 2022-10-14:
Hi @Barracuda09 ,

Please, could you point to the source part where the (software) pid filtering is done? Perhaps it's so simple to implement this and I could do it by myself.

Regards.

Barracuda09, 2022-10-14:
Hi @lars18th 

Sorry for the late reply.
Here is the reading part of the ChildPIPE:
https://github.com/Barracuda09/SATPI/blob/c4077cab8048d211d8b7181a9da05f5d1fc61409/src/input/childpipe/TSReader.cpp#L116-L129

But there is no real software filtering (only for parsing PMT, SDT etc.)
It would be possible to hack it in there just make a tmp `mpegts::PacketBuffer tmpbuffer;` and read the amount that should fit in buffer like it is with `buffer.getAmountOfBytesToWrite()`
But the problem at the moment is getting the PIDs out of (with somethink like `_deviceData.getFilterData()...` ) :
https://github.com/Barracuda09/SATPI/blob/c4077cab8048d211d8b7181a9da05f5d1fc61409/src/mpegts/Filter.h#L37-L40

But this is al a hack and Filter should be changed to do this job.

Barracuda09, 2022-10-14:
Here it should go somehow:
https://github.com/Barracuda09/SATPI/blob/c4077cab8048d211d8b7181a9da05f5d1fc61409/src/mpegts/Filter.cpp#L78-L95

At the moment `const mpegts::PacketBuffer &buffer` is const so it can not be changed but here is should be changed. Here you need to remove the PIDs (_in line 92_) that are not requested for `mpegts::PacketBuffer &buffer`

lars18th, 2022-10-14:
Hi @Barracuda09 ,

In order to not change the code a lot, and for maintaining the encapsulation, I've another different approach:

- The Filter class continues to have the `addData(id,buffer)` method with `const` parameters. And we add another member to create a "parallel/side" array/(binary)vector that indicates if a packet in the buffer requires to be discarded. Then when some process reads from the Buffer after the Filter processing, the code can check in the side-binary-vector if the packet could be discarded or not.

What you think about this approach?

lars18th, 2022-10-14:
Hi @Barracuda09 ,

And a dirty solution is to use the PID ID 8192 (0x2000) as a hot value to discard any output packets from the buffer. So we only need to change the PID ID of the packets inside the function `addData(id,buffer)` when `_pidTable.isPIDOpened(pid)` is false. However, in this case we need to remove the `const` of the Buffer.

Please, comment what you prefer.

Barracuda09, 2022-10-14:
Hi @lars18th 

I would keep the logic in the Filter class. 
So my approuch would be to change the name from `addData(..)` to `filterData(..)` and remove the const. and modify the buffer there. So no other class should know of any logic to handle and removed packets.
The `mpegts::PacketBuffer` is designed to directly prepair for decoding and/or sending.

lars18th, 2022-10-20:
Hi @Barracuda09 ,

I've implemented more or less what I need. All with strict encapsulation, so all is done in `mpegts::PacketBuffer` and `mpegts::Filter`. However, I've a problem:
- After changing the point where the ChildPipe adds the data to the buffer (with the call `_deviceData.getFilterData().addData(_feID, buffer);`) my code is executed. But inside the `mpegts::Filter` the member `_pidTable` seems to be empty. Therefore I can't filter-out any pid because I don't know the list of requested pids.

Any idea?

lars18th, 2022-10-20:
Hi @Barracuda09 ,

Perhaps the problem is because you're not updating the `_pidTable` inside the ChildPipe.

I don't see anything related to this part from the FrontEnd module: https://github.com/Barracuda09/SATPI/blob/c4077cab8048d211d8b7181a9da05f5d1fc61409/src/input/dvb/FrontendData.cpp#L358-L372

Perhaps it's necessary to change this function?
https://github.com/Barracuda09/SATPI/blob/c4077cab8048d211d8b7181a9da05f5d1fc61409/src/input/childpipe/TSReaderData.cpp#L54-L68

One question: your code is not passing _pid changes_ to all input devices?

Barracuda09, 2022-10-20:
Hi @lars18th 

Thanks for your effort so far!

No you are correct indeed, it is not parsed in ChildPIPE at the moment.
Maybe it is just copy that from FrontendData.cpp to TSReaderData.cpp

lars18th, 2022-10-21:
Hi @Barracuda09 ,

Finally I've done it! Quite complex because the ChildPIPE is not prepared for supporting any pid filtering. Nevertheless, I'll need some time to cleanup the code (a lot of unnecessary logs and other checks). And I want to add the option to enable/disable the internal pid filtering. I feel this could be added to FILE and/or STREAMER inputs as well.

Please could you check and merge my backport PR #164 in the meantime?
Regards.

Barracuda09, 2022-10-22:
Hi @lars18th 

That is nice you have implemented the filtering!

About #164 and backports for older GCC prior to 8.3 is for me a bit of a problem, because I maybe use in the future more C++17 (or C++20) features. And then I have to think about it to try to keep it compatible and compile it with 6.3.

That said, I try to keep away from that problem. Don't get me wrong, I appreciate your help and efforts as always. 

Is there in your case only 6.3?
Are these the only 'problems' that do not compile?

Kind regards

lars18th, 2022-10-22:
Hi @Barracuda09 ,

Regarding the internal pid filtering I'm finishing it. However, I want to do more cleanups. You can check initial changes at https://github.com/lars18th/SATPI/commit/c2fa2d8e49a852ea9395c3908e451c74db3f8433 for review.

But about #164, let me to clarify my proposal:
- In many platforms recent versions of GCC are not available. So, backward compatibility is desirable.
- You're free to use _any_ C++17 or C++20 features. I'm not proposing to make the code fully backwards compatible.
- I'm only providing a _fully optional_ patch to recover the compatibility with pre-C++17 compilers.
- All the changes are isolated. Very compact, and protected by defines.
- Only two changes are simple (one missing include, and the split of two commands to put first outside the if check). So these last are good to replace the current code.

So, now you can merge this change, and all continues equal. And in the future, if something will fail then I'll try to help recovering the lost compatibility. So, you don't lost anything. You continue developing equally.

Please, don't think that it's easy to upgrade in all platforms. Your project is made to run in embbeded devices, and there it's common to have only legacy compilers available.

lars18th, 2022-10-22:
Hi @Barracuda09 ,

This branch contains my current implementation of the pid filtering for ChildPIPE:
 https://github.com/lars18th/SATPI/tree/soft-pidfilter

Please, check it! It's fully functional. However, I want to improve it with:
1. Configurable option to enable/disable it.
2. Promote it to other inputs.

I will create a PR when it's finished. However, please let me know your comments.
Regards.

Barracuda09, 2022-10-23:
Hi @lars18th 

Many thanks for your effort and contribution so far and it looks good, but I have made some pointers/comment in here: https://github.com/lars18th/SATPI/tree/soft-pidfilter

Please do not see it as just criticism but as feedback.

Regards

lars18th, 2022-10-23:
Hi @Barracuda09 ,

Thank you for rewiving it. I'm pleased to read your feedback. Where it is?

lars18th, 2022-10-23:
Hi @Barracuda09 ,

My implementation of the configuration for the Child PIPE is done: https://github.com/lars18th/SATPI/commit/55de3e79b51b60ae50c56b3f4ec14a3f5d149c39

My idea is to move some code to the class Device (using virtuals) to make it more abstract. Then the same functionality will be available for all inputs. Including real "dvb tuners". The reason is because not all implementations have hardware pid filtering. So perhaps for some special tuners it has sense. Remember that by default the option is disbled. So it will not have any side effect.

I'm still waiting for your comments! 😉

Barracuda09, 2022-10-23:
> Hi @Barracuda09 ,
> 
> Thank you for reviewing it. I'm pleased to read your feedback. Where it is?

Here it is my feedback https://github.com/Barracuda09/SATPI/commit/c2fa2d8e49a852ea9395c3908e451c74db3f8433#r87629546

lars18th, 2022-10-26:
Hi @Barracuda09 ,

Thank you for the feedback. My final implementation is on #165 . Please check it.

Barracuda09, 2022-10-27:
Hi @lars18th 

Many thanks for you effort, I will check it out.

Barracuda09, 2022-11-01:
Hi @lars18th 

I have merged your pull request #165, Thanks for you efforts!

lars18th, 2022-11-02:
Hi @Barracuda09 ,

> I have merged your pull request #165, Thanks for you efforts!

Great! And I agree with your changes. I'll try to implement it soon for STREAM and FILE inputs.
Regards.

closed 2022-11-02 by lars18th