Barracuda09/SATPI #166: Github releases are broken and possible new release?

lgehr, 2022-10-26:
Hey!
Thanks for this great project! 

It seems that github releases are currently broken. V.1.6.2 seems to be tagged but it does not show up on the release page. 

On the same note: Would it be possible to get a new release (or tag :D)? V.1.6.2 seems to be almost 10 month old now.

Take care!

Barracuda09, 2022-10-27:
Hi @lgehr 

Thanks for the complement.

But the release does not have bins for SatPI, it is only a tag and with packed source code you can compile yourself.
Is this what you need?

Kind regards

lgehr, 2022-10-27:
>
>But the release does not have bins for SatPI, it is only a tag and with packed source code you can compile yourself.

A release (or tag) with packed sources whould make it easier for distributions to make a package. 
I try to make a package for alpine linux ATM.

Barracuda09, 2022-10-28:
Hi @lgehr 

I have made a new release, could you check it out

Barracuda09, 2022-10-28:
Here is user @Saentist who is trying to build for Alpine linux as well, see #162

lgehr, 2022-11-06:
Thanks for the release. Do you think you will make a 1.6.3 release/tag soon? V.1.6.2 is pretty old,still

Saentist, 2022-11-06:
> Thanks for the release. Do you think you will make a 1.6.3 release/tag soon? V.1.6.2 is pretty old,still

Clone latest source and build it 

[image]

hope it no take 52min to build as in my device case ;)