Barracuda09/SATPI #169: Generate PSI (Program-specific information) metadata for Translation M3U files

Barracuda09, 2022-11-04:
The idea is to generate PSI metadata on a specific requested frequency.

The PSI will be build up by the Translation M3U file so it is possible for clients to 'search' for programs and will be redirected to the correct translation entry.

Example:
```
#EXTINF:-1 satip-freq="200", Translation to: File uri="/usr/satpi/test.ts"
rtsp://@#1/?msys=file&uri="%2Fusr%2Fsatpi%2Ftest.ts"

#EXTINF:-1 satip-freq="201", Translation to: ChildPIPE exec="cat /usr/satpi/test.ts"
rtsp://@#1/?msys=childpipe&exec="cat%20%2Fusr%2Fsatpi%2Ftest.ts"

#EXTINF:-1 satip-freq="202", Translation to: Streamer - Multicast UDP input 224.0.1.3:15001
rtsp://@#1/?msys=streamer&uri="udp@224.0.1.3:15001"

#EXTINF:-1 satip-freq="203", Translation to: ChildPIPE exec="wget -qO- http://192.168.0.104:8875/?src=2%26fe=2%26freq=11052.75%26sr=22000%26msys=dvbs2%26mtype=8psk%26pol=h%26fec=23%26pids=0,1,16,17,18,1039,5420,5421,5424"
rtsp://@#1/?msys=childpipe&exec="wget%20-qO-%20http:%2F%2F192.168.0.104:8875%2F%3Fsrc=2%%26fe=2%%26freq=11052.75%%26sr=22000%%26msys=dvbs2%%26mtype=8psk%%26pol=h%%26fec=23%%26pids=0,1,16,17,18,1039,5420,5421,5424"

```
This should create PSI metadata for freq 200, 201, 202 and 203 when for example `rtsp://ip.of.my.box/?freq=10101` is requested from client during a channel search

Barracuda09, 2022-11-04:
Please use discussion #170 for ideas an possible sollutions

lars18th, 2022-11-05:
Hi @Barracuda09 ,

From my point of view this could be interesting. However, quite complex. I prefer before to implement these changes in the translation:
- Limit the translation to specific SRC. So if the client requests `satip://...?src=8&...` then only in this case (SRC==8) do the translation. Usefull to restrict the translation to one specific satellite position only. It could be done adding `satip-src=...` in the `EXTINF` part.
- Similar to the previous, provide a method to translate only if DVB-T or DVB-C is requested with `satip-mtype=...`.
- Add the option to _filter_ with an external tool the TS stream. For example with `satip-filter-exec=my_binary%20param`. In this case, the read is done with the translated input, but then injected to this external tool (with a PIPE) to filter it reading from the PIPE. Some tools like the TSDuck could be used then to generate the PSI metadata that you want to create the scanning tables.
- And finally remember that in the TODO is to generate the translated RTCP messages with the translation (see #123). So if DVB-S2 is requested and translated to Child PIPE, then send translated values for this request. Some clients refuse to tune when receive the not-translated messages.

I feel this could be more easy to implement. Do you think so?

Barracuda09, 2022-11-05:
Hi @lars18th 

- the `satip-src="xx"` is already implemented with #94
- the `satip-mtype="DVB-T"` I have to think about the use of it.
- this `satip-filter-exec=my_binary%20param` I don't see this as usefull for me, as I see the PSI metadata generation as a study object for me (and maybe with help, if there is any).
- Yes #123 have to look into this again.

lars18th, 2022-11-09:
Hi @Barracuda09 ,

1. YES, and thank you for updating the examples to note it.
2. This will be useful when you implement #123 and want to remap DVB-T and/or DVB-C transponders. In any other case is useless.
3. Perhaps this will not be useful for you. However, I feel it's interesting.
4. I hope you will want to implement it soon.

Regards.

lars18th, 2022-11-13:
Hi @Barracuda09 ,

Regarding you new changes (WIP) with the PSI metadata, some comments:

1. The new implementation of the RTCP description messages seems to work (almost with DVB-S mappings). So I get:
 `a=fmtp:33 ver=1.0;src=1;tuner=4,240,1,15,11953.5000,h,dvbs,qpsk,off,27500,34;pids=`
 However, as you can see the pid list is empty. You will fix this?
2. I'm not sure about the _final_ objective. So I feel I need to wait until you complete the implementation. But in any case, please leave the option to enable/disable this translation. For example, in my case I don't want to change at all the PSI data of _some_ remapped transponders. That's because I want to analyze the content, and not only to play them with a SAT>IP client. You'll support this option?
3. In case of a _configurable_ PSI translation. It could be some syntax in the mapping.m3u to manipulate it? In this case, could you first provide an example, please?

Regards.