Barracuda09/SATPI #172: Setting for Tuner Go to sleep

SaulGoodman1337, 2022-11-12:
Can we have a setting where we can specify in seconds when satpi will put the tuners into sleep mode after no tuner is active ?
Currently it looks like the tuners do not go into sleep mode at all.
With minisatip it worked great though.

Barracuda09, 2022-11-12:
Hi @SaulGoodman1337 

Thanks for your request!
Do you know how does that works? or how can I check this in Engima2 (I presume).

Regards,
Marc

SaulGoodman1337, 2022-11-12:
I think you'll find what you need here, right? 
https://www.kernel.org/doc/html/v4.8/media/kapi/dtv-core.html

there is something that sounds like it

```
int (*)(struct dvb_frontend *fe) sleep
    callback function used to put the tuner to sleep.
```

does this help?

SaulGoodman1337, 2022-11-13:
or this helps ? 
https://www.kernel.org/doc/html/v4.9/media/uapi/dvb/fe-set-voltage.html

SaulGoodman1337, 2022-11-13:
And Here in enigma2 frontend.cpp
https://github.com/OpenPLi/enigma2/blob/04ae50fa033ffd689f7ec4227caa385bad2f7990/lib/dvb/frontend.cpp#L2725

Barracuda09, 2022-11-13:
Hi @SaulGoodman1337 

Yes many thanks for your input, so basically it should turn the power off to the LNB after an adjustable time?

SaulGoodman1337, 2022-11-13:
Yes, exactly. In my case, that's 6 watts that are uselessly consumed when no one is watching. 
Thank you for an effort!

Barracuda09, 2022-11-13:
Yes , I understand the situation. I will try to look into this as soon as possible.

Thanks for you feedback!

SaulGoodman1337, 2022-11-16:
Awesome thanks. Can you possibly provide me with a binary for a gigablue 4k?