Barracuda09/SATPI #181: Artifacts with commit 9870875

nikolauzi11, 2023-01-15:
Hi @Barracuda09 , @lars18th 

After some time I wanted to update again.
Unfortunately, I found that commit 9870875 causes artifacts for me. Before that everything is ok.
I have attached a logfile.

Thanks
[satpi.zip](https://github.com/Barracuda09/SATPI/files/10420014/satpi.zip)

Barracuda09, 2023-01-15:
Hi @nikolauzi11 

Thanks for bringing this to my attention!
But it should have had no impact on the DVB frontend types.

I have to look into this.

Kind regards,
Marc

Barracuda09, 2023-01-16:
Hi @nikolauzi11 

To be clean this commit 93aee95919a856c2855ef6b6e692c970236d1371 works? So only after this, you get artifacts?

Or was this 08582c36491931a40cc62b5e777a1cfc7ed0b59e the last version that worked?

nikolauzi11, 2023-01-16:
Hi @Barracuda09 

Unfortunately, this makes it even worse.

Barracuda09, 2023-01-17:
Hi @nikolauzi11 

Thanks for the feedback!

Could you please try one more time, to modify the if in line 87-100
https://github.com/Barracuda09/SATPI/blob/03744a21a4e2a68c43359f792ba8c3c30f194edb/src/mpegts/PacketBuffer.h#L87-L100

To this:
```
		/// This function will return the number of completed TS Packets that are
		/// in this TS buffer
		std::size_t getNumberOfCompletedPackets() const {
			if (full()) {
				return NUMBER_OF_TS_PACKETS;
			}
			return (_writeIndex - RTP_HEADER_LEN) / TS_PACKET_SIZE;
		}

		/// This function will return the first un-filtered TS Packets AND RESETS to
		/// be processed.
		std::size_t getBeginOfUnFilteredPackets() const {
			const std::size_t index = _processedIndex;
			_processedIndex = _writeIndex;
			if (index == RTP_HEADER_LEN) {
				return 0;
			}
			return (index - RTP_HEADER_LEN) / TS_PACKET_SIZE;
		}

```

nikolauzi11, 2023-01-17:
Hi @Barracuda09 

With this, it looks better than yesterday, but there are still artifacts.
To make sure we are on the same level, I undid the change from yesterday.

Barracuda09, 2023-01-17:
Yes use the most current commit and use this patch

Are you using still the RPI with the sundtek tuner?

nikolauzi11, 2023-01-18:
Hi @Barracuda09 

I have now tested it again with the latest commit. Unfortunately there are still artifacts.
Yesterday I tested the patch with commit 9870875.
Yes, I still use a Raspberry Pi 3 B+ and the Sundtek dual tuner.

Barracuda09, 2023-01-18:
Hi @nikolauzi11 

Many thanks for your time, once more! that is much appreciated. 

Could you please try the latest commit

nikolauzi11, 2023-01-18:
Hi @Barracuda09 

> Many thanks for your time, once more! that is much appreciated.

No problem, I'm very happy to do that.

It looks very good now, this commit seems to fix the problem.

Thanks a lot!

nikolauzi11, 2023-01-18:
Unfortunately, artifacts still appear briefly every few minutes.

nikolauzi11, 2023-01-18:
After I restarted the VDR again, there are no more artifacts for now. I think I should observe it for a day or two in peace.

Barracuda09, 2023-01-19:
Hi @nikolauzi11 

Thanks for testing. I try to test as soon as possible again with my setup here.

nikolauzi11, 2023-01-19:
Hi @Barracuda09 

In fact, short artifacts still appear about every 10-20min.
I tried to capture this in a log file.
It occurred around 21:47:43.
[SatPI[33140].zip](https://github.com/Barracuda09/SATPI/files/10461236/SatPI.33140.zip)

Barracuda09, 2023-01-20:
Hi @nikolauzi11 

unfortunately I can not see any strange behavior in this log. I see that VDR is always very busy with scanning PMTs.
  
But around this time, VDR is switching (device 1) to a different channel to scan for PMT changes
```
Jan 19 21:47:37 satpi SatPI[33140]: RTSP Stream data from client None with IP 192.168.112.26 on Port 55480: PLAY rtsp://192.168.112.99/stream=101?src=1&freq=10993&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&sr=22000&fec=56 RTSP/1.0
Jan 19 21:47:37 satpi SatPI[33140]: CSeq: 265
Jan 19 21:47:37 satpi SatPI[33140]: Session: 0287292145
Jan 19 21:47:37 satpi SatPI[33140]: User-Agent: vdr-satip/2.4.1 (device 1)
```
```
Jan 19 21:47:58 satpi SatPI[33140]: RTSP Stream data from client None with IP 192.168.112.26 on Port 55480: PLAY rtsp://192.168.112.99/stream=101?src=1&freq=11052&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&sr=22000&fec=23 RTSP/1.0
Jan 19 21:47:58 satpi SatPI[33140]: CSeq: 295
Jan 19 21:47:58 satpi SatPI[33140]: Session: 0287292145
Jan 19 21:47:58 satpi SatPI[33140]: User-Agent: vdr-satip/2.4.1 (device 1)
Jan 19 21:47:58 satpi SatPI[33140]: <CRLF>
```

Maybe you could try for now, to try to reduce the channel scan in VDR and see if that changes this behavior.

nikolauzi11, 2023-01-20:
Hi @Barracuda09 

I have disabled "EPG scan" and "frontend reuse" for now.
However, I never had artifacts before commit 9870875.

nikolauzi11, 2023-01-20:
Hi @Barracuda09 

> I have disabled "EPG scan" and "frontend reuse" for now.

With that, unfortunately, there are still artifacts.

Barracuda09, 2023-01-21:
Hi @nikolauzi11 

> With that, unfortunately, there are still artifacts.

Yes, I was afraid of that. I will have to take a closer look at this problem.

Barracuda09, 2023-02-20:
Hi @nikolauzi11 

Could you please try the last commits, to see if this solves this issue?

Thanks!

Barracuda09, 2023-02-22:
Hi @nikolauzi11 

Thanks for trying, and sorry for the trouble. Could you please give a log of this?

Thanks

nikolauzi11, 2023-02-22:
Hi @Barracuda09 

Here is a log. I used the last commit from today.
The channel change was around 19:45:00.
[6425.zip](https://github.com/Barracuda09/SATPI/files/10807453/6425.zip)

nikolauzi11, 2023-02-22:
Hi @Barracuda09 

The stream just got completely stuck.

nikolauzi11, 2023-02-23:
Hi @Barracuda09 

Channel changes now work properly again and there were no problems for about 1 hour. After that, however, there were unfortunately short artifacts every few minutes again.
Here is also a log, the interesting time is about 21:34:49.
[12472.zip](https://github.com/Barracuda09/SATPI/files/10818409/12472.zip)

nikolauzi11, 2023-02-26:
Hi @Barracuda09 

It is possible that my statement from 2 days ago is not correct. I even had problems with the old version, which always worked fine. Something seems to have hung on the Raspi, possibly was something with the Sundtek driver. After a reboot at least the old version worked again. So I have to test the last commits again.

Barracuda09, 2023-02-26:
Hi @nikolauzi11 

Thanks for your feedback, and yes please try the latest version.

nikolauzi11, 2023-02-26:
Hi @Barracuda09 

I think the problem is solved.
I no longer have artifacts.
Thank you very much.

Barracuda09, 2023-02-27:
Hi @nikolauzi11 

Thanks for you feedback. And try it out some more.

nikolauzi11, 2023-03-03:
Hi @Barracuda09 

Unfortunately, there are problems after all. But it looks a little different than before. Before, the artifacts were always there, at intervals of a few minutes. Now it sometimes runs for a long time without problems and suddenly artifacts come again, but then they come regularly. If I then restart the VDR, are only times again gone, for a while.

nikolauzi11, 2023-05-15:
Hi @Barracuda09 

Is it possible that the problem was solved with one of the latest commits?
I have not had any problems with it for a few days now.

Barracuda09, 2023-05-16:
Hi @nikolauzi11 

Well it could be one or more 'optimizations' I have try to make to the code.
But the problem is that I did not experience it myself.

You could try to enable, in the frontend view, the **'Filter PCR for Timing'** to see if the artifacts appears again.
To be clear this is not necessary for tuners so leave it disabled for normal operation. This was previous, enabled by default.
And by changing some of the PID filtering it could (or needed) to check every packet to see if it was a PCR packet.

But please keep testing this, and I appreciate your feedback (as always).

nikolauzi11, 2023-05-16:
Hi @Barracuda09 

Activating 'Filter PCR for Timing' did not bring back the artifacts, then it must have been other optimizations.
Anyway, I'm happy that it works really well again now.

Thank you!

I will of course stay tuned.

Barracuda09, 2023-12-15:
Hi @nikolauzi11 

Did you use RTP-over-TCP support in VDR when testing this?
 
Because I just came across a bug that was there (probably) a long time (or maybe always).

The problem is/was the way the RTCP and RTP packets are send via network to the same port. That could mangle the packets by interlacing them.

Kind regards,
Marc

nikolauzi11, 2023-12-15:
Hi @Barracuda09 

I never used RTP-over-TCP because it always caused problems when I tested it and I didn't know whether it was due to the vdr-satip-plugin or SatPI.

Barracuda09, 2024-01-20:
I will close this defect see: #207 for more info

commit a4d302b8 (2023-02-20): Fix: Possible Artifacts with commit 9870875 should fix #181 #187
closed 2024-01-20 by Barracuda09