Barracuda09/SATPI #182: Build for RPI1

SaulGoodman1337, 2023-01-16:
Hi,
Can you possibly help me and provide me a build for the RPI1 or tell me which toolchain I can use for it?
I am somehow too stupid and do not get it built.

Barracuda09, 2023-01-16:
Which image do you use, are you trying to build it directly on the RPI?

SaulGoodman1337, 2023-01-16:
Raspberry Pi OS Lite (Legacy)
Release date: September 22nd 2022
System: 32-bit
Kernel version: 5.10
Debian version: 10 (buster)

Building directly on the Pi is unfortunately not possible. The thing needs partly felt 300 years to make an apt-get update .... :D

Barracuda09, 2023-01-16:
Try one of these:

[satpi_arm_dvbcsa_1.6.2.96~g03744a2.zip](https://github.com/Barracuda09/SATPI/files/10428727/satpi_arm_dvbcsa_1.6.2.96.g03744a2.zip)
[satpi_arm_1.6.2.96~g03744a2.zip](https://github.com/Barracuda09/SATPI/files/10428728/satpi_arm_1.6.2.96.g03744a2.zip)

SaulGoodman1337, 2023-01-16:
root@satpi-rpi1:/home/pi# ./satpi --no-daeomon
Illegal instruction

SaulGoodman1337, 2023-01-16:
root@satpi-rpi1:/home/pi# cat /proc/cpuinfo 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2835
Revision	: 100000d
Serial		: 00000000a14cd328
Model		: Raspberry Pi Model B Rev 2

SaulGoodman1337, 2023-01-16:
don't stress, i started the build on pi1 and let's see what happens

> root@satpi-rpi1:/home/pi/SATPI# make
> g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DHAS_NP_FUNCTIONS src/Version.cpp -o obj/Version.o
> g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DHAS_NP_FUNCTIONS src/InterfaceAttr.cpp -o obj/InterfaceAttr.o
> g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DHAS_NP_FUNCTIONS src/HeaderVector.cpp -o obj/HeaderVector.o
> g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DHAS_NP_FUNCTIONS src/HttpServer.cpp -o obj/HttpServer.o
> g++ -c  -I src -std=c++17 -Werror=vla -Wall -Wextra -Winit-self -pthread  -O2 -s -DNDEBUG -DHAS_NP_FUNCTIONS src/HttpcServer.cpp -o obj/HttpcServer.o
>

Barracuda09, 2023-02-25:
Issue should be resolved.

closed 2023-02-25 by Barracuda09