Barracuda09/SATPI #184: [Feature request] ICAM support

dni1337, 2023-02-01:
Any Chance to implement ICAM support for CAID 098D

look here:
https://github.com/catalinii/minisatip/issues/1003

Barracuda09, 2023-02-02:
Hi @dni1337

Yes I can give this a try, are you able to test this?

dni1337, 2023-02-02:
Yes, of Course. I can compile with the modified libdvbcsa and test.
Thx

SaulGoodman1337, 2023-02-02:
yep, me too !

Barracuda09, 2023-02-14:
Hi @dni1337 & @SaulGoodman1337 

I have tried to implement this, you could try to build it with this:
`make debug ENIGMA=yes LIBDVBCSA=yes ICAM=yes`

SaulGoodman1337, 2023-02-14:
> Hi @dni1337 & @SaulGoodman1337
> 
> I have tried to implement this, you could try to build it with this: `make debug ENIGMA=yes LIBDVBCSA=yes ICAM=yes`

thanks ! 

can you help me with this ? 
`/home/home/vu/opt/toolchains/stbgcc-8.3-0.4/bin/../lib/gcc/arm-unknown-linux-gnueabihf/8.3.0/../../../../arm-unknown-linux-gnueabihf/bin/ld: cannot find -ldvbcsa
collect2: error: ld returned 1 exit status
`

Barracuda09, 2023-02-14:
@SaulGoodman1337 

Did you build LIBDVBCSA with ICAM support? 
Then follow _`Cross Compiling with DVBCSA, then copy these files to your build environment for example ARM`_  steps here https://github.com/Barracuda09/SATPI/wiki/Cross-Compiling-SatPI

SaulGoodman1337, 2023-02-14:
i have no idea how to compile libdvbcsa for arm

Barracuda09, 2023-02-14:
Hi @SaulGoodman1337 

here is how I did it:
```
git clone https://github.com/glenvt18/libdvbcsa

cd libdvbcsa
source /../../../satpi/envARM.sh
./bootstrap
(maybe install tools that bootstrap needs) sudo apt-get install ...
./configure --host=arm-linux
patch -p1 < icam.diff
make
when finished libdvbcsa.so is in src/.libs
and dvbcsa.h is in src/dvbcsa
```

SaulGoodman1337, 2023-02-14:
`Wed Feb  1 14:01:22.5740 2023 [                src/decrypt/dvbapi/Client.cpp:358] Frontend: 8, PMT data to OSCam with adapter: 8  demux: 8  list_management: 0x03
Wed Feb  1 14:01:22.5740 2023 9F 80 32 82 00 3C 03 AC 4F 04 00 36 01 81 08 00  ..2..<..O..6....
Wed Feb  1 14:01:22.5740 2023 00 00 00 01 B9 27 0F 83 01 08 84 02 01 08 86 01  .....'..........
Wed Feb  1 14:01:22.5740 2023 08 87 01 08 09 04 18 54 E2 4F 09 04 18 50 E2 50  .......T.O...P.P
Wed Feb  1 14:01:22.5740 2023 09 04 18 38 EA 1E 09 04 18 68 F5 D8 09 04 09 8E  ...8.....h......
Wed Feb  1 14:01:22.5740 2023 FE 61                                            .a
Wed Feb  1 14:01:22.5740 2023 <CRLF>
Wed Feb  1 14:01:22.5740 2023 END
Wed Feb  1 14:01:22.5779 2023 [  src/input/dvb/Frontend_DecryptInterface.cpp:067] Frontend: 9, Start filter PID: 0264  demux: 0  filter: 0 (data 0x02 0xAC 0x4F mask 0xFF 0xFF 0xFF 0x00)
Wed Feb  1 14:01:22.5788 2023 [  src/input/dvb/Frontend_DecryptInterface.cpp:067] Frontend: 9, Start filter PID: 2590  demux: 0  filter: 1 (data 0x80 0x00 0x00 mask 0xF0 0x00 0x00 0x00)
Wed Feb  1 14:01:22.5792 2023 [                          src/mpegts/Filter.h:138] Frontend: 9, Updating PID filters...
Wed Feb  1 14:01:22.6065 2023 [                          src/mpegts/Filter.h:163] Frontend: 9, Set filter PID: 2590
Wed Feb  1 14:01:22.8481 2023 [                src/decrypt/dvbapi/Client.cpp:177] Frontend: 9, Send Filter Data with size 152 for demux: 0  filter: 1 PID 2590 TableID 0x80 0x30 0x8F 0x07 0x8D
Wed Feb  1 14:01:22.8502 2023 [  src/input/dvb/Frontend_DecryptInterface.cpp:067] Frontend: 9, Start filter PID: 2590  demux: 0  filter: 1 (data 0x81 0x00 0x00 mask 0xFF 0x00 0x00 0x00)
/usr/bin/satpi: symbol lookup error: /usr/bin/satpi: undefined symbol: dvbcsa_bs_key_set_ecm
root@vuduo4kse:~# 

`

Barracuda09, 2023-02-14:
Did you use the self compiled LIBDVBCSA?

Barracuda09, 2023-02-14:
You need to copy LIBDVBCSA.so you compiled to the Vu+ in `/home/root`

then start SatPI with something like: `LD_LIBRARY_PATH=/home/root/ /usr/bin/satpi`

SaulGoodman1337, 2023-02-14:
> Did you use the self compiled LIBDVBCSA?
yes, checked again

> You need to copy LIBDVBCSA.so you compiled to the Vu+ in `/home/root`
> then start SatPI with something like: `LD_LIBRARY_PATH=/home/root/ /usr/bin/satpi`
I have. nevertheless the same behavior

Barracuda09, 2023-02-14:
what does this show `objdump -T libdvbcsa.so | grep dvbcsa_bs_key`

SaulGoodman1337, 2023-02-14:
`[@ libdvbcsa]$ objdump -T src/.libs/libdvbcsa.so.1.0.1| grep dvbcsa_bs_key
00006568 g    DF .text	000001f4  Base        dvbcsa_bs_key_set
0000675c g    DF .text	000001f4  Base        dvbcsa_bs_key_set_ecm
000046d0 g    DF .text	0000002c  Base        dvbcsa_bs_key_alloc
000046fc g    DF .text	00000004  Base        dvbcsa_bs_key_free
`

Barracuda09, 2023-02-14:
So it is there, but for some reason it can not find it or does not use the correct lib.

Barracuda09, 2023-02-15:
Hi @SaulGoodman1337 

You could try to force the correct LIBDVBCSA.so with
`LD_PRELOAD=/home/root/libdvbcsa.so /usr/bin/satpi`

SaulGoodman1337, 2023-02-15:
So now quickly looked and it seems to work. i will check it more intensively in the next few days. thank you for your service

dni1337, 2023-02-17:
@Barracuda09 
Thank you very much.

I tested for about 1 hour now on
http://127.0.0.1:8875/?fe=3&freq=11171&pol=h&ro=0.35&msys=dvbs2&mtype=8psk&plts=off&sr=22000&fec=34&pids=0,102,132,1791,1795

IT'S WORKING PERFECTLY.

Again, thank you for implementing.

Best regrads

SaulGoodman1337, 2023-02-26:
force close on caid 1830. without ICAM=yes its working 

> root@vuduo4kse:~# LD_PRELOAD=/home/root/libdvbcsa.so /usr/bin/satpi --http-path /usr/share/satpi/web --no-daemon
Tue Feb  7 22:40:06.1156 2023 [                                 src/main.cpp:341] --- Starting SatPI version: 1.6.2.113~gf2e46ae Enigma ---
Tue Feb  7 22:40:06.1166 2023 [                                 src/main.cpp:342] Number of processors online: 4
Tue Feb  7 22:40:06.1169 2023 [                                 src/main.cpp:343] Default network buffer size: 80 KBytes
Tue Feb  7 22:40:06.1181 2023 [                        src/InterfaceAttr.cpp:132] eth0: 192.168.178.4 [00:1d:ec:16:a6:bd]
Tue Feb  7 22:40:06.1192 2023 [                        src/StreamManager.cpp:068] Enumerating all devices...
Tue Feb  7 22:40:06.1195 2023 [                   src/input/dvb/Frontend.cpp:179] Detecting frontends in: /dev/dvb
Tue Feb  7 22:40:06.1192 2023 [                src/decrypt/dvbapi/Client.cpp:371] Setting up DVBAPI client
Tue Feb  7 22:40:06.1273 2023 [                   src/input/dvb/Frontend.cpp:576] Frontend Name: Vuplus FE
Tue Feb  7 22:40:06.1275 2023 [                   src/input/dvb/Frontend.cpp:580] Frontend Stat: Use advanced signal stats
Tue Feb  7 22:40:06.1580 2023 [                   src/input/dvb/Frontend.cpp:656] Frontend Type: Satellite (DVB-S)
Tue Feb  7 22:40:06.1582 2023 [                   src/input/dvb/Frontend.cpp:660] Frontend Type: Satellite (DVB-S2)
Tue Feb  7 22:40:06.1584 2023 [                   src/input/dvb/Frontend.cpp:698] Frontend Type: Unknown 21
Tue Feb  7 22:40:06.1586 2023 [                   src/input/dvb/Frontend.cpp:702] Frontend Freq: 950000 Hz to 2150000 Hz
Tue Feb  7 22:40:06.1588 2023 [                   src/input/dvb/Frontend.cpp:703] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Tue Feb  7 22:40:06.1590 2023 [                   src/input/dvb/Frontend.cpp:710] Frontend  API: 5.10 (0x050A)
Tue Feb  7 22:40:06.1617 2023 [                   src/input/dvb/Frontend.cpp:576] Frontend Name: Vuplus FE
Tue Feb  7 22:40:06.1619 2023 [                   src/input/dvb/Frontend.cpp:580] Frontend Stat: Use advanced signal stats
Tue Feb  7 22:40:06.1918 2023 [                   src/input/dvb/Frontend.cpp:656] Frontend Type: Satellite (DVB-S)
Tue Feb  7 22:40:06.1920 2023 [                   src/input/dvb/Frontend.cpp:660] Frontend Type: Satellite (DVB-S2)
Tue Feb  7 22:40:06.1922 2023 [                   src/input/dvb/Frontend.cpp:698] Frontend Type: Unknown 21
Tue Feb  7 22:40:06.1924 2023 [                   src/input/dvb/Frontend.cpp:702] Frontend Freq: 950000 Hz to 2150000 Hz
Tue Feb  7 22:40:06.1926 2023 [                   src/input/dvb/Frontend.cpp:703] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Tue Feb  7 22:40:06.1928 2023 [                   src/input/dvb/Frontend.cpp:710] Frontend  API: 5.10 (0x050A)
Tue Feb  7 22:40:06.1953 2023 [                   src/input/dvb/Frontend.cpp:576] Frontend Name: Vuplus FE
Tue Feb  7 22:40:06.1955 2023 [                   src/input/dvb/Frontend.cpp:580] Frontend Stat: Use advanced signal stats
Tue Feb  7 22:40:06.2253 2023 [                   src/input/dvb/Frontend.cpp:656] Frontend Type: Satellite (DVB-S)
Tue Feb  7 22:40:06.2255 2023 [                   src/input/dvb/Frontend.cpp:660] Frontend Type: Satellite (DVB-S2)
Tue Feb  7 22:40:06.2257 2023 [                   src/input/dvb/Frontend.cpp:698] Frontend Type: Unknown 21
Tue Feb  7 22:40:06.2259 2023 [                   src/input/dvb/Frontend.cpp:702] Frontend Freq: 950000 Hz to 2150000 Hz
Tue Feb  7 22:40:06.2261 2023 [                   src/input/dvb/Frontend.cpp:703] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Tue Feb  7 22:40:06.2263 2023 [                   src/input/dvb/Frontend.cpp:710] Frontend  API: 5.10 (0x050A)
Tue Feb  7 22:40:06.2288 2023 [                   src/input/dvb/Frontend.cpp:576] Frontend Name: Vuplus FE
Tue Feb  7 22:40:06.2290 2023 [                   src/input/dvb/Frontend.cpp:580] Frontend Stat: Use advanced signal stats
Tue Feb  7 22:40:06.2582 2023 [                   src/input/dvb/Frontend.cpp:656] Frontend Type: Satellite (DVB-S)
Tue Feb  7 22:40:06.2585 2023 [                   src/input/dvb/Frontend.cpp:660] Frontend Type: Satellite (DVB-S2)
Tue Feb  7 22:40:06.2587 2023 [                   src/input/dvb/Frontend.cpp:698] Frontend Type: Unknown 21
Tue Feb  7 22:40:06.2589 2023 [                   src/input/dvb/Frontend.cpp:702] Frontend Freq: 950000 Hz to 2150000 Hz
Tue Feb  7 22:40:06.2592 2023 [                   src/input/dvb/Frontend.cpp:703] Frontend srat: 1000000 symbols/s to 45000000 symbols/s
Tue Feb  7 22:40:06.2594 2023 [                   src/input/dvb/Frontend.cpp:710] Frontend  API: 5.10 (0x050A)
Tue Feb  7 22:40:06.2618 2023 [                   src/input/dvb/Frontend.cpp:576] Frontend Name: Vuplus FE
Tue Feb  7 22:40:06.2621 2023 [                   src/input/dvb/Frontend.cpp:580] Frontend Stat: Use advanced signal stats
Tue Feb  7 22:40:06.2914 2023 [                   src/input/dvb/Frontend.cpp:656] Frontend Type: Satellite (DVB-S)
Tue Feb  7 22:40:06.2917 2023 [                   src/input/dvb/Frontend.cpp:660] Frontend Type: Satellite (DVB-S2)
Tue Feb  7 22:40:06.2919 2023 [                   src/input/dvb/Frontend.cpp:698] Frontend Type: Unknown 21
Tue Feb  7 22:40:06.2921 2023 [                   src/input/dvb/Frontend.cpp:702] Frontend Freq: 950000 Hz to 2150000 Hz
[... cut ...]
Tue Feb  7 22:40:11.5568 2023 [                src/decrypt/dvbapi/Client.cpp:177] Frontend: 1, Send Filter Data with size 152 for demux: 0  filter: 1 PID 7330 TableID 0x80 0x30 0x8F 0x07 0x8D
Tue Feb  7 22:40:11.5585 2023 [  src/input/dvb/Frontend_DecryptInterface.cpp:067] Frontend: 1, Start filter PID: 7330  demux: 0  filter: 1 (data 0x81 0x00 0x00 mask 0xFF 0x00 0x00 0x00)
Tue Feb  7 22:40:11.5742 2023 [                src/decrypt/dvbapi/Client.cpp:464] Frontend: 1, Received even(0x00) CW: 0x2D 0x54 0xCC 0x4D 0xEC 0x27 0x78 0x8B  index: 0
root@vuduo4kse:~#

Barracuda09, 2023-02-26:
Hi @SaulGoodman1337 

What does `cat /tmp/satpi.bt`  show?

Barracuda09, 2023-02-26:
Hi @SaulGoodman1337 

Thanks for testing!
It seems that I have made a mistake with none ICAM cards.
Could you please test the latest commit

SaulGoodman1337, 2023-02-26:
same behavior 

> Wed Feb  8 02:07:17.1612 2023 [  src/input/dvb/Frontend_DecryptInterface.cpp:067] Frontend: 1, Start filter PID: 0312  demux: 0  filter: 0 (data 0x02 0x6F 0xEC mask 0xFF 0xFF 0xFF 0x00)
> Wed Feb  8 02:07:17.1615 2023 [  src/input/dvb/Frontend_DecryptInterface.cpp:067] Frontend: 1, Start filter PID: 7301  demux: 0  filter: 1 (data 0x80 0x00 0x00 mask 0xF0 0x00 0x00 0x00)
> Wed Feb  8 02:07:17.1617 2023 [                          src/mpegts/Filter.h:140] Frontend: 1, Updating PID filters...
> Wed Feb  8 02:07:17.1855 2023 [                          src/mpegts/Filter.h:165] Frontend: 1, Set filter PID: 7301
> Wed Feb  8 02:07:17.8149 2023 [                src/decrypt/dvbapi/Client.cpp:177] Frontend: 1, Send Filter Data with size 152 for demux: 0  filter: 1 PID 7301 TableID 0x81 0x30 0x8F 0x07 0x8D
> Wed Feb  8 02:07:17.8171 2023 [  src/input/dvb/Frontend_DecryptInterface.cpp:067] Frontend: 1, Start filter PID: 7301  demux: 0  filter: 1 (data 0x80 0x00 0x00 mask 0xFF 0x00 0x00 0x00)
> root@vuduo4kse:~# cat /tmp/
> bcm/           minisatip.log  ntpv4.local    nxserver_ipc   satpi.bt       
> root@vuduo4kse:~# cat /tmp/satpi.bt 
> /usr/bin/satpi(_ZN5Utils15createBackTraceEPKc+0x34)[0xf3ccc]
> /usr/bin/satpi[0xd16f8]
> /lib/libc.so.6(__default_sa_restorer+0x0)[0xb6bf1ba0]
> /usr/bin/satpi(_ZN7decrypt6dvbapi4Keys3setEPKhii+0x70)[0x19ba84]
> /usr/bin/satpi(_ZN7decrypt6dvbapi16ClientProperties6setKeyEPKhii+0x48)[0x1a0ebc]
> /usr/bin/satpi(_ZN5input3dvb8Frontend6setKeyEPKhii+0x4c)[0x19f72c]
> /usr/bin/satpi(_ZN7decrypt6dvbapi6Client11threadEntryEv+0x918)[0x190038]
> /usr/bin/satpi(_ZN4base10ThreadBase15threadEntryBaseEv+0x80)[0xfc948]
> /usr/bin/satpi(_ZN4base10ThreadBase15threadEntryFuncEPv+0x2c)[0xfcb94]
> root@vuduo4kse:~# 
>

Barracuda09, 2023-02-26:
> Hi @SaulGoodman1337
> 
> What does `cat /tmp/satpi.bt` show?

Barracuda09, 2023-02-26:
Could you please try once more this new commit

SaulGoodman1337, 2023-02-26:
looks better. i will try this a few days. thanks

Barracuda09, 2023-02-26:
@SaulGoodman1337 

Yes, please try ICAM and none ICAM cards.

Thanks for you patience and help!