Barracuda09/SATPI #185: Access 1 childpipe "vtuner" from 2 clients

neo7530, 2023-02-02:
How can i connect 2 clients to the same service/frequency? I set up 2 childpipe frontends with one mapping.m3u when one Client is connected and Accesses the tuner, the 2nd Client gets a time out. The 2nd vtuner won't provide the same frequency. The only way to get a second Client connected is to Set up the same mapping with an other frequency with the 2nd tuner.

Barracuda09, 2023-02-02:
Hi @neo7530 

You can try to make a copy of the mapping.m3u, for example to mappingT1.m3u and mappingT2.m3u and use these for the tuners.

Or maybe I do not fully understand you question.

neo7530, 2023-02-02:
I've tried already. The log says that frequency 200 is already in use by mapping1.m3u... Will send a log tomorrow.

Barracuda09, 2023-02-02:
Could you also provide the frequency translation from the mapping.m3u

neo7530, 2023-02-03:
This can be closed. Me made a mistake by mixing TS-Reader with Childpipe in mapping. Now i understood the scheme and got it working with only one mapping-file.

Barracuda09, 2023-02-03:
Hi @neo7530 

Could you give an example for the mapping file you use? Maybe it is of use to other users as well

neo7530, 2023-02-03:
Here we go:

#EXTM3U

# To use Translation you need to set in the web interface
# - 'Advertise as' to exp. 'DVB-S2'
# - 'Transformation enable' should be checked
# - and a valid m3u file like this

#EXTINF:-1 satip-freq="201", Translation to: ChildPIPE exec="cat /usr/satpi/test.ts MPTS-File"
rtsp://@#1/?msys=childpipe&exec="cat%20multiprog.ts"

#EXTINF:-1 satip-freq="202", Translation to: ChildPIPE exec="wget -qO- HTTP-Stream from VU+-Box ARD HD"
rtsp://@#1/?msys=childpipe&exec="wget%20-qO-%20http:%2F%2F10.24.10.10:8001%2F1:0:19:283D:41B:1:FFFF0000:0:0:0:"

#EXTINF:-1 satip-freq="203", Translation to: ChildPIPE exec="wget -qO- HTTP-Stream from VU+-Box ZDF HD"
rtsp://@#1/?msys=childpipe&exec="wget%20-qO-%20http:%2F%2F10.24.10.10:8001%2F1:0:19:2B66:437:1:FFFF0000:0:0:0:"