Barracuda09/SATPI #187: Tuning problems on my VU+ Duo2 4k with 1xDVB-S FBC and 1x-DVB-C FBC (16 tuner)

SaulGoodman1337, 2023-02-16:
Have a wonderful good day Marc,
I have once again changed the receiver because I need DVB-C and I have chosen the Duo 2 4k. 
Here I have 1xDVB-S FBC and 1xDVB-C FBC installed with a total of 16 tuners.

Now I have a few problems here. 

1. the scan does not work. I start in the TVheadend a MUX scan or EPG scan comes out of the DVB-S tuners just no data.

2. tuning itself I also have problems. Sometimes it must retune a few times before it works.

I unfortunately have absolutely no idea where to start here and therefore I would pray you for help. gladly I can also set up nen remote access so you can look at it yourself.

thank you

Barracuda09, 2023-02-16:
Are the tuners correctly identified in the 'frontend overview' tab in the webui

SaulGoodman1337, 2023-02-16:
i am not sure. What I know in any case is that the tuner configuration in the TVHeadend had to be set manually. What SATPI had exported as SAT>IP was not correct.

Does that help you? I have attached two screens
[image]
[image]

SaulGoodman1337, 2023-02-16:
> What was not correctly exported? what does this show `http://192.168.0.112:8875/desc.xml`

All good. I think that some new version of Satpi has settled.

```
<root configId="0">
<specVersion>
<major>1</major>
<minor>1</minor>
</specVersion>
<device>
<deviceType>urn:ses-com:device:SatIPServer:1</deviceType>
<!-- filled in by server -->
<friendlyName>SatPI Server (192.168.178.4)</friendlyName>
<!-- filled in by server -->
<manufacturer>Marc Postema</manufacturer>
<manufacturerURL>https://github.com/Barracuda09/SATPI</manufacturerURL>
<modelDescription>SatPI is a SAT>IP Server for Linux</modelDescription>
<!-- filled in by server -->
<modelName>SatPI Server (192.168.178.4)</modelName>
<!-- filled in by server -->
<modelNumber>SatPI/1.6.2.100~g9b31903 Enigma</modelNumber>
<!-- filled in by server -->
<modelURL>https://github.com/Barracuda09/SATPI</modelURL>
<serialNumber>1S81A31231000007</serialNumber>
<!-- filled in by server -->
<UDN>uuid:50c958a8-e839-4b96-b7ae-001dec16a6bd</UDN>
<!-- filled in by server -->
<UPC>123456789012</UPC>
<iconList>
<icon>
<mimetype>image/png</mimetype>
<width>48</width>
<height>48</height>
<depth>24</depth>
<url>/assets/images/icons/sm.png</url>
</icon>
<icon>
<mimetype>image/png</mimetype>
<width>120</width>
<height>120</height>
<depth>24</depth>
<url>/assets/images/icons/lr.png</url>
</icon>
<icon>
<mimetype>image/jpeg</mimetype>
<width>48</width>
<height>48</height>
<depth>24</depth>
<url>/assets/images/icons/sm.jpg</url>
</icon>
<icon>
<mimetype>image/jpeg</mimetype>
<width>120</width>
<height>120</height>
<depth>24</depth>
<url>/assets/images/icons/lr.jpg</url>
</icon>
</iconList>
<!-- filled in by server -->
<presentationURL>http://192.168.178.4:8875/</presentationURL>
<!-- filled in by server -->
<!-- filled in by server -->
<satip:X_SATIPCAP>DVBS2-8,DVBC-8</satip:X_SATIPCAP>
<!-- filled in by server -->
<!-- filled in by server -->
<satip:X_SATIPM3U>/channellist.m3u</satip:X_SATIPM3U>
<!-- filled in by server -->
</device>
</root> 
```

But I have attached a debug log here. Here I have started a scan via TVheadend. 
The continues errors are also very disturbing. I ALWAYS have that when I switch to a channel. The switching times are sometimes 10 seconds or more. 
I see in the log also very many broken pipes... 

[image]
[satpi.zip](https://github.com/Barracuda09/SATPI/files/10754663/satpi.zip)

Barracuda09, 2023-02-16:
@SaulGoodman1337 

You could try to compile a regular build of SatPI (so no debug version) 
With:
`make ENIGMA=yes LIBDVBCSA=yes ICAM=yes `

SaulGoodman1337, 2023-02-16:
I can do that, but what's the point?
I have the problem since I have this receiver (6 weeks) and have until the day before yesterday used a normal satpi version.

Barracuda09, 2023-02-16:
Ok, what does cmd `top` say for CPU usage?

SaulGoodman1337, 2023-02-16:
I thought that's what you were getting at. The cpu gets bored. when all 16 tuners are active, i have about 15-20% cpu utilization at peak.

Barracuda09, 2023-02-16:
Ok, and memory usage?

SaulGoodman1337, 2023-02-16:
> Ok, and memory usage?

`Mem: 255672K used, 1364324K free, 66976K shrd, 8836K buff, 145024K cached`

looks good.

Barracuda09, 2023-02-16:
You could maybe try to disable in TvHeadend the `RTP/AVP/TCP transport support` just to see if that changes the behavior

SaulGoodman1337, 2023-02-16:
> You could maybe try to disable in TvHeadend the `RTP/AVP/TCP transport support` just to see if that changes the behavior

same behavior :(

the behavior here doesn't seem right to me either, does it?
```
Fri Feb  3 04:23:23.9178 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:24.0336 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:24.1425 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:24.2502 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:24.3597 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:24.4677 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:24.5750 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:24.6825 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:24.7896 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:24.8969 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:25.0148 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:25.1376 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:25.2474 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:25.3567 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:25.4655 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:25.5742 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:25.6867 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:25.8009 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:25.9080 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:26.0320 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:26.1383 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:26.2460 2023 [                   src/input/dvb/Frontend.cpp:801] Frontend: 4, FE_READ_STATUS: Resource temporarily unavailable (code 11)
Fri Feb  3 04:23:26.2470 2023 [                   src/input/dvb/Frontend.cpp:809] Frontend: 4, Not locked yet   (Timeout 3515 ms)...
Fri Feb  3 04:23:26.2475 2023 [                          src/mpegts/Filter.h:138] Frontend: 4, Updating PID filters...
Fri Feb  3 04:23:26.2482 2023 [                   src/input/dvb/Frontend.cpp:494] Frontend: 4, Opened /dev/dvb/adapter0/demux3 using fd: 37
Fri Feb  3 04:23:26.2518 2023 [                   src/input/dvb/Frontend.cpp:500] Frontend: 4, Set DMX buffer size to 18874368 Bytes
Fri Feb  3 04:23:26.2531 2023 [                   src/input/dvb/Frontend.cpp:516] Frontend: 4, Set DMX_SET_SOURCE with (Src: 3 - Offset: 32)
Fri Feb  3 04:23:26.3055 2023 [                          src/mpegts/Filter.h:163] Frontend: 4, Set filter PID: 0000
Fri Feb  3 04:23:26.3265 2023 [                          src/mpegts/Filter.h:163] Frontend: 4, Set filter PID: 0001
Fri Feb  3 04:23:26.3729 2023 [                          src/mpegts/Filter.h:163] Frontend: 4, Set filter PID: 0016
Fri Feb  3 04:23:26.3942 2023 [                          src/mpegts/Filter.h:163] Frontend: 4, Set filter PID: 0017
Fri Feb  3 04:23:26.4412 2023 [                          src/mpegts/Filter.h:163] Frontend: 4, Set filter PID: 0018
Fri Feb  3 04:23:26.4419 2023 [                   src/input/dvb/Frontend.cpp:439] Frontend: 4, Updating frontend (Finished in 3779 ms)
Fri Feb  3 04:23:26.4423 2023 [          src/output/StreamThreadRtcpBase.cpp:086] Frontend: 4, Restart RTCP/UDP stream to 192.168.178.2:50631
Fri Feb  3 04:23:26.4425 2023 [              src/output/StreamThreadBase.cpp:192] Frontend: 4, Restart RTP/UDP stream to 192.168.178.2:50630
Fri Feb  3 04:23:26.4428 2023 [                          src/HttpcServer.cpp:217] Send reply in 3990 ms
Fri Feb  3 04:23:26.4428 2023 RTSP/1.0 200 OK

```

Barracuda09, 2023-02-16:
That is because SatPI is reading the fe status to fast, and for some reason the driver is giving this "error"

SaulGoodman1337, 2023-02-16:
> I am more concerned about the reply time of 3990 ms

what does that mean?

Barracuda09, 2023-02-16:
That means TvHeadend is probably timing out before the reply is send.
Did you check https://github.com/Barracuda09/SATPI/wiki/Using-SatPI-as-frontend-for-Tvheadend

SaulGoodman1337, 2023-02-16:
> That means TvHeadend is probably timing out before the reply is send. Did you check https://github.com/Barracuda09/SATPI/wiki/Using-SatPI-as-frontend-for-Tvheadend

grace period was not set. set it and same behavior... the rest is set correctly

SaulGoodman1337, 2023-02-16:
can you include this?
https://www.kernel.org/doc/html/latest/userspace-api/media/dvb/fe-enable-high-lnb-voltage.html?highlight=fe_enable_high_lnb_voltage#c.DTV.fe.FE_ENABLE_HIGH_LNB_VOLTAGE

SaulGoodman1337, 2023-02-16:
I have now checked all the timeouts again and reset everything. The mux+epg scan doesn't seem to run into the timeout anymore.
but i still have a lot of continues errors on both tuners. DVB-C as well as DVB-S.

something is wrong here

Barracuda09, 2023-02-16:
Well there is a lot going on in the log you send (scanning on multiple frontends). 
This will interfere during logging text. This is not wise to log maybe al the time:
https://github.com/Barracuda09/SATPI/blob/9b31903e434adbe18d2a4cd4711add834cfc0e6f/src/mpegts/NIT.cpp#L214-L228

SaulGoodman1337, 2023-02-16:
I discovered something else interesting.
If I only scan DVB-S, it works reasonably well.
If I only scan DVB-C, it works reasonably well.
If I start both at the same time, DVB-S runs into a timeout.

> Well there is a lot going on in the log you send (scanning on multiple frontends). This will interfere during logging text. This is not wise to log maybe al the time:
> 
> https://github.com/Barracuda09/SATPI/blob/9b31903e434adbe18d2a4cd4711add834cfc0e6f/src/mpegts/NIT.cpp#L214-L228

yes, but you want to have a logfile?
Or what do you want to tell me with that?

Barracuda09, 2023-02-16:
Well it is also a remainder to myself.

SaulGoodman1337, 2023-02-16:
> I have to check this here is @nikolauzi11 having probably the same problem in (#181)

yes, that smells kind of similar. 
I hope you get this fixed. until then, thank you very much!

SaulGoodman1337, 2023-02-16:
LD_PRELOAD=/home/root/libdvbcsa.so /usr/bin/satpi --http-path /usr/share/satpi/web

Barracuda09, 2023-02-16:
have you enabled logging to syslog in SatPI?

SaulGoodman1337, 2023-02-18:
yes, I disabled it and still the same behavior.
I would give the softpid filter a chance. 
can you revert the commit #9870875 for the current branch to test?

Barracuda09, 2023-02-18:
Hi @SaulGoodman1337 

Thanks for testing!

You could checkout the it yourself in the master branch with:
```
git branch -f beforesoftpid 9870875
git checkout beforesoftpid
make ENIGMA=yes LIBDVBCSA=yes ICAM=yes
```

Barracuda09, 2023-02-18:
Hi @SaulGoodman1337 

I did the above, and that version seems better. I have had some issues the last month as well.

I have to check this, as it should have had no side affects.

SaulGoodman1337, 2023-02-18:
much better. i was able to complete a scan without a single continues error. now i only have 0-10 continues errors when switching channels. And interestingly, only for encrypted channels... any idea?
the version now has no icam support, right? not working anymore

Barracuda09, 2023-02-18:
@SaulGoodman1337 

No unfortunately there is no icam support in that commit

SaulGoodman1337, 2023-02-18:
Then I go back to the newer version for now.
But I can tell you that it runs better by far.

SaulGoodman1337, 2023-02-19:
just tried to cherrypick the icam stuff into the before branch, but can't build it.

> src/mpegts/PAT.cpp: In member function 'mpegts::TSData mpegts::PAT::generateFrom(FeID, const TransformationMap&)':
src/mpegts/PAT.cpp:87:20: error: 'pid' was not declared in this scope
  tmp[1]  = 0x40 | (pid & 0x1F) >> 8;
                    ^~~
src/mpegts/PAT.cpp:87:20: note: suggested alternative: 'id'
  tmp[1]  = 0x40 | (pid & 0x1F) >> 8;
                    ^~~
                    id
src/mpegts/PAT.cpp:89:13: error: 'scrambled' was not declared in this scope
  tmp[3]  = (scrambled & 0x3) << 6 | (payloadOnly & 0x3) << 4 | (cc & 0xF);
             ^~~~~~~~~
src/mpegts/PAT.cpp:89:38: error: 'payloadOnly' was not declared in this scope
  tmp[3]  = (scrambled & 0x3) << 6 | (payloadOnly & 0x3) << 4 | (cc & 0xF);
                                      ^~~~~~~~~~~
src/mpegts/PAT.cpp:89:65: error: 'cc' was not declared in this scope
  tmp[3]  = (scrambled & 0x3) << 6 | (payloadOnly & 0x3) << 4 | (cc & 0xF);
                                                                 ^~
src/mpegts/PAT.cpp:96:22: error: 'version' was not declared in this scope
  tmp[10] = (0xC0 | ((version & 0x1F) << 1) | (currIndicator & 0x01));
                      ^~~~~~~
src/mpegts/PAT.cpp:96:47: error: 'currIndicator' was not declared in this scope
  tmp[10] = (0xC0 | ((version & 0x1F) << 1) | (currIndicator & 0x01));
                                               ^~~~~~~~~~~~~
src/mpegts/PAT.cpp:113:6: warning: unused variable 'pmtPid' [-Wunused-variable]
  int pmtPid = 0x0100;
      ^~~~~~
make: *** [Makefile:211: obj/mpegts/PAT.o] Fehler 1

Barracuda09, 2023-02-20:
Hi @SaulGoodman1337 

Could you please try the latest commit, to see if this solves the issue.

SaulGoodman1337, 2023-02-20:
> Hi @SaulGoodman1337
> 
> Could you please try the latest commit, to see if this solves the issue.

nope, sorry....same behavior

Barracuda09, 2023-02-20:
Could you make a log once more

SaulGoodman1337, 2023-02-22:
> Could you make a log once more

Of course, it's in the appendix.
I started Satpi here, did "clear all statistic" in TVHeadend and then started a scan directly.
here are still a lot of continunity errors. 

[image]
[2023_02_22_1000_satpi.log](https://github.com/Barracuda09/SATPI/files/10802330/2023_02_22_1000_satpi.log)

Barracuda09, 2023-02-22:
Thanks @SaulGoodman1337 for you effort so far.

Could you please also include commit 06b70b4 (the latest) this will not log MPEGTS Tables

SaulGoodman1337, 2023-02-22:
sure.
[image]
[2023_02_22_1100_satpi.log](https://github.com/Barracuda09/SATPI/files/10802733/2023_02_22_1100_satpi.log)

Barracuda09, 2023-02-22:
Could you limit the scan for only 4 DVB-S2 and 4 DVB-C to see if that make any difference?

SaulGoodman1337, 2023-02-22:
rather worse. furthermore, SatPI seems to die spontaneously during scanning. This was already noticed during the 11 o'clock scan.
[2023_02_22_1200_satpi.log](https://github.com/Barracuda09/SATPI/files/10803132/2023_02_22_1200_satpi.log)

SaulGoodman1337, 2023-02-23:
with your latest commit :D :D
[image]

Barracuda09, 2023-02-23:
Hi @SaulGoodman1337 

Could you please check last commit? This should revert most of commit https://github.com/Barracuda09/SATPI/commit/98708753109d3e6744edc8dc7b9aeb8f6b60f416 for DVB frontends.

Barracuda09, 2023-02-23:
Hi @SaulGoodman1337 

Did you reboot the receiver at some time?

SaulGoodman1337, 2023-02-23:
Much better ! However, SatPI dies away to me still of the öfteren. now at the scan 4 times. 
currently I help myself here with a script. 
`#/bin/bash
while true; do 
	if pgrep satpi > /dev/null
	then
		echo "satpi running"
	else
		echo "starting satpi"
		LD_PRELOAD=/home/root/libdvbcsa.so /usr/bin/satpi --http-path /usr/share/satpi/web
	fi
	sleep 3
done
`

but as you can see still not perfect. i'm getting desperate :(
[image]

Barracuda09, 2023-02-23:
> but as you can see still not perfect. i'm getting desperate :(

Me too

SaulGoodman1337, 2023-02-23:
i think that is of interest. i'll try to lift that up
> [ 2446.899971] vmap allocation for size 28315648 failed: use vmalloc= to increase size.

SaulGoodman1337, 2023-02-23:
ahaaaaa...size 28315648 byte seemed so familiar. i had specified a DVR buffer (MB) of 27MB. Now I ask myself the question, whether you have influence on the kernel buffer? 16*27MB is a lot.
Like for example with Minisatip the buffer parameter. am I right?

I have now set the DVR buffer to 3MB, started a scan and look... not a single continunity error. 

I think this only applies to the new build from today. The DVR buffer is set to 27MB only since yesterday evening.

actually, however, I am even more confused than before :D
should I maybe set up a VPN profile for you and you "make yourself a picture" of it?

SaulGoodman1337, 2023-02-26:
so i have now tried your latest build and the whole thing works. thanks !
But it is so incredibly slow.... I have the problem that when a scan runs over the 16 tuners, I can no longer tune a station because it just takes too long. He then runs into a timeout.
I think a parallelization makes here sense, right?

Barracuda09, 2023-02-26:
Hi @SaulGoodman1337 

Sorry, i do not really understand what you mean here. Could you elaborate a bit more on parallelization in this contect?

SaulGoodman1337, 2023-02-26:
> Hi @SaulGoodman1337
> 
> Sorry, i do not really understand what you mean here. Could you elaborate a bit more on parallelization in this contect?

Splitting the tasks into threads like minisatip does. If I see it correctly, satpi processes one task after the other while minisatip, for example, is able to do things at the same time.

In the TVHeadend this is quite nice to see when you start a scan. It is very slowly activated one tuner after the other. Some tasks could be processed in parallel.

The problem is simply that when satpi is busy with other things, the tuning simply takes too long. do you understand? 
Example:
When I start a scan, why does satpi wait with the tuning of the DVB-C frequencies until all 8 DVB-S tuners are done with the tuning?

You should really take a look at my setup here and try something. You will quickly notice that satpi has reached its limits here.

SaulGoodman1337, 2023-02-26:
i'm really close to just throwing all this receiver shit out the window and not wasting my time with it anymore. 
It's just incredibly unreliable.

As soon as there is somehow a tuner active, just nothing works properly. Sometimes I need 20 seconds until a channel was opened, sometimes it goes without problems.
Sometimes nothing works and I have to restart satpi... It really sucks completely and I have already invested so much time in this topic that it is slowly too much for me.

please do not feel attacked but it just pisses me off. I've tried so many receivers and with each device I've always had some kind of fucking problems.

I could pull out my hair and scream at the same time.

Barracuda09, 2023-02-26:
Hi @SaulGoodman1337 

Well SatPI does use an thread for every frontend that is opened. I could not think of a reason why it should wait.

SaulGoodman1337, 2023-02-26:
Then I'm honestly at my wits' end. No idea what I should still do here. I guess it's not meant to be.
Thank you anyway

Barracuda09, 2023-02-26:
You should maybe not use all frontends for tuning. Just to start with only 4 DVB-S2 and 4 DVB-C. Then set in TvHeadend `initial scan` for only these frontends.

SaulGoodman1337, 2023-02-27:
Yeah but be honest, that sucks. If you have an interest in your software working with these devices, I can offer you vpn or I'll have them mail it to you. Otherwise I will put the project on hold. I sunk too much time into this topic and now as a family man time is precious.

The two suggestions I can still make.

Barracuda09, 2023-02-27:
Yes thanks, you can mail them to me, and I have a look.

This is a spare time project for me as well. Don't make a lot of money of this (it only costs money to buy stuff actually)

But why would you need all frontends to tune at once? Tuning needs a lot of PIDs to be opened which maybe to much for the receiver.

SaulGoodman1337, 2023-02-27:
This is actually not necessary, but even with only one tuner activated at a time, it does not work reliably. And I think I'm not alone with this.

Yes I can totally understand with the time. I am of course also willing to make a donation

Barracuda09, 2023-02-27:
I have here an Gigablue DVB-S2X and Vu+ DVB-S2 and DVB-C versions that work reliable with TvHeadend. The only problem are the drivers of Vu+ that are not working as I expect. With low bit rate PID (like scanning) it takes a long time to receive data from the frontends.

SaulGoodman1337, 2023-02-28:
I think we are talking at cross purposes. Satpi and minisatip also cause problems when I activate only 1 or 2 tuners. It seems to me that you have to do something different with these boxes than you do so far. Probably... I'll leave the assessment up to you.
So if you want, I'll have the box mailed to you. Let's chat privately somehow about the address.

Barracuda09, 2023-02-28:
Yes send me an mail, we first discus there further on a solution.