Barracuda09/SATPI #189: [Feature Request] DVB-S2X tunning support

lars18th, 2023-02-22:
Please, could you try to add support for `msys=dvbs2x` ?

Barracuda09, 2023-02-28:
What should be supported for these DVB-S2X tuners?

Barracuda09, 2023-02-28:
I have added `msys=dvbs2x` and that will use the SYS_DVBS2

sundtek, 2023-12-30:
you mean SYS_DVBS2X?

Barracuda09, 2024-01-08:
@sundtek 

Yes that should probably use SYS_DVBS2X. 

But probably there should be more added to make this work? I have not tried this myself, is there any test channels?

Barracuda09, 2024-01-08:
@sundtek 

How is this done in SundTek tuner driver for SYS_DVBS2X?

sundtek, 2024-01-08:
We only need SYS_DVBS2X and the symbolrate to be set, the other parameters are automatically detected.

Barracuda09, 2024-01-08:
Is this similar to #160?

But SYS_DVBS2X is still not standard in Linux v6.0

Barracuda09, 2024-01-08:
Here is an patch from Vu+linux kernel patch [patch](http://code.vuplus.com/repository/openvuplus_3.0.git/commit/?id=e79f10463a103a1e9e27e74d084082053d140fb3) that defines it as 21

```
From e79f10463a103a1e9e27e74d084082053d140fb3 Mon Sep 17 00:00:00 2001
From: hschang <chang@dev3>
Date: Wed, 28 Feb 2018 16:04:44 +0900
Subject: Support DVB-S2X for uno4k, ultimo4k, uno4kse.

---
 .../files/linux_3_1x_dvbs2x.patch                  | 81 ++++++++++++++++++++++
 .../linux-libc-headers/linux-libc-headers_3.19.bb  |  6 +-
 .../recipes-kernel/linux/linux-vuplus-3.14.28.inc  |  1 +
 .../linux-vuplus-3.14.28/linux_3_1x_dvbs2x.patch   | 81 ++++++++++++++++++++++
 .../linux/linux-vuplus-vuultimo4k_3.14.28.bb       |  2 +-
 meta-bsp/recipes-vuplus/enigma2/enigma2.bbappend   |  2 -
 meta-openvuplus/recipes-vuplus/enigma2/enigma2.bb  |  2 +-
 7 files changed, 170 insertions(+), 5 deletions(-)
 create mode 100644 meta-bsp/recipes-kernel/linux-libc-headers/files/linux_3_1x_dvbs2x.patch
 create mode 100644 meta-bsp/recipes-kernel/linux/linux-vuplus-3.14.28/linux_3_1x_dvbs2x.patch

diff --git a/meta-bsp/recipes-kernel/linux-libc-headers/files/linux_3_1x_dvbs2x.patch b/meta-bsp/recipes-kernel/linux-libc-headers/files/linux_3_1x_dvbs2x.patch
new file mode 100644
index 0000000..0933dc6
--- /dev/null
+++ b/meta-bsp/recipes-kernel/linux-libc-headers/files/linux_3_1x_dvbs2x.patch
@@ -0,0 +1,81 @@
+diff --git a/drivers/media/dvb-core/dvb_frontend.c b/drivers/media/dvb-core/dvb_frontend.c
+index 1f925e8..57ed889 100644
+--- a/drivers/media/dvb-core/dvb_frontend.c
++++ b/drivers/media/dvb-core/dvb_frontend.c
+@@ -170,6 +170,7 @@ static enum dvbv3_emulation_type dvbv3_type(u32 delivery_system)
+ 		return DVBV3_QAM;
+ 	case SYS_DVBS:
+ 	case SYS_DVBS2:
++	case SYS_DVBS2X:
+ 	case SYS_TURBO:
+ 	case SYS_ISDBS:
+ 	case SYS_DSS:
+@@ -893,6 +894,7 @@ static int dvb_frontend_check_parameters(struct dvb_frontend *fe)
+ 	switch (c->delivery_system) {
+ 	case SYS_DVBS:
+ 	case SYS_DVBS2:
++	case SYS_DVBS2X:
+ 	case SYS_TURBO:
+ 	case SYS_DVBC_ANNEX_A:
+ 	case SYS_DVBC_ANNEX_C:
+@@ -959,6 +961,7 @@ static int dvb_frontend_clear_cache(struct dvb_frontend *fe)
+ 	switch (c->delivery_system) {
+ 	case SYS_DVBS:
+ 	case SYS_DVBS2:
++	case SYS_DVBS2X:
+ 	case SYS_TURBO:
+ 		c->modulation = QPSK;   /* implied for DVB-S in legacy API */
+ 		c->rolloff = ROLLOFF_35;/* implied for DVB-S */
+@@ -2102,6 +2105,7 @@ static int dtv_set_frontend(struct dvb_frontend *fe)
+ 		switch (c->delivery_system) {
+ 		case SYS_DVBS:
+ 		case SYS_DVBS2:
++		case SYS_DVBS2X:
+ 		case SYS_ISDBS:
+ 		case SYS_TURBO:
+ 		case SYS_DVBC_ANNEX_A:
+diff --git a/include/uapi/linux/dvb/frontend.h b/include/uapi/linux/dvb/frontend.h
+index c56d77c..5aebaa9 100644
+--- a/include/uapi/linux/dvb/frontend.h
++++ b/include/uapi/linux/dvb/frontend.h
+@@ -165,6 +165,24 @@ typedef enum fe_code_rate {
+ 	FEC_3_5,
+ 	FEC_9_10,
+ 	FEC_2_5,
[... 173 lines cut ...]
 SRC_URI_append_vuzero4k = " \
 	file://vu_hd_1080p_default.patch \
 	file://enigma2_vuplus_uianimation_speed.patch \
-	file://enigma2_vuplus_dvbs2x.patch \
-	file://enigma2_vuplus_mis_pls.patch \
 	file://enigma2_vuplus_zero4k_blindscan.patch \
 "
 
diff --git a/meta-openvuplus/recipes-vuplus/enigma2/enigma2.bb b/meta-openvuplus/recipes-vuplus/enigma2/enigma2.bb
index 4df7f86..f7b15d2 100644
--- a/meta-openvuplus/recipes-vuplus/enigma2/enigma2.bb
+++ b/meta-openvuplus/recipes-vuplus/enigma2/enigma2.bb
@@ -233,7 +233,7 @@ DEPENDS += "${@base_contains("VUPLUS_FEATURES", "uianimation", "libgles libvugle
 RDEPENDS_${PN}_append_vuplus += "${@base_contains("VUPLUS_FEATURES", "uianimation", "libvugles2" , "", d)}"
 
 PN = "enigma2"
-PR = "r175"
+PR = "r176"
 
 inherit gitpkgv pythonnative
 
-- 
cgit 

```