Barracuda09/SATPI #191: Configuration with two cables on fbc twin tuner

Mofeywalker, 2023-03-15:
Hi @Barracuda09,

maybe you can help me to understand the configuration with 2 cables on the FBC tuner.
I have an VU+ Uno 4k with a FBC Twin Tuner and two cables connected.

How do I have to configure the FBC roots in satpi? Should one of the frontends be connected to A and the other to B? And when I check the "FBC linked" mark, does it consider the FBC-Connection? Should I then connect 3 of the remaining frontends to A and 3 to B?

Thanks for the software :)

Barracuda09, 2023-03-16:
Hi @Mofeywalker 

First thanks for trying SatPI.

Yes the "FBC linked" should consider the 'Link' connection to the specified root tuner. The rest of the tuners can we linked to any of the root tuners. So if you need only 2 tuners of B this is possible just use one root and one linked tuner.

Did you see this wiki topic: https://github.com/Barracuda09/SATPI/wiki/Setup-DVB-S2(X)-FBC-Tuners
If so, is this maybe not clear enough and should I update it?

Mofeywalker, 2023-03-16:
Thanks for getting back :)

So this means that I can split the physical tuners to the frontends e.g. in the following manner

| Frontend type | FBC Connection| Linked |
|----------|----------|--------|
| FBC Root  | A        | no     |
| FBC Root  | B        | no     |
| FBC      | A        | yes    |
| FBC      | A        | yes    |
| FBC      | B        | yes    |
| FBC      | B        | yes    |

And I can then decode multiple transponders in e.g. the Lowband H on the linked A Frontends and multiple transponders in the Highband V on the linked B frontends. A I correct?

Barracuda09, 2023-03-16:
What do you mean with decode exactly? But yes, this setup/config should work for receiving.

Mofeywalker, 2023-03-16:
Yes, sorry I meant receive. Thank you for the Information!

closed 2023-03-16 by Mofeywalker