Barracuda09/SATPI #195: LNB circular enigma2

Dima73, 2023-05-06:
Please help.
What values should be entered?
I entered 
Lof High (MHz) 107500
Lof Low (MHz) 107500
Lof Switch (MHz) 107500
but that's not it.

Barracuda09, 2023-05-07:
Hi @Dima73 

Could you say which brand and type of LNB you are using?

Dima73, 2023-05-07:
> Hi @Dima73
> 
> Could you say which brand and type of LNB you are using?
I could be wrong, but it looks like this.
https://www.inverto.tv/lnb/76/quad-high-band-circular-40mm-lnb-lo1075-ghz

Such LNBs are used for 36 Е/56 E satellites (NTV +/ Tricolor providers).
In enigma2 using this value https://github.com/OpenPLi/enigma2/blob/develop/lib/python/Components/NimManager.py#L55

Barracuda09, 2023-05-07:
Hi @Dima73 

You probably need to use:
Lof High (MHz) 0
Lof Low (MHz) 107500
Lof Switch (MHz) 0

See:
https://github.com/Barracuda09/SATPI/blob/ee0386a0162caa8dd107eb4982f4313f2e4cd5eb/src/input/dvb/delivery/Lnb.cpp#L87-L108

Dima73, 2023-05-07:
OK thanks.
I can only check it tomorrow.

By the way, I added your SATPI project to OpenPLi image.
[SatPI SAT>IP server for linux](https://github.com/OpenPLi/openpli-oe-core/commit/357ea4694298a8c4a5f64ca78715d0c29c024c52)

Dima73, 2023-05-07:
Checked it doesn't work.
Show message:
Service not found!\n(SID not found in PAT)
or
Tune failed!

Dima73, 2023-05-07:
info
https://github.com/catalinii/minisatip/issues/115

Barracuda09, 2023-05-08:
Thanks for testing.

Does the request have in the request:
- pol=r or v for right
- pol=l or h for left

What did you setup in DiSEqC Configuration in Frontend page?

Barracuda09, 2023-05-08:
Currently only pol=v/h is supported from request

https://github.com/Barracuda09/SATPI/blob/ee0386a0162caa8dd107eb4982f4313f2e4cd5eb/src/input/dvb/FrontendData.cpp#L175-L182

Dima73, 2023-05-08:
`What did you setup in DiSEqC Configuration in Frontend page?`
Simple LNB

Barracuda09, 2023-05-08:
Could you and for testing this:  https://github.com/Barracuda09/SATPI/blob/ee0386a0162caa8dd107eb4982f4313f2e4cd5eb/src/input/dvb/FrontendData.cpp#L175-L183  

To This:

```
const std::string pol = params.getParameter("pol"); 
 if (!pol.empty()) { 
 	if (pol == "h" || pol == "l") { 
 		_pol = Lnb::Polarization::Horizontal; 
 	} else if (pol == "v" || pol == "r") { 
 		_pol = Lnb::Polarization::Vertical; 
 	} 
 } 
```

Dima73, 2023-05-08:
I have a real circular LNB.
Real Channel list 36E is polarization L or R.

Dima73, 2023-05-08:
`Could you add for testing this: `
I do not have the ability to compile the satpi binary myself.

Dima73, 2023-05-08:
ARM GigaBlue UHD Quad 4 K

Barracuda09, 2023-05-08:
Here is the ARM build
[satpi.zip](https://github.com/Barracuda09/SATPI/files/11418995/satpi.zip)

Dima73, 2023-05-08:
I can only check in the evening.

By the way, you have defined variables https://github.com/Barracuda09/SATPI/blob/ee0386a0162caa8dd107eb4982f4313f2e4cd5eb/src/input/dvb/delivery/Lnb.h#L40

Barracuda09, 2023-05-08:
Yes I have defined it, but that is correct. But it did not make it in the code.
Because I can not test it myself

Barracuda09, 2023-05-08:
> Here is the ARM version with pol=r/l  Need this? 
> Lof High (MHz) 0
>  Lof Low (MHz) 107500
>  Lof Switch (MHz) 0

Yes that should stay that way

Dima73, 2023-05-08:
I don't understand something wrong...

GigaBlue UHD Quad 4 K use SATPI

VU+ Duo 4 K use satipclient
So I have a physical tuner with satellite 36e (circular LNB) in Duo 4 K.
Disable this tuner and using vtuner(satipclient) and  real channel list 36E.
So you can use (at least for testing)?
Or should I rescan all channels through vtuner(satipclient) ?

Barracuda09, 2023-05-08:
I don't know if you can reuse this channel list for vtuner.

You can always use a "direct" request with (wget or VLC):
```
http://ip.of.your.box:8875/?src=2&fe=2&freq=11052.75&sr=22000&msys=dvbs2&mtype=8psk&pol=h&fec=23&pids=0,1,16,17,18
```
and fill in the correct parameters for your test

Dima73, 2023-05-08:
I checked.
It's kind of chaos.
Only some transponders are scanned.
But in reality, these services are on other frequencies.
Example:
real services frequencies list https://flysat.com/en/package/ntv-plus-13/eutelsat-36b-36c
see: 11938 R

vtuner:
Wrong services for this frequencies(see screenshots
[image]
[image]
[satpi.log](https://github.com/Barracuda09/SATPI/files/11424001/satpi.log)

Channel stops showing after 5 minutes

Dima73, 2023-05-08:
Real frequencies service Матч Премьер 11785 R

Barracuda09, 2023-05-08:
What does this show in the satpi.log and how long does it take to receive data?
```
wget -qO- "http://192.168.1.43:8875/fe=1&freq=11785&sr=27500&msys=dvbs2&mtype=8psk&pol=r&fec=34&pids=0,1,16,17,18"
```

Barracuda09, 2023-05-08:
sorry forgot the _**?**_ please try this 
```
wget -qO- "http://192.168.1.43:8875/?fe=1&freq=11785&sr=27500&msys=dvbs2&mtype=8psk&pol=r&fec=34&pids=0,1,16,17,18"
```

Dima73, 2023-05-08:
??????????????????????
`root@vuduo4k:~# wget -qO- "http://192.168.1.43:8875/?fe=1&freq=11785&sr=27500&msys=dvbs2&mtype=8psk&pol=r&fec=34&pids=0,1,16,17,18"
Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         Gя                                                                                                                                                                                         G@   °m 
б  аeаfаgаhаiаjаkаn аp"аr$аt%аu&аv(аx)аy*аz+а{,а|-а}.а~0аЂ2а‚3аѓKа›UаҐБ©MxяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяG@ °MяяЙ     р         фP  
          чp  
          ш8   `?БяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяG@ PрЋЕиш 
pиPXк«E   )N rus  TC MrusЅѕІѕБВё ДГВ±ѕ»° (0+)Xк«P P >N rus  TC M0rusДА°№±ГАі - »µ№їЖёі (0+)ЗХЬЯШЮЭРв іХаЬРЭШШR3­яяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяG@ PрІяиш 
pиPTЉк«0  SN rus  TK MErusKARATE COMBAT-2022-Н. БЪаШТХаб - ». АЮзР. Ѕ. ПЭзгЪ - ґ. ігбвРТЮT‹к«E   8N rus  T@ M*rusјѕјµЅВЛ Ѕµґµ»ё (16+) - бЯЮавШТЭРп. BO_¶яяG@ Oсйя 
 pOy;кЁ!0 ' 0сMпrusїАµєА°БЅЛµ Бѕ·ґ°ЅёП (12+)Р - ЬХЫЮФаРЬР.  АХЦ.: јРЪбШЬ БгССЮвШЭ.  І аЮЫпx: ОЫШп їЮФЮЧХаЮТР, ІЫРФШбЫРТ ИЪЫпХТ, °ЭРбвРбШп °ЪРвЮТР, ЅШЫ єаЮЯРЫЮТ, ґЬШваШЩG  јгЫпа, °ЫсЭР єЮЧлаХТР, ІШЪвЮа АпСЮТ, °ЫХЪбХЩ єЮвЮТйШЪЮТ, ёТРЭ »РвгиЪЮ. АЮббШп. 2018h®GяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяG@ Pр№зиш 
pиPxБк«% @ ћN rus  TE Mђrusі°Ѕґ±ѕ»ѓЗµјїёѕЅ°В АѕББёё. OLIMPBET БГїµА»ёі°. ¶µЅЙёЅЛ. 1/4 ДёЅ°»°. 2-№ ј°ВЗ. їАПј°П ВА°ЅБ»ПЖёП. ЖБє° - "БвРТаЮЯЮЫмХ" (БвРТаЮЯЮЫмбЪШЩ ЪаРЩG )кЗOЇяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяG@ PтI%Биш 
pиPyoк«    KN rus  T$ M=rusНєѕЅѕјёє°. і»°ІЅѕµ (16+) - РЭРЫШвШзХбЪРп. АЮббШп. 2023ypк«  0 CN rus  T  M5rusЗ°БВЅ°П Бѕ±БВІµЅЅѕБВМ(16+)  - ШЭдЮаЬРжШЮЭЭРп. yqкG «E   IN rus  T  M;rusі»°ІЅЛµ ЅѕІѕБВё (16+)  - ШЭдЮаЬРжШЮЭЭРп. АЮббШп. 2023yrк«  0 =N rus  T  M/rus
ЗН· (16+)  - ШЭдЮаЬРжШЮЭЭРп. АЮббШп. 2023ysк«0   IN rus  T  M;rusG і»°ІЅЛµ ЅѕІѕБВё (16+)  - ШЭдЮаЬРжШЮЭЭРп. АЮббШп. 2023ytк«5    LN rus  T  M>rusАЛЅєё. Бµ№З°Б (16+)% - ШЭдЮаЬРжШЮЭЭРп. АЮббШп. 2021-2023yuк«U  E =N rus  T  M/rus
ЗН· (16+)G@ Jрh Л  р)G
АХУШЮЭРЫмЭлХѓJ­ь6ьD!ь:&ь7щьM)ь5р2  pрAJ­  pрA       !& 
 pрAщ  pрA)В0hNяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяG   - ШЭдЮаЬРжШЮЭЭРп. АЮббШп. 2023…љ…ЈяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяG@ Nр….Г  
[... cut ...]
ЗН· (16+)  - ШЭдЮаЬРжШЮЭЭРп.G  АЮббШп. 2023ЧдяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяG@ NрiГ  
pN:cкЁ@   ђNN rus  TC M@rus"Б°ББГѕ»ѕ" - "±ѕ»ѕЅМП"$ЗХЬЯШЮЭРв ёвРЫШШ. їапЬРп ваРЭбЫпжШп€µ9gяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяяG@ PрЕ3лшш 
pшPDк«"   0 CN rus  T  M5rusјёА БїѕАВ° (6+)  - ШЭдЮаЬРжШЮЭЭРп. АЮббШп. 2023Eк«"P P [N rus  TE MMrus
±°БєµВ±ѕ»>µґёЅ°П »ёі° ІВ±-2022-2023. ї»µ№-ѕДД. ДёЅ°». »ЮЪЮG@ °MяяЙ          р         фP  
          чp  `

Dima73, 2023-05-09:
Maybe workaround....

circularLNB = Lof Switch (MHz) ==107500 and Lof Low (MHz)== 107500 and Lof High (MHz) ==107500

const std::string pol = params.getParameter("pol"); 
 if (!pol.empty()) { 
 	if (pol == "h" && circularLNB) { 
 		_pol = Lnb::Polarization::CircularLeft 

and then change the code ...

Barracuda09, 2023-05-09:
@Dima73 

What is in the RTSP/HTTP request for **pol** parameter: **h/v** or **l/r**

>
>Maybe workaround....
>
>circularLNB = Lof Switch (MHz) ==107500 and Lof Low (MHz)== 107500 and Lof High (MHz) ==107500

This is for having not to only add LO freq for **Log Low (MHz)**: [Setup LNB](https://github.com/Barracuda09/SATPI/wiki/Setup-the-correct-LNB)

Dima73, 2023-05-09:
> Here is the ARM version with pol=r/l [satpi.zip](https://github.com/Barracuda09/SATPI/files/11419175/satpi.zip)
Did I check those changes yesterday?
[Add: LNB Circular Polarization (pol=l/r)](https://github.com/Barracuda09/SATPI/commit/a7c78c3cbda887ff13e7404e131b7e824e30c4cd) https://github.com/Barracuda09/SATPI/issues/195

Barracuda09, 2023-05-09:
Yes, they where in that zip archive.

I asked for the **satpi.log** after the wget test. In this comment see: https://github.com/Barracuda09/SATPI/issues/195#issuecomment-1538900122
To see if the requests are good and see if the freq is correct.

Dima73, 2023-05-09:
Yesterday it was already late and I turned off the PC.
I'll post the log tonight, but it's already clear that this is the wrong frequency.

Barracuda09, 2023-05-09:
Yes that could be, but I really like to see this in the log file.

Barracuda09, 2023-05-09:
I have thought about it some more, but I think the **Lof Low (MHz)** of 107500 is incorrect it should be 10750 MHz.
Because it LO of LNB is 10.75 GHz => 10750 MHz

Dima73, 2023-05-09:
I'll check back in three hours.

Dima73, 2023-05-09:
> I have thought about it some more, but I think the **Lof Low (MHz)** of 107500 is incorrect it should be 10750 MHz. Because the LO of LNB is 10.75 GHz => 10750 MHz
> 
> See [Setup LNB](https://github.com/Barracuda09/SATPI/wiki/Setup-the-correct-LNB)

Actually I used 10750.
I made a mistake in my first post, extra zero at the end.

Dima73, 2023-05-09:
`Tue May  9 21:55:08.6588 2023  6  [                     src/socket/TcpSocket.cpp:109] RTSP Client 192.168.1.108:38682 Connection closed with fd: 8
  Tue May  9 22:05:50.1458 2023  6  [                    src/socket/SocketAttr.cpp:138] HTTP Connection from 192.168.1.108 Port 45743 with fd: 8
  Tue May  9 22:05:50.1462 2023  7  [                          src/HttpcServer.cpp:150] HTTP Stream data from client None with IP 192.168.1.108 on Port 45743: GET /?fe=1&freq=11785&sr=27500&msys=dvbs2&mtype=8psk&pol=r&fec=34&pids=0,1,16,17,18 HTTP/1.1
  Tue May  9 22:05:50.1462 2023  7  Host: 192.168.1.109:8875
  Tue May  9 22:05:50.1462 2023  7  User-Agent: Wget
  Tue May  9 22:05:50.1462 2023  7  Connection: close
  Tue May  9 22:05:50.1462 2023  7  <CRLF>
  Tue May  9 22:05:50.1467 2023  6  [                        src/StreamManager.cpp:193] Found FrondtendID: 1 (fe=1)  StreamID: -1  SessionID: 0265553845
  Tue May  9 22:05:50.1470 2023  6  [                               src/Stream.cpp:210] Frontend: 1, StreamClient[0] with SessionID 0265553845 for dvbs2
  Tue May  9 22:05:50.1472 2023  6  [                   src/input/dvb/Frontend.cpp:410] Frontend: 1, Parsing transport parameters...
  Tue May  9 22:05:50.1473 2023  6  [               src/input/dvb/FrontendData.cpp:121] Frontend: 1, New frequency requested, clearing old channel data...
  Tue May  9 22:05:50.1479 2023  7  [                   src/input/dvb/Frontend.cpp:422] Frontend: 1, Parsing transport parameters (Finished)
  Tue May  9 22:05:50.1481 2023  7  [                               src/Stream.cpp:258] Frontend: 1, Found Streaming type: HTTP
  Tue May  9 22:05:50.1483 2023  6  [                   src/input/dvb/Frontend.cpp:426] Frontend: 1, Updating frontend...
  Tue May  9 22:05:50.1484 2023  6  [                          src/mpegts/Filter.h:137] Frontend: 1, Closing all active PID filters...
  Tue May  9 22:05:50.1990 2023  6  [                   src/input/dvb/Frontend.cpp:787] Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 9 (0 ms)
  Tue May  9 22:05:50.1991 2023  6  [              src/input/dvb/delivery/DVBS.cpp:186] Frontend: 1, Start tuning process for DVB-S(2)...
  Tue May  9 22:05:50.1993 2023  6  [              src/input/dvb/delivery/DVBS.cpp:224] Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Writing DiSEqC command with fd: 9
  Tue May  9 22:05:50.3099 2023  6  [         src/input/dvb/delivery/DiSEqcLnb.cpp:043] Frontend: 1, Sending LNB: Mini-Switch Src: 0
  Tue May  9 22:05:50.5144 2023  7  [              src/input/dvb/delivery/DVBS.cpp:263] Frontend: 1, Set Properties: Frequency 1035000
  Tue May  9 22:05:50.5146 2023  6  [                   src/input/dvb/Frontend.cpp:794] Frontend: 1, Tuned, waiting on lock...
  Tue May  9 22:05:50.9655 2023  6  [                   src/input/dvb/Frontend.cpp:806] Frontend: 1, Tuned and locked (FE status 0x1F)
  Tue May  9 22:05:50.9657 2023  6  [                          src/mpegts/Filter.h:157] Frontend: 1, Updating PID filters...
  Tue May  9 22:05:50.9659 2023  6  [                   src/input/dvb/Frontend.cpp:502] Frontend: 1, Opened /dev/dvb/adapter0/demux0 using fd: 10
  Tue May  9 22:05:50.9664 2023  6  [                   src/input/dvb/Frontend.cpp:508] Frontend: 1, Set DMX buffer size to 3145728 Bytes
  Tue May  9 22:05:50.9667 2023  6  [                   src/input/dvb/Frontend.cpp:524] Frontend: 1, Set DMX_SET_SOURCE with (Src: 0 - Offset: 18)
  Tue May  9 22:05:50.9874 2023  7  [                          src/mpegts/Filter.h:182] Frontend: 1, Set filter PID: 0000
  Tue May  9 22:05:51.0077 2023  7  [                          src/mpegts/Filter.h:182] Frontend: 1, Set filter PID: 0001
  Tue May  9 22:05:51.0280 2023  7  [                          src/mpegts/Filter.h:182] Frontend: 1, Set filter PID: 0016
  Tue May  9 22:05:51.0484 2023  7  [                          src/mpegts/Filter.h:182] Frontend: 1, Set filter PID: 0017
  Tue May  9 22:05:51.0686 2023  7  [                          src/mpegts/Filter.h:182] Frontend: 1, Set filter PID: 0018
  Tue May  9 22:05:51.0691 2023  6  [                   src/input/dvb/Frontend.cpp:447] Frontend: 1, Updating frontend (Finished in 920 ms)
  Tue May  9 22:05:51.0694 2023  6  [              src/output/StreamThreadHttp.cpp:065] Frontend: 1, HTTP set network buffer size: 437 KBytes
  Tue May  9 22:05:51.0697 2023  6  [              src/output/StreamThreadBase.cpp:138] Frontend: 1, Start HTTP stream to 192.168.1.108:45743
  Tue May  9 22:05:51.0699 2023  7  [                          src/HttpcServer.cpp:217] Send reply in 923 ms
  Tue May  9 22:05:51.0699 2023  7  HTTP/1.1 200 OK
  Tue May  9 22:05:51.0699 2023  7  Server: SatPI WebServer v0.1
  Tue May  9 22:05:51.0699 2023  7  Location: 
  Tue May  9 22:05:51.0699 2023  7  CSeq: 0
  Tue May  9 22:05:51.0699 2023  7  cache-control: no-cache
  Tue May  9 22:05:51.0699 2023  7  Content-Type: video/MP2T
  Tue May  9 22:05:51.0699 2023  7  <CRLF>
  Tue May  9 22:05:53.8612 2023  6  [                           src/mpegts/PAT.cpp:064] Frontend: 1, PAT - Section Length: 0109  TID: 13  Version: 0  secNr: 0 lastSecNr: 0  CRC: 0xC1A94D78
  Tue May  9 22:05:53.8644 2023  6  [                           src/mpegts/SDT.cpp:071] Frontend: 1, SDT - Section Length: 1012  Transport Stream ID: 13  Version: 15  secNr: 0  lastSecNr: 0  NetworkID: 0112  CRC: 0x5D0B9763
  Tue May  9 22:05:53.8664 2023  6  [                           src/mpegts/NIT.cpp:194] Frontend: 1, NIT - Section Length: 1008  NID: 0112  Version: 2  secNr: 0 lastSecNr: 1  CRC: 0x1341EDDB
  Tue May  9 22:05:53.8681 2023  6  [                           src/mpegts/NIT.cpp:194] Frontend: 1, NIT - Section Length: 0108  NID: 0112  Version: 2  secNr: 1 lastSecNr: 1  CRC: 0xEEF900D1
`

Dima73, 2023-05-09:
Strange
src/HttpcServer.cpp:150] RTSP Stream data from client None with IP 192.168.1.108 on Port 41100: PLAY rtsp://192.168.1.109:554/stream=100?src=1&freq=12191&pol=h&msys=dvbs&sr=27500&fec=34&ro=0.35&mtype=qpsk&plts=off RTSP/1.0

really zap to service 12341L 27500
[image]
Maybe satipclient set wrong value?

Barracuda09, 2023-05-09:
Could you try again with wget like this (And let it run):
```
wget -qO- "http://192.168.1.109:8875/?fe=1&freq=11785&sr=27500&msys=dvbs2&mtype=8psk&pol=r&fec=34&pids=0,1,16,17,18"  > /dev/null
```

during this could you get xml file like:
```
http://192.168.1.109:8875/SatPI.xml
```

Then stop wget again and please post result.

Dima73, 2023-05-09:
After stop wget
[3.log](https://github.com/Barracuda09/SATPI/files/11435496/3.log)

Barracuda09, 2023-05-09:
This seems to be correct for freq 11785
```
<transportStreamID>13</transportStreamID>
<networkID>112</networkID>
```
[image]

Dima73, 2023-05-10:
satipclient only pol h/v
https://github.com/oe-alliance/satip-client/blob/a12648d40698efe77f0ba5fa73e9e70f90c6d591/config.cpp#L304

Saentist, 2023-05-10:
Physically there is a possible to use linear LNB to receive Circular polarization,
without software changes.
DIY can be done with "dielectric plate"
https://olegkutkov.me/2020/09/20/reworking-linear-polarization-satellite-lnb-into-a-circular-polarized/

Dima73, 2023-05-10:
> Physically there is a possible to use linear LNB to receive Circular polarization, without software changes. DIY can be done with "dielectric plate" https://olegkutkov.me/2020/09/20/reworking-linear-polarization-satellite-lnb-into-a-circular-polarized/

I know it, but it's but it's not to do with it.

Barracuda09, 2023-05-10:
@Dima73 
I don't think the pol=h/v is the problem as h=l and v=r. As long you take that into account it should work.

The freq set in [vtuner.cpp#L280](https://github.com/oe-alliance/satip-client/blob/a12648d40698efe77f0ba5fa73e9e70f90c6d591/vtuner.cpp#L280) is somethink i do not know what it does, but it has some LNB calculations so it seems.

Barracuda09, 2023-05-10:
@Dima73 
What LNB did you setup in Enigma for the SAT>IP frontends?

Dima73, 2023-05-10:
> @Dima73 What LNB did you setup in Enigma for the SAT>IP frontends?
I tested both options:
When "Использовать круговой LNB"-->yes -->Circular LNB
When "Использовать круговой LNB"-->no -->Universal LNB
Here is a screenshot, here the value is yes
 https://user-images.githubusercontent.com/1756274/236901818-4e360052-0eb2-45c6-85f0-106ce22fe5e6.jpg

Barracuda09, 2023-05-10:
@Dima73 
I would expect if you use universal LNB in enigma it should cancel out the calculations in satipclient

Dima73, 2023-05-10:
there is always a calculation https://github.com/oe-alliance/satip-client/blob/a12648d40698efe77f0ba5fa73e9e70f90c6d591/vtuner.cpp#L272
By the way, is it getting data from the server?

Barracuda09, 2023-05-10:
@Dima73 

Yes it always does the calculation but it adds the LOF.
So Enigma is subtracts  the LOF to get the tune freq. then the vtuner adds it again to send the channel freq via SAT>IP protocol.
Then the SatPI subtracts the circular LOF to get the tune freq.

Yes the test here https://github.com/Barracuda09/SATPI/issues/195#issuecomment-1538924497 shows there is data.
And here https://github.com/Barracuda09/SATPI/issues/195#issuecomment-1540788077 show it tunes to the requested channel correctly.

Dima73, 2023-05-10:
`So my guess would be: use in Enigma the Universal LNB and that would do the trick.`
I tested it yesterday and it's only one channel out of dozens that worked.
I couldn't repeat this again.

Dima73, 2023-05-10:
Ok.
Use as universal LNB.
Finally, it somehow worked, I checked the switching to several channels, at least there is some kind of result.

The next problem, although this probably does not apply to satpi.
Right now I'm using a real 36E channel list.
But in the future I won't have a satellite dish.
Service scanning does not work through satipclient.
I have checked several times now.
Could this be a problem with the circular LNB?
Do I understand correctly that scanning and creating a channel list should work through satipclient when using the original universal LNB (I myself can not check)?

Maybe it's still worth adding support for the circular LNB in satipclient?
If possible...

Dima73, 2023-05-10:
I'm sorry.
The scan seems to be working.

Dima73, 2023-05-12:
[Barracuda09](https://github.com/Barracuda09)
Two days of testing everything is ok.
Thank you very much.
The topic can be closed.

Barracuda09, 2023-05-12:
@Dima73 

Thanks for your efforts and implementation in @OpenPLi

As requested this will be closed

closed 2023-05-12 by Barracuda09