Barracuda09/SATPI #197: DVB-C/DVB-S/DVB-T simultaneously

Dima73, 2023-06-26:
Is it possible?
Example /etc/vtuner.conf:
`0=desc:SatPI Server (192.168.1.90),force_plts:1,ipaddr:192.168.1.90,tcpdata:1,tuner_type:DVB-C,uuid:50c958a8-e839-4b96-b7ae-00179a271630,vtuner_type:satip_client
1=desc:SatPi Server (192.168.1.90),force_plts:1,ipaddr:192.168.1.90,tcpdata:1,tuner_type:DVB-S,uuid:50c958a8-e839-4b96-b7ae-00179a271630,vtuner_type:satip_client`

Barracuda09, 2023-06-26:
Could you explain what this **`1=desc:........`** etc means and how it is used to make a RTSP request:
```
rtsp://192.168.1.90/?fe=3&freq=385&sr=6900&msys=dvbc&mtype=256qam&fec=35&pids=0,1,16,17,18
```

Barracuda09, 2023-06-26:
@Dima73 

I do not get the thread, over at OpenPLI, were this questions comes from.
Without some information from the people involved, I can not give support to this question.

Dima73, 2023-06-26:
I understand.
Now they know who to contact.
I will close this question.

closed 2023-06-26 by Dima73