Barracuda09/SATPI #201: Freezing when watching encrypted channels

sekinooo, 2023-11-30:
Hi, I have issue watching encrypted channels, 
Every time ECM is updated, picture will freeze for half of a second. Anyone can help me here what could be wrong?
FTA channels are running well.

Server: VU+ SOLO 4K, Open ATV 7.3, FBC tuners
Client, VU+ ZERO 4K, Open ATV 7.3, EMU: CCcam, ECM < 100ms
LAN: 1 GBps
SATPI Version | 1.6.2.161~ga9aab2c
CC Errors | 4789

------------LOG----
>   Thu Nov 30 19:37:33.5202 2023  6  Found FrondtendID: 3 (fe=3)  StreamID: 102  SessionID: 0259402104
>   Thu Nov 30 19:37:33.5203 2023  6  Frontend: 3, StreamClient with SessionID 0259402104
>   Thu Nov 30 19:37:33.5203 2023  6  Frontend: 3, Parsing transport parameters...
>   Thu Nov 30 19:37:33.5204 2023  7  Frontend: 3, Parsing PID parameter: delpids=2960,2910,2920,0,2900,2921,2950
>   Thu Nov 30 19:37:33.5204 2023  7  Frontend: 3, Parsing transport parameters (Finished)
>   Thu Nov 30 19:37:33.5205 2023  6  Frontend: 3, Updating frontend...
>   Thu Nov 30 19:37:33.5205 2023  6  Frontend: 3, Updating PID filters...
>   Thu Nov 30 19:37:33.5408 2023  7  Frontend: 3, Remove filter PID: 2900 - Packet Count: 000000295:000004 - PMT
>   Thu Nov 30 19:37:33.5610 2023  7  Frontend: 3, Remove filter PID: 2910 - Packet Count: 000156848:000031
>   Thu Nov 30 19:37:33.5812 2023  7  Frontend: 3, Remove filter PID: 2920 - Packet Count: 000005055:000044
>   Thu Nov 30 19:37:33.6013 2023  7  Frontend: 3, Remove filter PID: 2921 - Packet Count: 000004910:000010
>   Thu Nov 30 19:37:33.6215 2023  7  Frontend: 3, Remove filter PID: 2950 - Packet Count: 000000097:000000
>   Thu Nov 30 19:37:33.6417 2023  7  Frontend: 3, Remove filter PID: 2960 - Packet Count: 000000042:000002
>   Thu Nov 30 19:37:33.6419 2023  6  Frontend: 3, Updating frontend (Finished in 121 ms)
>   Thu Nov 30 19:37:33.6420 2023  7  Send reply in 121 ms
>   Thu Nov 30 19:37:33.6420 2023  7  RTSP/1.0 200 OK
>   Thu Nov 30 19:37:33.6420 2023  7  Server: satpi/1.6.2.161-ga9aab2c
>   Thu Nov 30 19:37:33.6420 2023  7  RTP-Info: url=rtsp://192.168.1.23/stream=102
>   Thu Nov 30 19:37:33.6420 2023  7  CSeq: 46
>   Thu Nov 30 19:37:33.6420 2023  7  Session: 0259402104
>   Thu Nov 30 19:37:33.6420 2023  7  Range: npt=0.000-
>   Thu Nov 30 19:37:33.6420 2023  7  <CRLF>
>   Thu Nov 30 19:37:35.4949 2023  7  RTSP Stream data from client None with IP 192.168.1.24 on Port 51878: PLAY rtsp://192.168.1.23:554/stream=102?delpids=2941 RTSP/1.0
>   Thu Nov 30 19:37:35.4949 2023  7  CSeq: 47
>   Thu Nov 30 19:37:35.4949 2023  7  Session: 0259402104
>   Thu Nov 30 19:37:35.4949 2023  7  <CRLF>
>   Thu Nov 30 19:37:35.4950 2023  6  Found FrondtendID: 3 (fe=3)  StreamID: 102  SessionID: 0259402104
>   Thu Nov 30 19:37:35.4951 2023  6  Frontend: 3, StreamClient with SessionID 0259402104
>   Thu Nov 30 19:37:35.4951 2023  6  Frontend: 3, Parsing transport parameters...
>   Thu Nov 30 19:37:35.4952 2023  7  Frontend: 3, Parsing PID parameter: delpids=2941
>   Thu Nov 30 19:37:35.4952 2023  7  Frontend: 3, Parsing transport parameters (Finished)
>   Thu Nov 30 19:37:35.4953 2023  6  Frontend: 3, Updating frontend...
>   Thu Nov 30 19:37:35.4953 2023  6  Frontend: 3, Updating PID filters...
>   Thu Nov 30 19:37:35.5157 2023  7  Frontend: 3, Remove filter PID: 2941 - Packet Count: 000000019:000000
>   Thu Nov 30 19:37:35.5159 2023  6  Frontend: 3, Updating frontend (Finished in 20 ms)
>   Thu Nov 30 19:37:35.5159 2023  7  Send reply in 21 ms
>   Thu Nov 30 19:37:35.5159 2023  7  RTSP/1.0 200 OK
>   Thu Nov 30 19:37:35.5159 2023  7  Server: satpi/1.6.2.161~ga9aab2c
>   Thu Nov 30 19:37:35.5159 2023  7  RTP-Info: url=rtsp://192.168.1.23/stream=102
>   Thu Nov 30 19:37:35.5159 2023  7  CSeq: 47
>   Thu Nov 30 19:37:35.5159 2023  7  Session: 0259402104
>   Thu Nov 30 19:37:35.5159 2023  7  Range: npt=0.000-
>   Thu Nov 30 19:37:35.5159 2023  7  <CRLF>
>   Thu Nov 30 19:37:36.4949 2023  7  RTSP Stream data from client None with IP 192.168.1.24 on Port 51878: PLAY rtsp://192.168.1.23:554/stream=102?delpids=18 RTSP/1.0
>   Thu Nov 30 19:37:36.4949 2023  7  CSeq: 48
>   Thu Nov 30 19:37:36.4949 2023  7  Session: 0259402104
>   Thu Nov 30 19:37:36.4949 2023  7  <CRLF>Frontend Info

---Frontend Info--
Stream Index	3
Enable	
Attached	no
Type	DVB-S(2) FBC (Slot A)
Name	Vuplus FE
Path	/dev/dvb/adapter0/frontend2
Freq Range	950000 Hz to 2150000 Hz
Symbol Rate Range	1000000 symbols/s to 45000000 symbols/s
Linux DVB API version	0x050A
Stream Info
Stream URL	[URL Link](http://192.168.1.23:8875/frontend.html)
Channel Info
Network Name	Not Found
Channel Name	Not Found
Delivery System	UNKNOWN DELSYS
Tune Frequency (kHz)	0
Modulation	
Fec	
Symbol Rate	0
Monitor Info
Status	0
Signal Strength	0
snr	0
ber	0
unc	0
PID	
CC Errors	4789
Configuration
DVR Buffer (MB)	
5
RTCP Signal Update Freq	
1
Filter PCR for timing	
Wait On Tuning Lock Timeout (ms)	
1000
Turn off LNB Voltage during teardown	
Enable slightly higher LNB Voltage	
List of PIDs to add to requests (CSV)	
0,1,16,17,18
Transformation Info
Transformation enable	
Advertise as	
DVB-S2
Transformation Mapping M3U	
mapping.m3u
Transformation Frequency	0
FBC Configuration
FBC Connection	
A
FBC Linked	
Send DiSEqC Via Linked Root Tuner	
DiSEqC Configuration
DiSEqC	
Unicable (EN50494)
Repeat DiSEqC	
0
Channel Freq (MHz)	
1680
Channel Slot (0-32)	
2
Delay before write	
10
Delay after write	
15
PIN (256 disabled)	
256
LNB Configuration
LNB 1
Lof High (MHz)	
10600
Lof Low (MHz)	
9750
Lof Switch (MHz)	
11700

Barracuda09, 2023-12-01:
Hi @sekinooo 

The decoding takes place in the Client: VU+ ZERO 4K

sekinooo, 2023-12-01:
Hi @Barracuda09 

I'm aware that decoding is taking place on client side. Do you know what can cause the freezing? 
ECM is recieved under 100ms, so this is not the issue with latency, I think...
When physically connecting Client instead of server, local decoding is working without freezing.

Any idea what can cause the freezing when decoding on client side?

Barracuda09, 2023-12-01:
Hi @sekinooo 

I do not have a idea yet, but a few questions:
- But the PIDs are not changing in between?
- Could you try to decode it in SatPI with OSCam?

sekinooo, 2023-12-01:
Hi @Barracuda09

So using OScam, didn't solve the issue.

> But the PIDs are not changing in between?

Where can I find if PIDS are cahnging? In SATPI log? Or Frontend Monitor?

> Could you try to decode it in SatPI with OSCam?

I have empty tab in SATPI > OScam, I donwloaded version from openATV feed. I need another version?

Barracuda09, 2023-12-01:
Try replacing the current bin with this one: [satpi_arm_1.6.2.161~ga9aab2c_Enigma.zip](https://github.com/Barracuda09/SATPI/files/13524497/satpi_arm_1.6.2.161.ga9aab2c_Enigma.zip)

sekinooo, 2023-12-01:
Hi @Barracuda09 

running OScam directly in SATPI solve the issue with freezing!!! Hooray 👍 

but 2 new issues arised.

1. OScam on Server box, is no longer decoding channels on local box while viewing, due to change in DVBAPI. Can I have 2 DVBAPIs? Or how to solve that?

2. OScam must be turned off on Client box, while watching channels from SATPI, but I need OScam to watch channels on local box from other sat. position. How to block OScam on Client to not decoding channels from SATPI (other sat position), but decode from other satellite pos. ?

Barracuda09, 2023-12-01:
Answers:
1. You can try to set in (SatPI) the **OSCam Configure Page** option **OSCam Aadapter offset** to 10
2. i do not exactly what you mean with this. So the decoded Channels from SatPI still go through the decoding on client side?

Barracuda09, 2023-12-01:
As far as I know you can not add an extra entry (or connection), you should be able to do the same with the offset in SatPI

sekinooo, 2023-12-01:
OK, so I have 1 DVBAPI. I can choose to feed OSCAM in SATPI or in SERVER box, what should I do to have both things working same time? Should I remove this from Oscam.conf?
`[dvbapi]
enabled = 1
au = 1
pmt_mode = 4
listen_port = 15012
user = TEST
read_sdt = 2
boxtype = pc`
 and put this?
`[dvbapi]
enabled = 1
user = auto
au = 1
pmt_mode = 0
request_mode = 0
boxtype = dreambox`

and both will be working?

sekinooo, 2023-12-01:
When I choose in OScam.conf `boxtype = dreambox` local box is decoding local channels, SATPI will NOT connect to OScam.
When I choose in OScam.conf `boxtype = pc` local box is NOT decoding local channels, SATPI will connect to OScam.

I think `OSCam Aadapter offset to 10` has no effect here when mentioned situation above :(
Can you make SATPI version to work with `boxtype = dreambox` ?

Maybe I will need to ask on OScam board to help solve this issue with DVBAPI ...

Barracuda09, 2023-12-01:
Or you(we) should try to figure out why during client decode the picture freezes?

Saentist, 2023-12-02:
This is reconnect, do you use WIFI?

sekinooo, 2023-12-03:
NO, LAN cable, 1GBps switch in between. Reconnect is doing every 30 - 60s. Sometimes after 3-4 minutes SATPI stops streaming completly and I need to manually switch channel to reconect.

Barracuda09, 2023-12-03:
Do you have the web interface of SatPI constantly open? and maybe on more devices?

sekinooo, 2023-12-03:
I have it opened on PC, to see the log...

Barracuda09, 2023-12-03:
It is for me a bit difficult to debug, because I do not observe this problem.

 - _I'm Using Frontend 3, disabled 1 and 2, does it have any impact?_ -> no that should have no impact
 - _What is doing DVR Buffer (MB) in Frontend settings?_ -> that is how much the 'driver' should allocate for DVR buffer, you could increment this for testing

sekinooo, 2023-12-03:
I made some more tests. LAN capacity is not an issue here, I can stream FTA 4k channel from other receiver without any issue.

_SATPI Streaming FTA channels_ - no freezes at all, working flawless!
SATPI decoding - _OScam connected to local OScam server_ - no freezes at all, working flawless!
SATPI decoding - _OScam connected to Client's OScam server_ - 99% working without issue, I would consider this pretty same as streaming local OScam running on server. There is reconnection every few minutes which doesn't affect picture, no freeze.
_Decoding via Client box_ - OScam/CCcam **everytime** when ECM is updated there is reconnection in SATPI log = picture freeze for half of a second.

Barracuda09, 2023-12-03:
If decoding via client box is not working reliable when ECM changes, this makes me think that this is probably an issue in the client.

- Client and Sever are on the same internal network? or is there an internet connection between client OSCam etc.
- Which client do you use? satipclient? 

SatPI is just passing the requested PIDs to the client without interruption when client handles the decoding.
When SatPI is decoding, only a few requested (by OSCam) packets are send to OSCam, so no big bandwidth is needed.

Barracuda09, 2023-12-04:
Could you share which channel (frequency) and satellite this is?

sekinooo, 2023-12-04:
Astra 23.5, https://en.kingofsat.net/find.php?question=jednotka

sekinooo, 2023-12-04:
OK, one more question, should I enable local client decoding, or server decoding? Or it doesn't matter. I set it on server decoding, cause local decoding is not usable...

sekinooo, 2023-12-04:
`This XML file does not appear to have any style information associated with it. The document tree is shown below.
<data>
<appdata>
<uptime>54</uptime>
<appversion>1.6.2.161~ga9aab2c Enigma</appversion>
<uuid>53617450-495F-5365-7276-001dec0d2d9e</uuid>
</appdata>
<streams>
<stream1>
<streamindex>1</streamindex>
<enable>
<inputtype>checkbox</inputtype>
<value>false</value>
</enable>
<attached>no</attached>
<rtcpSignalUpdate>
<inputtype>number</inputtype>
<value>1</value>
<minvalue>1</minvalue>
<maxvalue>5</maxvalue>
</rtcpSignalUpdate>
<frontendname>Vuplus FE</frontendname>
<pathname>/dev/dvb/adapter0/frontend0</pathname>
<freq>950000 Hz to 2150000 Hz</freq>
<symbol>1000000 symbols/s to 45000000 symbols/s</symbol>
<dvbversion>0x050A</dvbversion>
<dvrbuffer>
<inputtype>number</inputtype>
<value>10</value>
<minvalue>0</minvalue>
<maxvalue>30</maxvalue>
</dvrbuffer>
<waitOnLockTimeout>
<inputtype>number</inputtype>
<value>1000</value>
<minvalue>0</minvalue>
<maxvalue>3500</maxvalue>
</waitOnLockTimeout>
<dvbcsa_bs_batch_size>
<inputtype>number</inputtype>
<value>32</value>
<minvalue>0</minvalue>
<maxvalue>32</maxvalue>
</dvbcsa_bs_batch_size>
<status>0</status>
<signal>0</signal>
<snr>0</snr>
<ber>0</ber>
<unc>0</unc>
<filter>
<pidcsv/>
<totalCCErrors>0</totalCCErrors>
<filterPCR>
<inputtype>checkbox</inputtype>
<value>false</value>
</filterPCR>
<addUserPids>
<inputtype>text</inputtype>
<value>0,1,16,17,18</value>
</addUserPids>
<channelname>Not Found</channelname>
<networkname>Not Found</networkname>
<pat/>
<pmtlist/>
<sdt>
<transportStreamID>0</transportStreamID>
<networkID>0</networkID>
</sdt>
<nit>
<networkID>0000</networkID>
[... 1777 lines cut ...]
<inputtype>text</inputtype>
<value>/usr/bin</value>
</appDataPath>
<syslog>
<inputtype>checkbox</inputtype>
<value>false</value>
</syslog>
<logDebug>
<inputtype>checkbox</inputtype>
<value>true</value>
</logDebug>
</configdata>
<ssdp>
<annouceTime>
<inputtype>number</inputtype>
<value>60</value>
<minvalue>0</minvalue>
<maxvalue>1800</maxvalue>
</annouceTime>
<bootID>32</bootID>
<deviceID>2</deviceID>
<ttl>1</ttl>
<satipserver0>192.168.1.24 [SatPI/1.6.2.161~ga9aab2c]</satipserver0>
</ssdp>
</data>`

Barracuda09, 2023-12-06:
@sekinooo 
here is a new satipclient with extended timeouts: [satipclient_ARM_ext_timeouts.zip](https://github.com/Barracuda09/SATPI/files/13579870/satipclient_ARM_ext_timeouts.zip)

Barracuda09, 2023-12-06:
@sekinooo
here is a new satipclient with extended timeouts and buffers: [satipclient_ARM_ext_timeouts_and_buffers.zip (Stripped) ](https://github.com/Barracuda09/SATPI/files/13582824/satipclient_ARM_ext_timeouts_and_buffers.zip)