Barracuda09/SATPI #203: vuduo4kse DVB-C not working

alturismo, 2023-12-16:
Hi,

may a hint what im doing wrong, following the guides and actually im stuck ;)

base, vuduo4kse with dvb-c / dvb-s setup, tvheadend as satip client

dvb-s /unicable) tuners seems working so far
only it looks like tvheadend has some small issues with micro stuttering with streams coming from the vu compared to the dvb-s card builded into the pc, wanted to try with dvbapi decryption, but looks like i also need a modified libdvbca.so ... abother time therefore ;)

major issue now, dvb-c wont work at all
- tuners are detected
- tuners are disabled in the vu (also tried while they are enabled)

[image]

- no changes where done besides ticking enable in satpi

[image]

- i tested (i guess all) combinations in tvheadend tuner setup

[image]

- while tuning -C with satpi, i get no bandwith coming in

[image]

may i missed something ...

attached also a log where i actually dont get smarter ;)

[log_satpi.txt](https://github.com/Barracuda09/SATPI/files/13691510/log_satpi.txt)

for any Tipps, thanks ahead

and of course, -C Tuners (1-4) are working fine directly on the VU, so reception is ok

Barracuda09, 2023-12-16:
Hi @alturismo 

First of all thanks for giving this a go!.

A few questions and things to do:

- could you please enable these two options and zip and send syslog (probably /var/log/messages and messages.0)
 and do some things you try to do to get it working.
[image]

- What do you mean with this?
  > only it looks like tvheadend has some small issues with micro stuttering with streams coming from the vu

- From the log, seems that the tuners are locking to some signal (time 06:52:03.4460). 
 Was this during scanning?

```
  Sat Dec 16 06:52:02.9448 2023  6  Frontend: 13, StreamClient with SessionID 0241484422 for dvbc
  Sat Dec 16 06:52:02.9449 2023  6  Frontend: 13, Parsing transport parameters...
  Sat Dec 16 06:52:02.9450 2023  6  Frontend: 13, New frequency requested (0.0000 -> 330.0000), clearing old channel data...
  Sat Dec 16 06:52:02.9452 2023  6  Frontend: 13, Updating frontend...
  Sat Dec 16 06:52:02.9452 2023  6  Frontend: 13, Closing all active PID filters...
  Sat Dec 16 06:52:02.9956 2023  6  Frontend: 13, Opened /dev/dvb/adapter0/frontend12 for Read/Write with fd: 9 (0 ms)
  Sat Dec 16 06:52:02.9956 2023  6  Frontend: 13, Start tuning process for DVB-C(2)...
  Sat Dec 16 06:52:02.9957 2023  6  Frontend: 13, Tuned, waiting on lock...
  Sat Dec 16 06:52:03.4460 2023  6  Frontend: 13, Tuned and locked (FE status 0x1F)  <*****************
  Sat Dec 16 06:52:03.4461 2023  6  Frontend: 13, Updating PID filters...
  Sat Dec 16 06:52:03.4462 2023  6  Frontend: 13, Opened /dev/dvb/adapter0/demux12 using fd: 10
  Sat Dec 16 06:52:03.4464 2023  6  Frontend: 13, Set DMX buffer size to 3145728 Bytes
  Sat Dec 16 06:52:03.4466 2023  6  Frontend: 13, Set DMX_SET_SOURCE with (Src: 12 - Offset: 32)
  Sat Dec 16 06:52:03.5685 2023  6  Frontend: 13, Updating frontend (Finished in 623 ms)
  Sat Dec 16 06:52:03.5685 2023  6  Frontend: 13, RTP/UDP set network buffer size: 160 KBytes
  Sat Dec 16 06:52:03.5686 2023  6  Frontend: 13, Start RTP/UDP stream to 192.168.1.80:51712
  Sat Dec 16 06:52:03.5686 2023  6  Frontend: 13, Start RTCP/UDP stream to 192.168.1.80:51713

```

alturismo, 2023-12-16:
> * What do you mean with this?
>   >

i see when using this solution some small "micro stuttering" on decoding channels, this compared to using the same channels from the same signal (satellite) on the same server (tvheadend) but with the DD pcie TV Card inside the Server instead the SatIP Stream coming from VU+.

But thats a different story and i ll debug this deeper once DVB-C is up and running, there i can compare to my other SatIP (Fritz) Solution which run flawlessly, so i can then rather say its a SatIP issue or not ...

no prio currently ;) but i ll probably return on this ;)

> * Was this during scanning?

nope, nothing to scan here, TVHeadend is already running with all channels, as you may see in my upper Post i already have DVB-C running (SatIP) from a fritz Tuner (4x), and DVB-S is also already all settled and running from the DVB Card (DD Max S8).

> * could you please enable these two options and zip and send syslog (probably /var/log/messages and messages.0)
>   and do some things you try to do to get it working.

ok, log attached

[messages.log](https://github.com/Barracuda09/SATPI/files/13692724/messages.log)

alturismo, 2023-12-16:
> First of all thanks for giving this a go!.

and forgot, thanks to you for providing this ;) only reason why i switched from VTi to OATV ...

and i just need some time to compile me a working libdvbca.so ... so i could try the oscam solution against the "micro stuttering" i mentioned above, but that will be another story / thread ;)

Barracuda09, 2023-12-16:
You probably do not need libdvbca, you need an SatPI with OSCam support compiled in. That is not standard in OpenATV or OpenPLi

Which SatPI version is current in this distro?

alturismo, 2023-12-16:
> Which SatPI version is current in this distro?

root@vuduo4kse:~# satpi --version
SatPI version: 1.6.2.161~ga9aab2c

but i also tested with another version here from another thread, i think you provided it for debugging purposes

root@vuduo4kse:~# /usr/bin/satpi_debug_oscam --version
SatPI version: 1.6.2.161~ga9aab2c Enigma

on this version i can also configure oscam, i see its connecting, i see it tries to descramble, but after a few seconds it quits, so my thought was as i readed 

[image]

prolly some modified one ...

Barracuda09, 2023-12-16:
No the libdvbca from OpenATV should work.. I expect.

Does OSCam quit giving reaction?

Barracuda09, 2023-12-16:
Could you give me a syslog of the DVB-S included?

alturismo, 2023-12-16:
> Could you give me a syslog of the DVB-S included?

attached, and here some screens, looking all ok ... but screen stays black ;)

here from oscam log

[image]

also connection looking fine

[image]

tvheadend looking ok (besides the small contin errors ... but running now since a few minutes)

[image]

also frontend is ok

[image]

but it stays "black" ;)

[image]

attached the message from this testrun

[messages2.log](https://github.com/Barracuda09/SATPI/files/13693635/messages2.log)

Barracuda09, 2023-12-16:
But FTA dvb-s(2) are working?

alturismo, 2023-12-16:
> But FTA dvb-s(2) are working? Edit: Because it is getting data from the frontends.. But the frontends for DVB-C seems not to get any data

yes ;)

also working fine when i let the descrambling done in tvheadend (dvbapi) instead the satpi integration using the satpi tuners

[image]

[image]

[image]

so yes, Tuners are working (DVB-S), besides the "micro stutters" from time to time ... but not painful and i investigate this further once DVB-C is running ... ;)

about DVB-C ... well, i can say the VU is working with DVB-C ;) just as sample now ...

[image]
[image]

Barracuda09, 2023-12-16:
Yes I mean, that SatPI does not read any data. There is data (probably) because the tuner is locked. But probably there is a bug in SatPI how it handles DMX_SET_SOURCE?

Barracuda09, 2023-12-16:
Could please try this in VLC (Open Network Stream):
```
http://192.168.1.13:8875/?freq=450&sr=6900&msys=dvbc&mtype=256qam&fec=35&specinv=1&pids=0,21http,6100,6110,6120,6121,6123
```

alturismo, 2023-12-17:
> Could you please try this in VLC (Open Network Stream):

i can open it, it stays open, but no pic ...

[image]

[image]

it should be one of these tuners ...

[image]

alturismo, 2023-12-17:
> Could you attach SatPI.xml here? just do:

attached (had to zip it, xml not supported ;))

[SatPI.zip](https://github.com/Barracuda09/SATPI/files/13694982/SatPI.zip)

Barracuda09, 2023-12-17:
Hi @alturismo ,

Thanks for you input, and I think I know the problem(bug), thanks to your SatPI.xml.

The DVB-C tuners do not show to which Root tuner they are connected to, so they are not properly configured.
I will try to fix this as soon as possible

alturismo, 2023-12-17:
> Thanks for you input, and I think I know the problem(bug), thanks to your SatPI.xml.

thanks for looking into it, exited to see if its an "easy" fix ;)

> The DVB-C tuners do not show to which Root tuner they are connected to, so they are not properly configured.

funny, once i looked into it to check if there is something i could set ... ;)

in the end, shell i open a seperate thread about the oscam dvbapi issue or wait until the next release.

btw. i also tried to start satpi like this (after copy the 2 files to /home/root/

LD_LIBRARY_PATH=/home/root/ /usr/bin/satpi

while there is now

[image]

Barracuda09, 2023-12-17:
Could you try this test version. Just replace it with the current one
[satpi_ARM_1.6.2.161~TEST_Enigma.zip](https://github.com/Barracuda09/SATPI/files/13697668/satpi_ARM_1.6.2.161.TEST_Enigma.zip)

alturismo, 2023-12-17:
> Could you try this test version. Just replace it with the current one

thanks alot for the testbuild, made a quick "run through", sadly same result, tested with A and B config, tested while tuners are activated /deactivated in VU Box, also tested with FBC linked tticked, unticked.

attached log and SatPI.xml

[SatPI.zip](https://github.com/Barracuda09/SATPI/files/13697820/SatPI.zip)
[messages3.log](https://github.com/Barracuda09/SATPI/files/13697821/messages3.log)

alturismo, 2023-12-18:
> Could you try this test version. Just replace it with the current one

ok, i assume im 2 steps further now ...

1/ DVB-C is working, but ... ;) only if i start enabling from Tuner 1 up ... which then breaks the local playback while in use

[image]

i remembered, when i 1st started up satpi i had dvb-c working but no dvb-s, then i readed about disabling tuners in the box so satpi can access them ... thats why i disabled tuner 1-4 (-S && -C), -S worked then, -C stopped working then ...

but, this setup will then stop "dual" usage, stb nativ and satip ... either or ... either satpi uses the tuner 1+ ... or the stb is using the tuner 1+ ... ok, could be "workaround" by using the satpi as "leading" tuners also for local playback.
the 1st "wins" ... either local stb playback is using dvb-c, then satpi stays like shown above, or satpi takes tuner 1+ and local is dead.

2/ oscam dvbapi, partly working and no smooth playback

- partly, either icam is natively not supported in the oatv image libdvbcsa as i never manage to decode such a channel, or see next part and may my "conclusion"

- no smooth playback, if oscam is working, i see alot of stuttering, pic errors, ... like a (really) bad signal ... and as 1+ tuners are active ... its worse and worse ... so may back to "partly", could be like icam channels are just way too slow and there is no picture coming up here ... as the "signal" is too weak, or just really not supported ;)
- also, max 1 decode seems supported, as soon i start a 2nd stream the 1st stops to decode anyhow

---
back to the DVB-C "issue", so it was working before (not knowing the root Tuner) but needs a total control and all tuners binded to satpi (here at least).

Barracuda09, 2023-12-18:
Could you send the syslog of this situation?

Barracuda09, 2023-12-18:
Ok now I understand you use ICAM? Did you compile libdvbcsa yourself?

alturismo, 2023-12-18:
> Ok now I understand you use ICAM? Did you compile libdvbcsa yourself?

nope, as mentioned above and you replied (may i missunderstood)

[image]

now, for syslog i prepared as followed

- VU is in standby (no tuner activity from VU)
- oscam dvbapi enabled in satpi (disabled in tvheadend )
- logging enabled in satpi
- tuners in satpi as follow

[image]

- scenario, i zap through channels now, starting with FTA -C, then oscam -C, then oscam -S

FTA -C is quite ok, just really small stuttering
oscam -C is not ok (even on only 1 tuned channel)

sample what i described upper

[image]

oscam -S is not ok (even on only 1 tuned channel)

sample what i described upper

[image]

finally i zap through some icam channels

i hope thats what you looking for to understand ? ;)

[messages.zip](https://github.com/Barracuda09/SATPI/files/13701328/messages.zip)

Barracuda09, 2023-12-18:
Did you enable in TvHeadend the `RTP/AVP/TCP transport supported` option

[image]

alturismo, 2023-12-18:
> Did you enable in TvHeadend the `RTP/AVP/TCP transport supported` option

nope, default its not ticked

alturismo, 2023-12-18:
> Did you enable in TvHeadend the `RTP/AVP/TCP transport supported` option

i would say this makes no difference here, it looks to me more like ... higher bandwith, more fails

[image]

this happens also on streaming only (decoding done in TVH), as higher the bandwith, as more dropouts ...

hard to describe ... seems my vuduo4kse is not so compatible ;)

Barracuda09, 2023-12-18:
Stange... it is hard to debug, as here it seems not that bad and decrypted in SatPI. I have vuuno4k
[image]

Barracuda09, 2023-12-18:
this Test is DVB-C as well

alturismo, 2023-12-18:
> this Test is DVB-C as well

may one more question, is it "as is" if satip using dvb-c that local usage is not possible while satpi is using it ?

i guess its due its an FBC Tuner ...

alturismo, 2023-12-18:
> Stange... it is hard to debug, as here it seems not that bad and decrypted in SatPI. I have vuuno4k

and you also use openatv with standard libdvbcsa ? nothing extra compiled ?

Barracuda09, 2023-12-18:
Here an FTA DVB-S2 example using a Raspberry PI with Sundtek tuners.
[image]

Barracuda09, 2023-12-18:
Well to be honest I do not use Enigma.. I have on my vuuno4k DVB-C just stock image but never setup Enigma on it.
and have on my vuuno4k DVB-S2 version OpenPLi but never setup Enigma on it or OSCam because i do not own a subscription.
Problem I do not have any antenna cables near the TV only 1Gbs network cable

> may one more question, is it "as is" if satip using dvb-c that local usage is not possible while satpi is using it ?
> 
> i guess its due its an FBC Tuner ...

But if you mean sharing the same Tuner/frontend that will most likely fail. But it should just work if one uses Tuner A/Frontent 1 and other Tuner D/Frontend 4

> and you also use openatv with standard libdvbcsa ? nothing extra compiled ?

I use mostly my own compiled libs. But do not know if the are better then OpenATV or OpenPLi (my guess, it does not matter)

alturismo, 2023-12-18:
> But if you mean sharing the same Tuner/frontend that will most likely fail. But it should just work if one uses Tuner A/Frontent 1 and other Tuner D/Frontend 4
> 
> >

thats what i expected and not working actually (DVBC), i assume due FBC Tuner ...
working as expected on DVBS (also FBC)

> I use mostly my own compiled libs. But do not know if they are better then OpenATV or OpenPLi (my guess, it does not matter)

i ll give it a try, have to setup a virtual ARM enviroment to compile it actually, will report back if that makes any difference

thanks for the informations

Barracuda09, 2023-12-18:
> thats what i expected and not working actually (DVBC), i assume due FBC Tuner ...
> working as expected on DVBS (also FBC)

I have vuuno4k with FBC tuners (DVB-S2 and DVB-C). You situation is different because you have 2 Root FBC tuners and of different types. I do not know if this has to be setup differently in SatPI (probably, because it does not work in you situation).

alturismo, 2023-12-18:
> I have vuuno4k with FBC tuners (DVB-S2 and DVB-C). Your situation is different because you have 2 Root FBC tuners and of different types. I do not know if this has to be setup differently in SatPI (probably, because it does not work in you situation).

yep, looks like openatv doesnt have the proper drivers for dvb-c fbc tuners

i digged a little deeper and changed this setting

[image]

from auto to a tuner higher then the 1st one ... and then, also on the VU directly it always stays black ...

so tuner configuration for satpi and dvb-c (fbc) as second tuner is prolly a mess ;)

Barracuda09, 2023-12-18:
What would happen if you switch the tuners around in the slots?  So DVB-C in slot A and DVB-S2 in slot B. will that make the problem move to DVB-S2?

alturismo, 2023-12-18:
> What would happen if you switch the tuners around in the slots? So DVB-C in slot A and DVB-S2 in slot B. will that make the problem move to DVB-S2?

cant imagine this would change anything, it seems the nature of dvb-c fbc tuner in the vu+ with oatv.
its not working the same way as the dvb-s fbc tuner where the assignement is functional.

and i guess (really just a guess) its a fw issue with openatv, as the preferred tuner assignement inside openatv also breaks tuning the channel (completely aside satpi), meaning, whenever i assign a prio tuner aside the 1st one, tuning will break ... thats why i "thought" dvb-c was broken in satpi, cause i assigned dvb-c tuners 5-8 (similar to dvb-s) ...

[image]

while on DVB-S there is a major difference (working as expected)

1/ assigned in VU as active, unticked in satpi (to not use them), working as expected
2/ disabled in VU as inactive, assigned in satpi (to use only them), working as expected

[image]

i hope its more understandable ...

so basically, this could be workaround by just using locally also a satip tuner instead direct (normal) tuner access, current state (here), only useable without descrambling ...

but ... when descrambling comes now into the game, when using the satpi oscam feature ... there are the dropouts which makes it more or less unusable here, way too many errors with picture dropouts as posted above.

so basically, can be closed, and i could lean on other threads here about oscam (icam) support and stability,
and about the small glitches on FTA ... lets forget about it ;)

alturismo, 2023-12-18:
in addition, here a small comparision, 2 DVB-C FTA Streams running simultan from VU

1/ left side, satpi usage
2/ right side, m3u stream directly from openatv

[image]

Barracuda09, 2023-12-19:
You could try OpenPLi

Barracuda09, 2023-12-19:
This is the same problem #187. Also not resolved

alturismo, 2023-12-19:
> This is the same problem #187. Also not resolved

yep, also saw this today morning, but didnt saw if you received the receiver or whatever there was planned ...

may a question, you are using a vu with an FBC dvbc tuner, with openpli and is working as expected without the limitations of non working tuner if the "2nd" access starts (either VU direct or satpi)

Barracuda09, 2023-12-19:
> yep, also saw this today morning, but didnt saw if you received the receiver or whatever there was planned ...

No, unfortunately We did not arrange anything.

> may a question, you are using a vu with an FBC dvbc tuner, with openpli and is working as expected without the limitations of non working tuner if the "2nd" access starts (either VU direct or satpi)

I use on the vuuno4k DVB-C version the stock Vu+ image, but I do not watch channels via Enigma (and is not configured).
[image]
[image]

alturismo, 2023-12-19:
> I use on the vuuno4k DVB-C version the stock Vu+ image, but I do not watch channels via Enigma (and is not configured).

and you dont use any decryption ... i assume on stock i definately would need a custom libdvbca.so ;)

alturismo, 2023-12-19:
> You could try OpenPLi

out of curiosity, i ll give it a shot, anything special i have to be aware off ?

alturismo, 2023-12-19:
> You could try OpenPLi

actually, i couldnt imagine it makes a difference in Quality ... but i have to admit ...

while 1 DVB-S stream is active on the Box itself
[image]

insane improvement ... well, actually expected behaviour how it should be ;)

[image]

ok, now i need a working oscam version ... if that works out ... i would have to laugh ;)

alturismo, 2023-12-19:
> > You could try OpenPLi
> 
> out of curiosity, i ll give it a shot, anything special i have to be aware off ?

1 note, what also doesnt work is "Dual" DVB-C usage ... 1st tuner needs to be free for either local or satpi usage ;) but once satpi is running stable we can always use local satip tuner setup ;)

Barracuda09, 2023-12-19:
Send me an e-mail, we arrange it further there with libs

alturismo, 2023-12-19:
> You could try OpenPLi

last report today ;) testing with the satpi version provided by you here in this thread, sadly about the same behaviour ...

1/ many dropouts
2/ no icam decryptions working (i guess libdvbcsa ...)

i test now how it runs as pure satpi Server for TVHeadend for a little, i guess i can drop the oscam idea

Barracuda09, 2024-02-09:
Hi @alturismo 

It is some time ago, but If you're still interested, could you try this version [satpi_1.6.2.188~g19b05a3 Enigma.zip](https://github.com/Barracuda09/SATPI/files/14221909/satpi_1.6.2.188.g19b05a3.Enigma.zip).

alturismo, 2024-02-09:
> Hi @alturismo 
> 
> It is some time ago, but If you're still interested, could you try this version [satpi_1.6.2.188~g19b05a3 Enigma.zip](https://github.com/Barracuda09/SATPI/files/14221909/satpi_1.6.2.188.g19b05a3.Enigma.zip).

Sure, just some work to do, i ll report 😉

alturismo, 2024-02-14:
> Hi @alturismo
> 
> It is some time ago, but If you're still interested, could you try this version [satpi_1.6.2.188~g19b05a3 Enigma.zip](https://github.com/Barracuda09/SATPI/files/14221909/satpi_1.6.2.188.g19b05a3.Enigma.zip).

ok, new windows mounted, sat dish mounted ;)

1st lookup

1/ looking way better signal stability, looks now like minisatip with 0 dropped frames
2/ oscam still only working for non icam channels, but working nice with non icam ...

Barracuda09, 2024-02-15:
> 2/ oscam still only working for non icam channels, but working nice with non icam ...

Yes that could be, because this binary was not at all compiled with ICAM support.

Barracuda09, 2024-02-15:
But all tuners work? So the DVB-S2 and DVB-C?

Barracuda09, 2024-02-15:
For ICAM there is an issue  #184. There some report that it is working.

alturismo, 2024-02-15:
> For ICAM there is an issue  #184. There some report that it is working.

Not here 😉

alturismo, 2024-02-15:
> But all tuners work? So the DVB-S2 and DVB-C? as described in https://github.com/Barracuda09/SATPI/issues/203#issue-2044622253

Yes 👍

Barracuda09, 2024-02-16:
Hi @alturismo could you try (if you have some time) this ICAM version: [satpi_1.6.2.192~g614852e Enigma.zip](https://github.com/Barracuda09/SATPI/files/14316132/satpi_1.6.2.192.g614852e.Enigma.zip)

Do you still have the `libdvbcsa.so.1.0_ICAM.zip` I emailed you some time ago?
-> Copy that one to `/usr/lib/libdvbcsa.so.1`

alturismo, 2024-02-17:
> Hi @alturismo could you try (if you have some time) this ICAM version: [satpi_1.6.2.192~g614852e Enigma.zip](https://github.com/Barracuda09/SATPI/files/14316132/satpi_1.6.2.192.g614852e.Enigma.zip)

made a run through, same result

> Do you still have the `libdvbcsa.so.1.0_ICAM.zip` I emailed you some time ago?
> -> Copy that one `(libdvbcsa.so.1.0.1)` to `/usr/lib/libdvbcsa.so.1`

yes and also tested with your .so.1, sadly still the same
using the "original" .so.1 one will likely crash satpi when starting a stream, so needed to use the provided one anyway ;)

also, while using it with non icam streams, oscam usage will end in ts errors again (framedrops).

thanks for taking your time 👍 

satpi itself without internal oscam is working stable (using tvheadend oscam dvbapi to decode), just very long zapping timings compared to minisatip or legacy usage, this just as note, talking about 4 - 5 seconds (satpi) compared to 1 - 2 seconds (minisatip) compared to instant (legacy FBC) ;).

Barracuda09, 2024-02-17:
Could you try to use this frontend.html it shows if libdvbcsa has ICAM enabled [frontend.zip](https://github.com/Barracuda09/SATPI/files/14318360/frontend.zip) like here:

[image]

Barracuda09, 2024-02-17:
Sorry for wasting your time, but could I ask one more favor.
 
> yes and also tested with your .so.1, sadly still the same
> using the "original" .so.1 one will likely crash satpi when starting a stream, so needed to use the provided one anyway ;)

Sorry my bad, you are correct. I messed up.. This version is detecting the correct libdvbcsa so it will not crash [satpi_1.6.2.192~g614852e_Enigma.zip](https://github.com/Barracuda09/SATPI/files/14318506/satpi_1.6.2.192.g614852e_Enigma.zip)

alturismo, 2024-02-17:
> Sorry my bad, you are correct. I messed up.. This version is detecting the correct libdvbcsa so it will not crash

ok, now its working to decode, normal && icam channels ;) 1 step further.

> Could you try to use this frontend.html it shows if libdvbcsa has ICAM enabled [frontend.zip](https://github.com/Barracuda09/SATPI/files/14318360/frontend.zip) like here:

[image]

yep, as expected its enabled ;)

but sadly i have to say, same behaviour as before about framedrops when using oscam dvbapi from satpi

[image]

this does not happen on either

1/ using satpi without dvbapi decode, tvheadend does the decode using dvbapi (same oscam Server of course)
2/ using minisatip with dvbapi decode, no dropped frames there either (same oscam Server of course)

so, technically its now working as it should, can fully decode the streams (also icam) using your libdvbcsa.so.1, but usable its not as its dropping way too many frames with the satpi included decoder.

also, technically way longer zapping times compared to minisatip as described above.

so, i keep this version now running but without oscam decode, just to see what happens in terms of dropped frames without decoding, looking so far so good without it ;)

[2 more replies]