Barracuda09/SATPI #207: permanent artifacts with commit 410b47f

nikolauzi11, 2024-01-12:
Hi @Barracuda09 

First of all, I would like to wish you all the best for the New Year.
Unfortunately there are permanent artifacts with the last commit(410b47f).
The client is still the VDR.

Barracuda09, 2024-01-12:
Hi @nikolauzi11 

Yes, you too have a great new year,

So the commit before this was beter? or is this stiil as #181

nikolauzi11, 2024-01-12:
Hi @Barracuda09 

With commit bdab5f2 it still looks good. I always try to stay somewhat up to date.

Barracuda09, 2024-01-16:
Hi @nikolauzi11 

Thanks for your support! it is appreciated.

Could you try for testing, to modify Filter.h in:
https://github.com/Barracuda09/SATPI/blob/410b47f5d9aaed43d41976b7ea6c345f171397e1/src/mpegts/Filter.h#L205-L215

And change Line 211 to:
```
			_mutex.unlock();
			const bool done = openPid(pid);
			_mutex.tryLock(15000);
```
---

And at this place:
https://github.com/Barracuda09/SATPI/blob/410b47f5d9aaed43d41976b7ea6c345f171397e1/src/mpegts/Filter.h#L220-L230

And change Line 226 to:
```
			_mutex.unlock();
			const bool done = closePid(pid);
			_mutex.tryLock(15000);
```

nikolauzi11, 2024-01-16:
Hi @Barracuda09 

This makes it work again.
Thanks!