Barracuda09/SATPI #208: After update no stream data no picture

lordduckstein, 2024-02-01:
Hello,

after the update. The SATIP stream is broken. I find in the Tuners in DVBviewer but no picture. Before the update the system were running.

Here my system
- GigablueUHD UE 4K, 2 x Sat FBC-Tuner,
- Inverto IDLU-UWT110CUO10-32P Multiswitch

Here my log 
[log.txt](https://github.com/Barracuda09/SATPI/files/14128424/log.txt)

[image]
[image]
[image]
[image]
[image]
[image]

The fronted 2 is not configured, because I use the in Engima2. Ich have OpenATV image with the new Devel-Image 7.5 but in other images 7.3 or 7.4 it is also not working.

I think it must issues after the last changes ...

Thank you
Lordi

Barracuda09, 2024-02-01:
Hi @lordduckstein 

Thank you for your feedback. It seems that you do not get a lock to the requested frequency. 
I have a few questions:
- Did it work before then?
- What kind of LNB is connected to the multiswitch?
- Did you adjust the `Repeat DiSEqC` to more then 1?

```
  Thu Feb  1 17:41:53.0191 2024  6  Frontend: 1, Not locked yet   (FE status 0x01)...   <--- *** No Lock ***
  Thu Feb  1 17:41:52.5683 2024  6  Frontend: 1, Tuned, waiting on lock...
  Thu Feb  1 17:41:52.5682 2024  7  Frontend: 1, Set Properties: Frequency 1420000
  Thu Feb  1 17:41:52.5682 2024  6  Frontend: 1, Closing /dev/dvb/adapter0/frontend0 with fd: 12
  Thu Feb  1 17:41:52.4398 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Thu Feb  1 17:41:52.3166 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Thu Feb  1 17:41:52.1934 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Thu Feb  1 17:41:52.0701 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Thu Feb  1 17:41:51.9468 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Thu Feb  1 17:41:51.8235 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Thu Feb  1 17:41:51.7002 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Thu Feb  1 17:41:51.6767 2024  6  HTTP Client 192.168.178.44:56891 Connection closed with fd: 9
  Thu Feb  1 17:41:51.5765 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Thu Feb  1 17:41:51.5533 2024  6  Frontend: 1, Sending DiSEqC: [0x70] [0x05] [0x34] [0x03] - DiSEqC Src: 0 - UB: 0
  Thu Feb  1 17:41:51.5533 2024  7  Frontend: 1, Using LNB with: Lof-High=10600 MHz  Lof-Low=9750 MHz  Lof-Switch=11700 MHz -> IF=1432 MHz
  Thu Feb  1 17:41:51.5082 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Writing DiSEqC command with fd: 12
  Thu Feb  1 17:41:51.5079 2024  6  Frontend: 1, Start tuning process for DVB-S(2)...
  Thu Feb  1 17:41:51.5078 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 11 (0 ms)
  Thu Feb  1 17:41:51.4574 2024  6  Frontend: 1, Closing all active PID filters...
  Thu Feb  1 17:41:51.4574 2024  6  Frontend: 1, Updating frontend...
  Thu Feb  1 17:41:51.4573 2024  7  Frontend: 1, Parsing transport parameters (Finished)
  Thu Feb  1 17:41:51.4572 2024  7  Frontend: 1, Parsing PID parameter: pids=0
  Thu Feb  1 17:41:51.4570 2024  6  Frontend: 1, New frequency requested (0.0000 -> 12032.0000), clearing old channel data...
  Thu Feb  1 17:41:51.4569 2024  6  Frontend: 1, Parsing transport parameters...
  Thu Feb  1 17:41:51.4569 2024  6  Frontend: 1, StreamClient with SessionID 0263094224 for dvbs2
  Thu Feb  1 17:41:51.4567 2024  7  Frontend: 1, Found Streaming type: RTP/AVP
  Thu Feb  1 17:41:51.4566 2024  6  Found FrondtendID: x (fe=x)  StreamID: x  SessionID: 0263094224  New Session: true
  Thu Feb  1 17:41:51.4561 2024  7  <CRLF>
  Thu Feb  1 17:41:51.4561 2024  7  Transport: RTP/AVP;unicast;client_port=47742-47743
  Thu Feb  1 17:41:51.4561 2024  7  CSeq: 2
  Thu Feb  1 17:41:51.4561 2024  7  RTSP Stream data from client None with IP 192.168.178.44 on Port 56892: SETUP rtsp://192.168.178.58:554/?src=1&freq=12032&msys=dvbs2&plts=off&fec=910&pol=h&ro=0.35&sr=27500&mtype=qpsk&pids=0 RTSP/1.0
  Thu Feb  1 17:41:51.4551 2024  7  <CRLF>
  Thu Feb  1 17:41:51.4551 2024  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
```

Kind regards,
Marc

lordduckstein, 2024-02-01:
> Hi @lordduckstein
> 
> Vielen Dank für Ihr Feedback. Es scheint, dass Sie keine Sperre für die angeforderte Frequenz erhalten. I have a few questions:
> 
> * Hat es vorher funktioniert?
> * What kind of LNB is connected to the multiswitch?
> * Did you adjust the `Repeat DiSEqC` to more then 1?
> 
> ```
>   Thu Feb  1 17:41:53.0191 2024  6  Frontend: 1, Not locked yet   (FE status 0x01)...   <--- *** No Lock ***
>   Thu Feb  1 17:41:52.5683 2024  6  Frontend: 1, Tuned, waiting on lock...
>   Thu Feb  1 17:41:52.5682 2024  7  Frontend: 1, Set Properties: Frequency 1420000
>   Thu Feb  1 17:41:52.5682 2024  6  Frontend: 1, Closing /dev/dvb/adapter0/frontend0 with fd: 12
>   Thu Feb  1 17:41:52.4398 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
>   Thu Feb  1 17:41:52.3166 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
>   Thu Feb  1 17:41:52.1934 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
>   Thu Feb  1 17:41:52.0701 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
>   Thu Feb  1 17:41:51.9468 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
>   Thu Feb  1 17:41:51.8235 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
>   Thu Feb  1 17:41:51.7002 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
>   Thu Feb  1 17:41:51.6767 2024  6  HTTP Client 192.168.178.44:56891 Connection closed with fd: 9
>   Thu Feb  1 17:41:51.5765 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
>   Thu Feb  1 17:41:51.5533 2024  6  Frontend: 1, Sending DiSEqC: [0x70] [0x05] [0x34] [0x03] - DiSEqC Src: 0 - UB: 0
>   Thu Feb  1 17:41:51.5533 2024  7  Frontend: 1, Using LNB with: Lof-High=10600 MHz  Lof-Low=9750 MHz  Lof-Switch=11700 MHz -> IF=1432 MHz
>   Thu Feb  1 17:41:51.5082 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Writing DiSEqC command with fd: 12
>   Thu Feb  1 17:41:51.5079 2024  6  Frontend: 1, Start tuning process for DVB-S(2)...
>   Thu Feb  1 17:41:51.5078 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 11 (0 ms)
>   Thu Feb  1 17:41:51.4574 2024  6  Frontend: 1, Closing all active PID filters...
>   Thu Feb  1 17:41:51.4574 2024  6  Frontend: 1, Updating frontend...
>   Thu Feb  1 17:41:51.4573 2024  7  Frontend: 1, Parsing transport parameters (Finished)
>   Thu Feb  1 17:41:51.4572 2024  7  Frontend: 1, Parsing PID parameter: pids=0
>   Thu Feb  1 17:41:51.4570 2024  6  Frontend: 1, New frequency requested (0.0000 -> 12032.0000), clearing old channel data...
>   Thu Feb  1 17:41:51.4569 2024  6  Frontend: 1, Parsing transport parameters...
>   Thu Feb  1 17:41:51.4569 2024  6  Frontend: 1, StreamClient with SessionID 0263094224 for dvbs2
>   Thu Feb  1 17:41:51.4567 2024  7  Frontend: 1, Found Streaming type: RTP/AVP
>   Thu Feb  1 17:41:51.4566 2024  6  Found FrondtendID: x (fe=x)  StreamID: x  SessionID: 0263094224  New Session: true
>   Thu Feb  1 17:41:51.4561 2024  7  <CRLF>
>   Thu Feb  1 17:41:51.4561 2024  7  Transport: RTP/AVP;unicast;client_port=47742-47743
>   Thu Feb  1 17:41:51.4561 2024  7  CSeq: 2
>   Thu Feb  1 17:41:51.4561 2024  7  RTSP Stream data from client None with IP 192.168.178.44 on Port 56892: SETUP rtsp://192.168.178.58:554/?src=1&freq=12032&msys=dvbs2&plts=off&fec=910&pol=h&ro=0.35&sr=27500&mtype=qpsk&pids=0 RTSP/1.0
>   Thu Feb  1 17:41:51.4551 2024  7  <CRLF>
>   Thu Feb  1 17:41:51.4551 2024  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
> ```
> 
> Kind regards, Marc

Hello Marc,

thanks for your support

1.) Yes It has worked before, only I have update the system. I have made a heavy mistake "never touch an running system".

2.) It is an Durline eco Quad-LNB

3.) which DiSEqC settings do you mean in enigma2?
 
[image]

There is no DiSEcQ settings

or in SATIP

I have try with DiSEcQ Repeat = 0
[image]

or before  with DiSEcQ Repeat = 8

[image]

Nothing happens...

Thank you

Lordi

lordduckstein, 2024-02-01:
Do you have old ipk`s then I can figure out it is SATIP or other things like enigma2?

Barracuda09, 2024-02-01:
> Do you have old ipk`s from SATPI then I can figure out it is SATPI or other things like enigma2?

No I do not have one, but it should work with this version.

I have some further questions:
- Can you give the Enigma2 tuner settings for Tuner A (Like you gave for Tuner B)
- The Channel freq of Unicable II starts at 0 So SCR1 is `Channel Slot (0-32)` is 0
  - Example (See below) CH1 (SCR1) is:
    - `Channel Freq (MHz)` - 1210 
    - `Channel Slot (0-32)` - 0
```
User band (channel) frequencies | Configurable; Default 32 User bands:
  | CH1: 1210 MHz (EN50494+EN50607)
  | CH2: 1420 MHz (EN50494+EN50607)
  | CH3: 1680 MHz (EN50494+EN50607)
```

lordduckstein, 2024-02-01:
Hello Marc,

Tuner A is not configured in Enigma2, because It is not work. SATPI or Enigma2 don´t like this, when the tuner is in both are configured

[image]

Yes I know that we have to start with "0"

Here again the settings for Tuner A in SATPI

[image]

Here the actuell Fronted overview the other tuner I have disabled

[image]

It is not working

Here a new log with only Tuner A is enabled:

[log2.txt](https://github.com/Barracuda09/SATPI/files/14129660/log2.txt)

Lordi

Barracuda09, 2024-02-01:
I see you use `Channel Freq` of 1420 with `Channel Slot` 0 -> Should that not be set to 1

Barracuda09, 2024-02-01:
You could increase 
- `Delay before write` to 15
- `Delay after write` to 30

lordduckstein, 2024-02-01:
Hello Marc,

no it don´t work 

[log3.txt](https://github.com/Barracuda09/SATPI/files/14129853/log3.txt)

Barracuda09, 2024-02-01:
Did you try setting it just both at 100

lordduckstein, 2024-02-01:
Hello Marc,

what should I say your are an "GOD"

IT WORKS

You are my Hero

CU
Lordi

Barracuda09, 2024-02-01:
@lordduckstein No problem

lordduckstein, 2024-02-01:
Hello Marc, 

A Last question this work with the Root-Tuner.

How are the settings with the other  no-root-tuner?

What you think?

Thank you
Lordi

closed 2024-02-02 by lordduckstein