Barracuda09/SATPI #210: Only 8 Tuners (1 FBC) works

foehlmann, 2024-02-05:
Hello everyone,

I have the problem that only the first 8 tuners of my box work. I have a VU+ Duo 4k SE with 2x DVB-S2 FBC. The first 8 frontends work without any problems. But from number 8 onwards no packets are transmitted, no matter what settings I make. Does anyone have any tips? I have already checked out the latest source from Github and started cross compiled on the box.
[image]
[image]

Barracuda09, 2024-02-05:
Hi @foehlmann 

Thanks for you interest in SatPI.

I know of this problem, but have so far not come up with a solution. Issues #203 #187 seem to have (or look to have) the problem.

Unfortunately I can not test this myself, as I do not own this hardware.

foehlmann, 2024-02-05:
Unfortunately the same problem... 1-8 running, from 9 no packages.

[image]

Barracuda09, 2024-02-05:
Which version do you use of SatPI (Home Page)

foehlmann, 2024-02-05:
1.6.2.146~g91dabfb

Barracuda09, 2024-02-05:
That is quite an old version, which image are you using?

foehlmann, 2024-02-05:
OpenPLi 9.0-release (2023-12-27-release-9.0)

I had cross compiled an binary from the git today. That causes the same error 😒

foehlmann, 2024-02-06:
The same.

I have already considered reflashing an image and not configuring the tuners in Enigma at all. Do you have any recommendations on which image works best with SatPI?

Barracuda09, 2024-02-06:
OpenPLi 9.0 should be ok.

Do you know how to replace the SatPI binary? So I can give you an test version?
So you can give me a log?

foehlmann, 2024-02-06:
Sure..

Barracuda09, 2024-02-06:
could you try this version [satpi_1.6.2.188~g19b05a3 Enigma(Debug).zip](https://github.com/Barracuda09/SATPI/files/14178341/satpi_1.6.2.188.g19b05a3.Enigma.Debug.zip) and give me the log of the startup of SatPI?

foehlmann, 2024-02-06:
`Starting satpi:/usr/bin/satpi: /usr/lib/libstdc++.so.6: version `GLIBCXX_3.4.29' not found (required by /usr/bin/satpi)
/usr/bin/satpi: /lib/libc.so.6: version `GLIBC_2.34' not found (required by /usr/bin/satpi)`

I think you are using some newer versions that the openpli 9 image

`find /lib -iname "libc-*" 
/lib/libc-2.33.so`

Barracuda09, 2024-02-06:
Yes sorry with GCC12. Use this version [satpi_1.6.2.188~g19b05a3 Enigma(Debug).zip](https://github.com/Barracuda09/SATPI/files/14178773/satpi_1.6.2.188.g19b05a3.Enigma.Debug.zip)

foehlmann, 2024-02-06:
`/usr/bin/satpi: cannot execute binary file`

need to leave for an appointment for about 1,5hrs .. will start later with that

Barracuda09, 2024-02-06:
It is compiled with libdvbcsa, maybe you need to install that too?

foehlmann, 2024-02-06:
Thats installed:
`libdvbcsa1 - 1.1.0-r0.0`

Barracuda09, 2024-02-06:
i assume it has the correct rights to execute the program?

foehlmann, 2024-02-06:
It has +x , yes.

If i try to start it direct without the init.d service:
`./satpi: cannot execute binary file: Exec format error`

foehlmann, 2024-02-06:
```
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 54.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x42
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0x00f
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 54.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x42
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0x00f
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 54.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x42
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0x00f
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 54.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x42
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0x00f
CPU revision	: 3

```

Barracuda09, 2024-02-06:
Strange... what does `file ./satpi` say

foehlmann, 2024-02-06:
Just flashed the box again... now your new binary works - and uses all tuners (the first 10 for now)

[image]

Barracuda09, 2024-02-06:
could you give me the log of the startup of satpi or enable log to syslog and startup included

foehlmann, 2024-02-06:
Enables log to syslog and started over:
`Feb  6 15:09:08 vuduo4kse local1.info SatPI[26465]: [                           src/HttpServer.cpp:058] Setting up HTTP server
Feb  6 15:09:08 vuduo4kse local1.info SatPI[26465]: [                           src/RtspServer.cpp:048] Setting up RTSP server
Feb  6 15:09:09 vuduo4kse local1.info SatPI[26465]: [                     src/upnp/ssdp/Server.cpp:105] Setting up SSDP server with BOOTID: 14  annouce interval: 60 Sec  TTL: 1
Feb  6 15:09:10 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63653 with fd: 8
Feb  6 15:09:12 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63654 with fd: 9
Feb  6 15:09:12 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63656 with fd: 10
Feb  6 15:09:12 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63655 with fd: 11
Feb  6 15:09:12 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63657 with fd: 12
Feb  6 15:09:12 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63658 with fd: 13
Feb  6 15:09:13 vuduo4kse local1.info SatPI[26465]: [                     src/socket/TcpSocket.cpp:109] HTTP Client 192.168.1.192:63655 Connection closed with fd: 11
Feb  6 15:09:13 vuduo4kse local1.info SatPI[26465]: [                     src/socket/TcpSocket.cpp:109] HTTP Client 192.168.1.192:63653 Connection closed with fd: 8
Feb  6 15:09:13 vuduo4kse local1.info SatPI[26465]: [                     src/socket/TcpSocket.cpp:109] HTTP Client 192.168.1.192:63654 Connection closed with fd: 9
Feb  6 15:09:13 vuduo4kse local1.info SatPI[26465]: [                     src/socket/TcpSocket.cpp:109] HTTP Client 192.168.1.192:63658 Connection closed with fd: 13
Feb  6 15:09:13 vuduo4kse local1.info SatPI[26465]: [                     src/socket/TcpSocket.cpp:109] HTTP Client 192.168.1.192:63656 Connection closed with fd: 10
Feb  6 15:09:13 vuduo4kse local1.info SatPI[26465]: [                     src/socket/TcpSocket.cpp:109] HTTP Client 192.168.1.192:63657 Connection closed with fd: 12
Feb  6 15:09:14 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63659 with fd: 8
Feb  6 15:09:20 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63660 with fd: 9
Feb  6 15:09:20 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63662 with fd: 10
Feb  6 15:09:20 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63661 with fd: 11
Feb  6 15:09:20 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63664 with fd: 12
Feb  6 15:09:20 vuduo4kse local1.info SatPI[26465]: [                    src/socket/SocketAttr.cpp:156] HTTP Connection from 192.168.1.192 Port 63663 with fd: 13
Feb  6 15:09:22 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 1:  Root 1  Index 0  Offset 0  Connected 0  Linked 0
Feb  6 15:09:22 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 2:  Root 1  Index 1  Offset 0  Connected 0  Linked 1
Feb  6 15:09:22 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 3:  Root 0  Index 2  Offset 0  Connected 0  Linked 1
Feb  6 15:09:22 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 4:  Root 0  Index 3  Offset 0  Connected 0  Linked 1
Feb  6 15:09:22 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 5:  Root 0  Index 4  Offset 0  Connected 0  Linked 1
Feb  6 15:09:22 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 6:  Root 0  Index 5  Offset 0  Connected 0  Linked 1
Feb  6 15:09:22 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 7:  Root 0  Index 6  Offset 0  Connected 0  Linked 1
Feb  6 15:09:22 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 8:  Root 0  Index 7  Offset 0  Connected 0  Linked 1
Feb  6 15:09:23 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 9:  Root 1  Index 8  Offset 8  Connected 9  Linked 0
Feb  6 15:09:23 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 10:  Root 1  Index 9  Offset 8  Connected 9  Linked 1
Feb  6 15:09:23 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 11:  Root 0  Index 10  Offset 8  Connected 9  Linked 1
Feb  6 15:09:23 vuduo4kse local1.info SatPI[26465]: [               src/input/dvb/delivery/FBC.cpp:104] Frontend 12:  Root 0  Index 11  Offset 8  Connected 9  Linked 1
[... cut ...]
Feb  6 15:09:23 vuduo4kse local1.info SatPI[26465]: [                           src/Properties.cpp:074] Setting HTTP Port to: 8875
Feb  6 15:09:23 vuduo4kse local1.info SatPI[26465]: [                           src/Properties.cpp:081] Setting RTSP Port to: 554
Feb  6 15:09:23 vuduo4kse local1.info SatPI[26465]: [                           src/Properties.cpp:086] Setting WEB Path to: /usr/share/satpi/web
Feb  6 15:09:23 vuduo4kse local1.info SatPI[26465]: [                           src/Properties.cpp:090] Setting App Data Path to: /usr/bin
`

foehlmann, 2024-02-06:
Strange things happening: Now i have packets on all frontends, but cant get the tuner locked.. so no stream

Barracuda09, 2024-02-06:
16 frontends at the same time?

What does `top` show?

Barracuda09, 2024-02-06:
That are "keep alive" packets, you can see that they will not increase that quickly.

But could you recap what is working like
- Does only frontend 1 working?
- Does only frontend 8 working?

foehlmann, 2024-02-06:
For now: It's not working... no lock on any frontend. 
Until this morning the old version keeps running on frontend 1-8.

I replaced the binary again with the old one - no luck.. 
Can't think of anything that can cause this.
Rebooted, reset settings, ...

If i stop satpi and try locally on the receiver it tunes fine... lock on all channels.

Barracuda09, 2024-02-06:
Did you enable `Send DiSEqC Via Linked Root Tuner`

Barracuda09, 2024-02-06:
You need to update the web folder with current one of github

foehlmann, 2024-02-06:
> Did you enable `Send DiSEqC Via Linked Root Tuner`

yes

Barracuda09, 2024-02-06:
Could you zip and post XML file http://192.168.1.13:8875/SatPI.xml

foehlmann, 2024-02-06:
[SatPI.xml.zip](https://github.com/Barracuda09/SATPI/files/14184116/SatPI.xml.zip)

foehlmann, 2024-02-06:
> You need to update the web folder with current one of github

done that

Barracuda09, 2024-02-06:
Could you give a beter "mental" picture of the connections which inputs is connected to which LNB (satelliet) So:
- Tuner in slot A
  - Input A -> Which sat?
  - input B -> Which sat?
- Tuner in slot B
  - Input A -> Which sat?
  - input B -> Which sat?

foehlmann, 2024-02-06:
Tuner in slot A
Input A -> Astra 19.2
input B -> None
Tuner in slot B
Input A -> None
input B -> None

Barracuda09, 2024-02-06:
Ok, so no psychical connection to tuner in slot B

I see that you have for frontend 1:
- `Channel Freq (MHz)` = 1210  (I assume this is SCR1 of JESS LNB)
- `Channel Slot (0-32)` = 1  (Then this should be 0. **So always subtract 1**) 

Please take a look at: https://github.com/Barracuda09/SATPI/wiki/Setup-Unicable-and-JESS

I am not sure if it is possible (at this point in SatPI) to connect Tuner Slot B software whise to Tuner Slot A - Input A. But I assume this works in Enigma2

foehlmann, 2024-02-06:
> Ok, so no psychical connection to tuner in slot B
>
> I see that you have for frontend 1:
>
> * `Channel Freq (MHz)` = 1210  (I assume this is SCR1 of JESS LNB)
> * `Channel Slot (0-32)` = 1  (Then this [...]

That was working in the old setup, strange - will correct that.

> I am not sure if it is possible (at this point in SatPI) to connect Tuner Slot B software whise to Tuner Slot A - Input A. But I assume this works in Enigma2

Works fine in enigma2, yes. So probably an splitter would help me out?

Barracuda09, 2024-02-07:
> That was working in the old setup, strange - will correct that.
- I am not sure this has changed over time
> 
> > I am not sure if it is possible (at this point in SatPI) to connect Tuner Slot B software whise to Tuner Slot A - Input A. But I assume this works in Enigma2
> 
> Works fine in enigma2, yes. So probably an splitter would help me out?
- Could you give a screen capture of how this setup is handled in Enigma.
- Yes a splitter you could try, I give that a better chance.

Barracuda09, 2024-02-09:
Hi @foehlmann 

Thanks for the feedback, that is appreciated!

Could you try this version [satpi_1.6.2.188~g19b05a3 Enigma.zip](https://github.com/Barracuda09/SATPI/files/14221909/satpi_1.6.2.188.g19b05a3.Enigma.zip), it is a release version without debug info and should be a bit faster.

foehlmann, 2024-02-09:
@Barracuda09 sure, tomorrow 😊 

If I change the version now, it will be the last thing I do. Then the TV will cut out for my wife... that's not good.

foehlmann, 2024-02-10:
I´ve replaced the binary. Let´s see how it goes today 🙂
Test in production environment ...

Barracuda09, 2024-02-10:
Do you, by the way, use satipclient?

foehlmann, 2024-02-10:
I'll try to... for now i'm using tvheadend + player on apple tv.

satipclient stutters ... i have just compiled your newest fork and moved the binary to a vuzero4k and give it a try.

Barracuda09, 2024-02-10:
I use it on OpenPLi 9.0 which you can adjust vtuner for satipclient with a dialog. Which you can set for TCP mode.

How did you compile it?

Barracuda09, 2024-02-10:
Here is the bin I use for my vuzero4k [satipclient.zip](https://github.com/Barracuda09/SATPI/files/14229675/satipclient.zip)

logicencoder, 2026-09-14:
Solved this in our fork: https://github.com/logicencoder/satpi

The second FBC bank on the Duo 4K SE (frontend8-15, bcm7335/NEXUS) needed several driver-level fixes, all verified live with DVBViewer + Enigma2 running alongside:

- DiSEqC on FBC children must go through a dup() of the already-open root frontend fd (minisatip model); bcm7335_send_diseqc_msg on the child fd times out
- NEXUS demux: DMX_SET_SOURCE on the same fd with the frontend index, mid-stream ADD_PID
- SIGSEGV on whitespace-only RTSP keepalive (headers[0] on empty vector) and SIGPIPE on client disconnect
- When the lock wait times out with explicit plts/fec/ro hints, retune once with them relaxed to AUTO — clients forward stale channel-list values (e.g. plts=off on pilots-on transponders) that can never lock
- SSDP: bind with --iface-name so LOCATION isn't 0.0.0.0

Working config + write-up in docs/. Happy to upstream the patches if there's interest.

Barracuda09, 2026-09-15:
Thanks for looking into this issue.
I am happy to merge your excellent contributions.