Barracuda09/SATPI #211: No signal signal strenght on tvheadend

aurelmarius, 2024-02-14:
I'm running satpi on ubuntu 20.04 with 2 usb WinTV dualhd dvb-c tunners. I notice that signal strenght info doesn't shows in tvheadend like my hdhomerun tunners.
[image][image][image]

Saentist, 2024-02-15:
This is depends on "drivers" installed
if used build-in in kernel not expect more then basic
try
https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder

Barracuda09, 2024-02-15:
Hi @aurelmarius and @Saentist 

It seems that in the screen shots there is in the `Monitor Info` section the `Signal Strength`but No `snr`
So maybe something else is wrong too?

Saentist, 2024-02-15:
Driver module is not capable to send DVBAPI 5.x information use 3.x
In TVH Legacy mode /Force old status/ sometime fix this.
Playing with multiplying values and checkbox sometime fix problem.
[image]

Barracuda09, 2024-02-15:
@Saentist and @aurelmarius 

I think your solution is when TVH is retrieving the signal status from the drivers. But I believe this signal is send by SatPI via RTP.

The problem is fixed with(I think) to disable this in TVH (See TV-adapters parameters):
- Check tuner-number in signal-status messages

The explanation is:
When TVH expects a specific tuner number, but if that tuners is used, SatPI will use a different tuner number. And with this enabled TVH will discard the RTCP info, because is expects for example 1 and maybe SatPI uses 5 and sends that.

aurelmarius, 2024-02-15:
I'm running media tree hwe kernel with hauppauge patches. I tried changing those settings but no luck. I can live with this, no problem 😀

Barracuda09, 2024-02-16:
So disabling `Check tuner-number in signal-status messages` does not work?

Strange here it does, which version of TVH are you using?

aurelmarius, 2024-02-16:
TVH Build: 4.3-2229~g6b5defc76 (2024-02-12T17:57:09+0000)
[image]
[image]

aurelmarius, 2024-02-16:
[image]
I think the problem is that Signal strenght is 65534 and TVH expects 100 or 240. Testing with minisatip, it reports correct info.

Barracuda09, 2024-02-16:
I think @Saentist had a point with:
> Driver module is not capable to send DVBAPI 5.x information use 3.x

SatPI is probably using 5.x and only falls back to 3.x when 5.x fails.

Saentist, 2024-02-16:
There is a simple check
if is in `%` dvbapi 3.x 
if is in `dBm` dvbapi 5.x
But not respected by all driver developers.
Still some drivers report signal/quality from `0-FF (255)` other from `0-FFF (4095)` also `0-FFFF (65535)` 
most strange is when is mixed ex: signal = 215 but quality = 45800
There is no **Standard**

@Barracuda09 
If you can suggest any code improvement to TVH and post PR, I can try force thinks to be merged faster.

aurelmarius, 2024-02-16:
If i run tvheadend on that box i get signal in %.

Barracuda09, 2024-02-16:
@aurelmarius Did you change something in between? Because the first screen shot was 221:
***

[image]
***
And second 65534:
***

[image]

aurelmarius, 2024-02-16:
First screen was with media tree kernel without hwe, second is with mediatree-hwe-kernel.

Barracuda09, 2024-02-21:
Hi @aurelmarius 

Could you please try this version [satpi_1.6.2.193~g66c090c_Enigma.zip](https://github.com/Barracuda09/SATPI/files/14356473/satpi_1.6.2.193.g66c090c_Enigma.zip) to see if the Signal status are there?

There is also a new frontend.html included, you need to update. There is now also a `Force Old Style Signal Status` see:
[image]

Barracuda09, 2024-02-21:
Please checkout and build new commit 35c6341db3375a3e21bf27f3477bb9ed2cb5ab18
Please not: there is also a new frontend.html