Barracuda09/SATPI #212: Match frontend number in tvheadend

aurelmarius, 2024-02-15:
Is there any workaround to match the frontend number in tvheadend with the frontend in satpi ?
For example, I have 4 dvb-c tunners in satpi exported to tvheadend via sat>ip but i also have 2 cable providers. I want to use 2 tunners with the first provider and other 2 with the second provider. In tvh I have created two networks and assigned first 2 tunners to one network and the other to the second network, but it doesn't work as expected because satpi doesn't know what frontend is in use in tvh.
Thanks.

Barracuda09, 2024-02-16:
This should work when enabling `FE supported` in TV Adapters Parameters.

So you have 4 Frontends in SatPI:
- 1 Frontend Provider X
- 2 Frontend Provider X
- 3 Frontend Provider Y
- 4 Frontend Provider Y

SatPI will try to open the requested fe=x, but when in use it will try another DVB-C tuner that is not in use.

Barracuda09, 2024-02-16:
Strange, could you give a syslog (enable `Log messages to syslog` in SatPI) and do a request.

aurelmarius, 2024-02-16:
Tested again now and you are right. If FE is not in use it will choose that one. But here comes the problem, if the provider X fe is in use and other client is requesting for example mux: "482MHz" from that provider, satpi will enable other fe that is on provider Y but requesting mux from provider X.
An option like strict mode would help, so if tvh is requesting stream from FE1 and FE1 is not available then don't choose other FE.

Barracuda09, 2024-02-16:
What is the 'other' client, Not TVH?

aurelmarius, 2024-02-16:
Yes, other client through tvh. Only tvh is connecting to satpi.

Barracuda09, 2024-02-16:
So I understand:
-  1 Frontend Provider X  --> **in use by TVH?**
-  2 Frontend Provider X  --> **Request by TVH 482MHz?**
-  3 Frontend Provider Y
-  4 Frontend Provider Y

does not work?

aurelmarius, 2024-02-16:
For example:

Satpi FE1 -> Provider X
Satpi FE2 -> Provider X
Satpi FE3 -> Provider Y
Satpi FE4 -> Provider Y

TVH -> Satpi FE1 -> Network X
TVH -> Satpi FE2 -> Network X
TVH -> Satpi FE3 -> Network Y
TVH -> Satpi FE4 -> Network Y

Kodi Living -> TVH Provider X -> Satpi FE1 (Ok)
Ios Client -> TVH Provider X -> Satpi FE2 (Ok)
Now if Kodi Bedroom -> TVH Provider X -> Satpi FE3 (Not Ok)
Instead choosing FE3 it should return something like "No free adapters"

Barracuda09, 2024-02-16:
Yes now I get the picture, but I should think this is a TVH problem? it should know there are no free adapters for Provider X.

Questions:
- Am I wrong in assuming this?
- But what is TVH and requesting? could you give a SatPI log of this situation?

aurelmarius, 2024-02-16:
Oh, you are right. Yes, i just tested now and it seems that even there are no free adapters for Provider X tvh is still requesting stream from Satpi.
The tunner management in tvh is wrong because is treating all tunners from satpi server as a quad tunner not as separate tunners.

aurelmarius, 2024-02-16:
@Barracuda09 Later Edit: I think i found the problem. I was running DVBC-4 tunner configuration in TVH. If i set Tunner Configuration to auto now i see DVB-T frontend also and DVB-C doesn't respect FE number anymore because in TVH are more tunners than on satpi.
[image]
[image]

aurelmarius, 2024-02-16:
I solved it. Settings Tunner Configuration: DVBC-2,DVB-C2 works as expected

Barracuda09, 2024-02-16:
Yes your tuner can handle DVB-C/T/T2
So SatPI will say in http://192.168.10.247:8875/desc.xml
4 DVB-C tuners
4 DVB-T tuners
4 DVB-T2 tuners

This is all according to specs, but then TVH thinks there are then 12 tuners (maybe correctly?)

aurelmarius, 2024-02-16:
Yes, in desc.xml are 12 tunners.

http://192.168.10.247:8875/ DVBT-4,DVBT2-4,DVBC-4 /channellist.m3u

But on satpi web interface only 4, so frontend number from tvh doesn't match fe number in satpi.

[image]

The workaround is to set in tvh dvbc-2,dvbc-2. So now I have 2 tunners for provider x and 2 for provider y and everything works ok.

Barracuda09, 2024-02-16:
I am not sure you should call it a workaround. I think it is how it should work.

SatPI only advertise(capable of) that there are tuners:
4 DVB-C tuners
4 DVB-T tuners
4 DVB-T2 tuners

For SatPI it is only 4 tuners. And you should setup TVH what tuners to use and how to use them (which tuner type).

aurelmarius, 2024-02-16:
Ok, then i get it wrong from the beginning. Thank you for your help.

Barracuda09, 2024-02-16:
No problem, you're welcome!!

And thanks you for giving SatPI a go.

Barracuda09, 2024-02-21:
Will close this one, see https://github.com/Barracuda09/SATPI/issues/212#issuecomment-1948330615

closed 2024-02-21 by Barracuda09