Barracuda09/SATPI #213: Problems with multistream channels for ST2-88.0E

xxj737, 2024-03-27:
Hi! I'm trying to tune in china multistream channels from ST2-88.0E via SatPI and I've just encountered some problems. 
My SatPI server is installed on a zgemma h7s (with PnP DVB-S2x multistream tuner) and my client is for tvheadend and TransEdit_4_2  or  vlc media player.

ST2 88.0E multistream
11633H30000 PLSN 260334 GOLD MIS 038, 048, 052
11669H30000 PLSN 248232 GOLD MIS 034, 036, 040, 050

As you can see in the following pictures, I can't tune some channels.
[log.txt](https://github.com/Barracuda09/SATPI/files/14775871/log.txt)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

log  For example, 
rtsp://192.168.10.129:554/stream=101?fe=1&src=1&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232

Request failed, unable to connect

I would be grateful if you could fix these problems. Nowadays, there are more and more satellite TV channels using multistream technology.

Thanks in advance.

xxj737, 2024-03-27:
[image]
[image]
[image]
[image]

xxj737, 2024-03-27:
[image]

ZGEMMA H7S  OS Enigma2 Linux 

This channellist.m3u is not a list of programs received by the device. It is an invalid M3U file that defaults to installation.

Barracuda09, 2024-03-28:
Hi @xxj737 

Looking at the log:
```
  Wed Mar 27 13:28:45.3595 2024  6  RTSP Connection from 192.168.10.110 Port 50312 with fd: 9
  Wed Mar 27 13:28:45.3597 2024  7  RTSP Stream data from client None with IP 192.168.10.110 on Port 50312: SETUP rtsp://192.168.10.129/?src=1&fe=1&freq=11669&sr=30000&msys=dvbs2&mtype=qpsk&pol=h&ro=0.20&plts=on&isi=1&plsm=combo&plsc=248232&pids=0 RTSP/1.0
  Wed Mar 27 13:28:45.3597 2024  7  Transport: RTP/AVP/TCP;interleaved=0-1
  Wed Mar 27 13:28:45.3597 2024  7  CSeq: 1
  Wed Mar 27 13:28:45.3597 2024  7  <CRLF>
  Wed Mar 27 13:28:45.3599 2024  6  Found FrondtendID: 1 (fe=1)  StreamID: -1  SessionID: 0284952674
  Wed Mar 27 13:28:45.3601 2024  7  Frontend: 1, Found Streaming type: RTP/AVP/TCP
  Wed Mar 27 13:28:45.3602 2024  6  Frontend: 1, StreamClient with SessionID 0284952674 for dvbs2
  Wed Mar 27 13:28:45.3603 2024  6  Frontend: 1, Parsing transport parameters...
  Wed Mar 27 13:28:45.3604 2024  6  Frontend: 1, New frequency requested (0.0000 -> 11669.0000), clearing old channel data...
  Wed Mar 27 13:28:45.3607 2024  7  Frontend: 1, Parsing PID parameter: pids=0
  Wed Mar 27 13:28:45.3607 2024  7  Frontend: 1, Parsing transport parameters (Finished)
  Wed Mar 27 13:28:45.3608 2024  6  Frontend: 1, Updating frontend...
  Wed Mar 27 13:28:45.3609 2024  6  Frontend: 1, Closing all active PID filters...
  Wed Mar 27 13:28:45.4112 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 10 (0 ms)
  Wed Mar 27 13:28:45.4113 2024  6  Frontend: 1, Start tuning process for DVB-S(2)...
  Wed Mar 27 13:28:45.4114 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Writing DiSEqC command with fd: 10
  Wed Mar 27 13:28:45.7201 2024  7  Frontend: 1, Using LNB with: Lof-High=10600 MHz  Lof-Low=9750 MHz  Lof-Switch=11700 MHz -> IF=1919 MHz
  Wed Mar 27 13:28:45.7202 2024  6  Frontend: 1, Sending LNB: Mini-Switch Src: 0
  Wed Mar 27 13:28:45.8008 2024  7  Frontend: 1, Set Properties: Frequency 1919000
  Wed Mar 27 13:28:45.8009 2024  7  Frontend: 1, Set Properties: StreamID 130656257
  Wed Mar 27 13:28:45.8010 2024  6  Frontend: 1, Tuned, waiting on lock...
  Wed Mar 27 13:28:46.2527 2024  6  Frontend: 1, Not locked yet   (FE status 0x00)...
  Wed Mar 27 13:28:46.5559 2024  6  Frontend: 1, Not locked yet   (Timeout 1144 ms)...
  Wed Mar 27 13:28:46.5560 2024  6  Frontend: 1, Updating PID filters...
  Wed Mar 27 13:28:46.5561 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/demux0 using fd: 11
  Wed Mar 27 13:28:46.5565 2024  6  Frontend: 1, Set DMX buffer size to 3145728 Bytes
  Wed Mar 27 13:28:46.5567 2024  6  Frontend: 1, Set DMX_SET_SOURCE with (Src: 0 - Offset: 0)
  Wed Mar 27 13:28:46.5673 2024  7  Frontend: 1, Set filter PID: 0000
  Wed Mar 27 13:28:46.5775 2024  7  Frontend: 1, Set filter PID: 0001
  Wed Mar 27 13:28:46.5876 2024  7  Frontend: 1, Set filter PID: 0016
  Wed Mar 27 13:28:46.5978 2024  7  Frontend: 1, Set filter PID: 0017
  Wed Mar 27 13:28:46.6079 2024  7  Frontend: 1, Set filter PID: 0018
  Wed Mar 27 13:28:46.6081 2024  6  Frontend: 1, Updating frontend (Finished in 1247224 us)
  Wed Mar 27 13:28:46.6082 2024  6  Frontend: 1, RTP/TCP set network buffer size: 43 KBytes
  Wed Mar 27 13:28:46.6082 2024  6  Frontend: 1, Start RTP/TCP stream to 192.168.10.110:50312
  Wed Mar 27 13:28:46.6083 2024  6  Frontend: 1, Start RTCP/TCP stream to 192.168.10.110:50312
  Wed Mar 27 13:28:46.6086 2024  7  Frontend: 1, Start Reader and Monitor Thread
  Wed Mar 27 13:28:46.6086 2024  7  Send reply in 1248 ms
  Wed Mar 27 13:28:46.6086 2024  7  RTSP/1.0 200 OK
  Wed Mar 27 13:28:46.6086 2024  7  Server: satpi/1.6.2.199~g5b13ad1
  Wed Mar 27 13:28:46.6086 2024  7  CSeq: 1
  Wed Mar 27 13:28:46.6086 2024  7  Session: 0284952674;timeout=60
  Wed Mar 27 13:28:46.6086 2024  7  Transport: RTP/AVP/TCP;unicast;client_ip=192.168.10.110;interleaved=0-1
  Wed Mar 27 13:28:46.6086 2024  7  com.ses.streamID: 100
  Wed Mar 27 13:28:46.6086 2024  7  <CRLF>
  Wed Mar 27 13:28:46.6099 2024  7  RTSP Stream data from client None with IP 192.168.10.110 on Port 50312: PLAY rtsp://192.168.10.129/stream=100?addpids=1,16,17,1000,1001,1002,1003,8187 RTSP/1.0
  Wed Mar 27 13:28:46.6099 2024  7  Session: 0284952674
  Wed Mar 27 13:28:46.6099 2024  7  CSeq: 2
  Wed Mar 27 13:28:46.6099 2024  7  <CRLF>
  Wed Mar 27 13:28:46.6100 2024  6  Found FrondtendID: 1 (fe=1)  StreamID: 100  SessionID: 0284952674
  Wed Mar 27 13:28:46.6101 2024  6  Frontend: 1, StreamClient with SessionID 0284952674
  Wed Mar 27 13:28:46.6102 2024  6  Frontend: 1, Parsing transport parameters...
  Wed Mar 27 13:28:46.6113 2024  7  Frontend: 1, Parsing PID parameter: addpids=1,16,17,1000,1001,1002,1003,8187
  Wed Mar 27 13:28:46.6114 2024  7  Frontend: 1, Parsing transport parameters (Finished)
  Wed Mar 27 13:28:46.6115 2024  6  Frontend: 1, Updating frontend...
  Wed Mar 27 13:28:46.6115 2024  6  Frontend: 1, Updating PID filters...
  Wed Mar 27 13:28:46.6217 2024  7  Frontend: 1, Set filter PID: 1000
  Wed Mar 27 13:28:46.6319 2024  7  Frontend: 1, Set filter PID: 1001
  Wed Mar 27 13:28:46.6420 2024  7  Frontend: 1, Set filter PID: 1002
  Wed Mar 27 13:28:46.6522 2024  7  Frontend: 1, Set filter PID: 1003
  Wed Mar 27 13:28:46.6625 2024  7  Frontend: 1, Set filter PID: 8187
  Wed Mar 27 13:28:46.6626 2024  6  Frontend: 1, Updating frontend (Finished in 51060 us)
  Wed Mar 27 13:28:46.6626 2024  7  Send reply in 52 ms
  Wed Mar 27 13:28:46.6626 2024  7  RTSP/1.0 200 OK
[... cut ...]
  Wed Mar 27 13:28:48.7164 2024  7  Session: 0284952674
  Wed Mar 27 13:28:48.7164 2024  7  <CRLF>
  Wed Mar 27 13:28:48.7175 2024  6  RTSP Client 192.168.10.110:50312 Connection closed with fd: 9
```

I see request:
```
SETUP rtsp://192.168.10.129/?src=1&fe=1&freq=11669&sr=30000&msys=dvbs2&mtype=qpsk&pol=h&ro=0.20
                             &plts=on&isi=1&plsm=combo&plsc=248232&pids=0 RTSP/1.0
``` 
Which client is sending this request?  I see `plsm=combo`, in SatPI `plsm=` should be a number (0 = root and 1 = gold) and not combo

xxj737, 2024-03-29:
Dear @Barracuda09     
Thank you very much for your feedback
 The client is TransEdit_4.2  
 As you can see in the following pictures
Is there an error in the settings?
[image]
[image]
[image]
[image]
[Log.txt](https://github.com/Barracuda09/SATPI/files/14805919/Log.txt)
[0880.zip](https://github.com/Barracuda09/SATPI/files/14805923/0880.zip)

xxj737, 2024-03-29:
I tune 11669 H 3000 at VLC，I started playing 11699, but failed
  For example
rtsp://192.168.10.129:554/?fe=1&src=1&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187

[image]
[VLC LOG.txt](https://github.com/Barracuda09/SATPI/files/14805988/VLC.LOG.txt)

Barracuda09, 2024-03-29:
What is your satellite setup:
- Any DiSEqc switches
- Which type of LNBs

xxj737, 2024-03-29:
Dear @Barracuda09
--DiSEqc  1.0   port 3  (C)
--SI2169D  DVBS2X
[image]

xxj737, 2024-03-29:
SATPI>frontend.html

[image]

xxj737, 2024-03-29:
[image]

Barracuda09, 2024-03-29:
So if I understand correctly,  ST2-88.0E is on port 3?  So **src=3**

Then try in VLC this (src=3):
```
rtsp://192.168.10.129:554/?fe=1&src=3&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187
```

xxj737, 2024-03-30:
I will play the link again in VLC, but still couldn't tune
rtsp://192.168.10.129:554/?fe=1&src=3&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187
 [rtsp://192.168.10.129:554/?fe=1&src=3&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187](url)
[log 3-30.txt](https://github.com/Barracuda09/SATPI/files/14811339/log.3-30.txt)

Barracuda09, 2024-03-30:
**src=3** -> is the DiSEqc switches input were ST2-88.0E is connected to
**fe=1** -> is the frontend were ST2-88.0E is connected to
```
rtsp://192.168.10.129:554/?fe=1&src=3&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187
```
**But please, could you enable this**:
[image]

xxj737, 2024-04-01:
Hi  Barracuda09
Record the log as follows

  Sat Mar 30 09:08:33.6798 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Sat Mar 30 09:09:01.6831 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Sat Mar 30 09:09:02.6837 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Sat Mar 30 09:09:03.6844 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Sat Mar 30 09:09:23.7780 2024  6  RTSP Connection from 192.168.10.118 Port 44054 with fd: 9
  Sat Mar 30 09:09:23.7799 2024  7  RTSP Stream data from client None with IP 192.168.10.118 on Port 44054: OPTIONS rtsp://192.168.10.129:554/?fe=1&src=3&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187 RTSP/1.0
  Sat Mar 30 09:09:23.7799 2024  7  CSeq: 2
  Sat Mar 30 09:09:23.7799 2024  7  User-Agent: LibVLC/3.0.18-rc2 (LIVE555 Streaming Media v2016.11.28)
  Sat Mar 30 09:09:23.7799 2024  7  <CRLF>
  Sat Mar 30 09:09:23.7801 2024  7  Send reply in 0 ms
  Sat Mar 30 09:09:23.7801 2024  7  RTSP/1.0 200 OK
  Sat Mar 30 09:09:23.7801 2024  7  Server: satpi/1.6.2.199~g5b13ad1
  Sat Mar 30 09:09:23.7801 2024  7  CSeq: 2
  Sat Mar 30 09:09:23.7801 2024  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Sat Mar 30 09:09:23.7801 2024  7  <CRLF>
  Sat Mar 30 09:09:23.7868 2024  7  RTSP Stream data from client None with IP 192.168.10.118 on Port 44054: DESCRIBE rtsp://192.168.10.129:554/?fe=1&src=3&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187 RTSP/1.0
  Sat Mar 30 09:09:23.7868 2024  7  CSeq: 3
  Sat Mar 30 09:09:23.7868 2024  7  User-Agent: LibVLC/3.0.18-rc2 (LIVE555 Streaming Media v2016.11.28)
  Sat Mar 30 09:09:23.7868 2024  7  Accept: application/sdp
  Sat Mar 30 09:09:23.7868 2024  7  <CRLF>
  Sat Mar 30 09:09:23.7870 2024  7  Send reply in 0 ms
  Sat Mar 30 09:09:23.7870 2024  7  RTSP/1.0 404 Not Found
  Sat Mar 30 09:09:23.7870 2024  7  Server: satpi/1.6.2.199~g5b13ad1
  Sat Mar 30 09:09:23.7870 2024  7  CSeq: 3
  Sat Mar 30 09:09:23.7870 2024  7  Content-Type: application/sdp
  Sat Mar 30 09:09:23.7870 2024  7  Content-Base: rtsp://192.168.10.129/
  Sat Mar 30 09:09:23.7870 2024  7  Content-Length: 0
  Sat Mar 30 09:09:23.7870 2024  7  <CRLF>
  Sat Mar 30 09:09:23.7937 2024  6  RTSP Client 192.168.10.118:44054 Connection closed with fd: 9
  Sat Mar 30 09:09:23.7975 2024  6  RTSP Connection from 192.168.10.118 Port 44068 with fd: 9
  Sat Mar 30 09:09:23.7991 2024  7  RTSP Stream data from client None with IP 192.168.10.118 on Port 44068: SETUP rtsp://192.168.10.129:554/?fe=1&src=3&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187 RTSP/1.0
  Sat Mar 30 09:09:23.7991 2024  7  CSeq: 0
  Sat Mar 30 09:09:23.7991 2024  7  Transport: RTP/AVP;unicast;client_port=9324-9325
  Sat Mar 30 09:09:23.7991 2024  7  <CRLF>
  Sat Mar 30 09:09:23.7992 2024  6  Found FrondtendID: 1 (fe=1)  StreamID: -1  SessionID: 0268528852
  Sat Mar 30 09:09:23.7995 2024  7  Frontend: 1, Found Streaming type: RTP/AVP
  Sat Mar 30 09:09:23.7996 2024  6  Frontend: 1, StreamClient with SessionID 0268528852 for dvbs2
  Sat Mar 30 09:09:23.7996 2024  6  Frontend: 1, Parsing transport parameters...
  Sat Mar 30 09:09:23.7997 2024  6  Frontend: 1, New frequency requested (0.0000 -> 11669.0000), clearing old channel data...
  Sat Mar 30 09:09:23.7999 2024  7  Frontend: 1, Parsing PID parameter: pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187
  Sat Mar 30 09:09:23.8000 2024  7  Frontend: 1, Parsing transport parameters (Finished)
  Sat Mar 30 09:09:23.8001 2024  6  Frontend: 1, Updating frontend...
  Sat Mar 30 09:09:23.8001 2024  6  Frontend: 1, Closing all active PID filters...
  Sat Mar 30 09:09:23.8505 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 10 (0 ms)
  Sat Mar 30 09:09:23.8505 2024  6  Frontend: 1, Start tuning process for DVB-S(2)...
  Sat Mar 30 09:09:23.8505 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Writing DiSEqC command with fd: 10
  Sat Mar 30 09:09:24.1635 2024  7  Frontend: 1, Using LNB with: Lof-High=10600 MHz  Lof-Low=9750 MHz  Lof-Switch=11700 MHz -> IF=1919 MHz
  Sat Mar 30 09:09:24.1636 2024  6  Frontend: 1, Sending LNB: Mini-Switch Src: 2
  Sat Mar 30 09:09:24.2443 2024  7  Frontend: 1, Set Properties: Frequency 1919000
  Sat Mar 30 09:09:24.2443 2024  7  Frontend: 1, Set Properties: StreamID 130656292
  Sat Mar 30 09:09:24.2444 2024  6  Frontend: 1, Tuned, waiting on lock...
  Sat Mar 30 09:09:24.6961 2024  6  Frontend: 1, Not locked yet   (FE status 0x00)...
  Sat Mar 30 09:09:24.9994 2024  6  Frontend: 1, Not locked yet   (Timeout 1148 ms)...
  Sat Mar 30 09:09:24.9995 2024  6  Frontend: 1, Updating PID filters...
  Sat Mar 30 09:09:24.9996 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/demux0 using fd: 11
  Sat Mar 30 09:09:25.0000 2024  6  Frontend: 1, Set DMX buffer size to 3145728 Bytes
  Sat Mar 30 09:09:25.0002 2024  6  Frontend: 1, Set DMX_SET_SOURCE with (Src: 0 - Offset: 0)
  Sat Mar 30 09:09:25.0107 2024  7  Frontend: 1, Set filter PID: 0000
[... cut ...]
  Sat Mar 30 09:10:20.3695 2024  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Sat Mar 30 09:10:20.3695 2024  7  Session: 0268528852
  Sat Mar 30 09:10:20.3695 2024  7  <CRLF>
  Sat Mar 30 09:10:31.6978 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Sat Mar 30 09:10:32.7010 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Sat Mar 30 09:10:33.7018 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back

Barracuda09, 2024-04-01:
Here (https://github.com/Barracuda09/SATPI/issues/213#issuecomment-2027516028) I see you did not setup a DiSEqC Switch.
You need to setup the DiSEqC Switch like so (if you use a switch):
[image]

xxj737, 2024-04-01:
I will provide the logs again

rtsp://192.168.10.129:554/?fe=1&src=3&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187

[2024-4-1 log.txt](https://github.com/Barracuda09/SATPI/files/14823712/2024-4-1.log.txt)

`  Mon Apr  1 21:01:05.9770 2024  6  --- Starting SatPI version: 1.6.2.199~g5b13ad1 ---
  Mon Apr  1 21:01:05.9775 2024  6  Number of processors online: 2
  Mon Apr  1 21:01:05.9776 2024  6  Default network buffer size: 80 KBytes
  Mon Apr  1 21:01:05.9782 2024  6  eth0: 192.168.10.129 [00:17:9a:2b:02:44]
  Mon Apr  1 21:01:05.9784 2024  6  Enumerating all devices...
  Mon Apr  1 21:01:05.9785 2024  6  Detecting frontends in: /dev/dvb
  Mon Apr  1 21:01:05.9808 2024  6  Frontend Name: DVB-S2
  Mon Apr  1 21:01:05.9809 2024  6  Frontend Stat: Use advanced signal stats
  Mon Apr  1 21:01:05.9809 2024  6  Frontend Type: Satellite (DVB-S)
  Mon Apr  1 21:01:05.9809 2024  6  Frontend Type: Satellite (DVB-S2)
  Mon Apr  1 21:01:05.9809 2024  6  Frontend Freq: 950000 Hz to 2150000 Hz
  Mon Apr  1 21:01:05.9810 2024  6  Frontend srat: 0 symbols/s to 45000000 symbols/s
  Mon Apr  1 21:01:05.9810 2024  6  Frontend  API: 5.10 (0x050A)
  Mon Apr  1 21:01:05.9812 2024  6  Frontend  FBC: Frontend 1  Root 0  Index 0(A)  Offset -8  Connected 0  Linked 0
  Mon Apr  1 21:01:05.9829 2024  6  Frontend Name: DVB-S2
  Mon Apr  1 21:01:05.9829 2024  6  Frontend Stat: Use advanced signal stats
  Mon Apr  1 21:01:05.9829 2024  6  Frontend Type: Satellite (DVB-S)
  Mon Apr  1 21:01:05.9830 2024  6  Frontend Type: Satellite (DVB-S2)
  Mon Apr  1 21:01:05.9830 2024  6  Frontend Freq: 950000 Hz to 2150000 Hz
  Mon Apr  1 21:01:05.9830 2024  6  Frontend srat: 0 symbols/s to 45000000 symbols/s
  Mon Apr  1 21:01:05.9831 2024  6  Frontend  API: 5.10 (0x050A)
  Mon Apr  1 21:01:05.9832 2024  6  Frontend  FBC: Frontend 2  Root 0  Index 1(B)  Offset -8  Connected 0  Linked 0
  Mon Apr  1 21:01:05.9847 2024  6  Frontend Name: DVB-T/T2/C
  Mon Apr  1 21:01:05.9847 2024  6  Frontend Stat: Use advanced signal stats
  Mon Apr  1 21:01:05.9848 2024  6  Frontend Type: Terrestrial (DVB-T2)
  Mon Apr  1 21:01:05.9848 2024  6  Frontend Type: Terrestrial (DVB-T)
  Mon Apr  1 21:01:05.9848 2024  6  Frontend Type: Cable (Annex A)
  Mon Apr  1 21:01:05.9848 2024  6  Frontend Type: Cable (Annex C)
  Mon Apr  1 21:01:05.9849 2024  6  Frontend Freq: 48250000 Hz to 863250000 Hz
  Mon Apr  1 21:01:05.9849 2024  6  Frontend srat: 0 symbols/s to 9360000 symbols/s
  Mon Apr  1 21:01:05.9849 2024  6  Frontend  API: 5.10 (0x050A)
  Mon Apr  1 21:01:05.9851 2024  6  Frontend  FBC: Frontend 3  Root 0  Index 2(C)  Offset -8  Connected 0  Linked 0
  Mon Apr  1 21:01:05.9855 2024  6  Frontends found: 3
  Mon Apr  1 21:01:05.9855 2024  6  Setting up TS Reader using path: /usr/bin
  Mon Apr  1 21:01:05.9869 2024  6  Setting up TS Streamer
  Mon Apr  1 21:01:05.9964 2024  6  Setting HTTP Port to: 8875
  Mon Apr  1 21:01:05.9965 2024  6  Setting RTSP Port to: 554
  Mon Apr  1 21:01:05.9965 2024  6  Setting WEB Path to: /usr/share/satpi/web
  Mon Apr  1 21:01:05.9965 2024  6  Setting App Data Path to: /usr/bin
  Mon Apr  1 21:01:06.0023 2024  6  Setting up RTSP server
  Mon Apr  1 21:01:06.0081 2024  6  Setting up SSDP server with BOOTID: 33  annouce interval: 60 Sec  TTL: 1
  Mon Apr  1 21:01:06.0085 2024  6  Setting up HTTP server
  Mon Apr  1 21:01:06.0572 2024  6  HTTP Connection from 192.168.10.119 Port 38520 with fd: 8
  Mon Apr  1 21:01:06.0594 2024  7  Client: 192.168.10.119 requested desc.xml
  Mon Apr  1 21:01:06.4535 2024  6  HTTP Connection from 192.168.10.101 Port 24614 with fd: 9
  Mon Apr  1 21:01:06.4538 2024  7  Client: 192.168.10.101 requested desc.xml
  Mon Apr  1 21:01:11.8647 2024  6  HTTP Connection from 192.168.10.101 Port 5300 with fd: 10
  Mon Apr  1 21:01:11.8649 2024  6  HTTP Connection from 192.168.10.101 Port 14907 with fd: 11
  Mon Apr  1 21:01:12.0306 2024  6  HTTP Connection from 192.168.10.101 Port 4942 with fd: 12
  Mon Apr  1 21:01:12.0309 2024  6  HTTP Connection from 192.168.10.101 Port 27704 with fd: 13
  Mon Apr  1 21:01:12.0358 2024  6  HTTP Connection from 192.168.10.101 Port 48058 with fd: 14
  Mon Apr  1 21:01:12.0453 2024  6  HTTP Connection from 192.168.10.101 Port 44917 with fd: 15
  Mon Apr  1 21:01:18.0159 2024  6  HTTP Connection from 192.168.10.101 Port 28773 with fd: 16
  Mon Apr  1 21:01:18.0160 2024  6  HTTP Connection from 192.168.10.101 Port 3204 with fd: 17
  Mon Apr  1 21:01:18.0161 2024  6  HTTP Connection from 192.168.10.101 Port 4788 with fd: 18
  Mon Apr  1 21:01:18.0164 2024  6  HTTP Connection from 192.168.10.101 Port 2157 with fd: 19
  Mon Apr  1 21:01:25.8664 2024  6  HTTP Client 192.168.10.119:38520 Connection closed with fd: 8
  Mon Apr  1 21:01:31.9989 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Mon Apr  1 21:01:32.9997 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Mon Apr  1 21:01:34.0004 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Mon Apr  1 21:02:02.0036 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Mon Apr  1 21:02:03.0045 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
[... cut ...]
  Mon Apr  1 21:05:00.1525 2024  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Mon Apr  1 21:05:00.1525 2024  7  Session: 0252988701
  Mon Apr  1 21:05:00.1525 2024  7  <CRLF>
  Mon Apr  1 21:05:02.0374 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Mon Apr  1 21:05:03.0415 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Mon Apr  1 21:05:04.0416 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
`

xxj737, 2024-04-01:
[image]

xxj737, 2024-04-01:
HI Barracuda09 
Very good!
Thank you for your guidance！

[image]
[image]

xxj737, 2024-04-01:
There is a new bug
  rstp===Many mosaics
rtsp://192.168.10.129:554/?fe=1&src=3&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187

http====is  ok，Normal tuning

http://192.168.10.129:8875/?fe=1&src=3&freq=11669&pol=h&msys=dvbs2&sr=30000&fec=35&ro=0.35&mtype=qpsk&plts=off&isi=036&plsc=248232&pids=0,1,16,17,20,1000,1001,1002,1003,1010,1020,1030,8187

Barracuda09, 2024-04-01:
Hi @xxj737 

Are you using VLC? Then try to follow these steps https://rg4.net/archives/1104.html to enable `RTP over TCP`

xxj737, 2024-04-04:
HI  Barracuda09
Thank you, setup completed

xxj737, 2024-04-04:
HI Barracuda09
When my client is TransEdit
How to use it?

TransEdit_4.2
[Log.txt](https://github.com/Barracuda09/SATPI/files/14874437/Log.txt)

[image]
[image]

`
  Fri Apr  5 00:38:41.0944 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:38:42.0952 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:38:43.0960 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:39:11.0974 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:39:12.0982 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:39:13.0989 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:39:41.1015 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:39:42.1022 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:39:43.1030 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:39:54.7991 2024  6  HTTP Client 192.168.10.101:2120 Connection closed with fd: 8
  Fri Apr  5 00:39:55.5622 2024  6  HTTP Connection from 192.168.10.101 Port 2383 with fd: 8
  Fri Apr  5 00:40:04.8110 2024  6  HTTP Client 192.168.10.101:2383 Connection closed with fd: 8
  Fri Apr  5 00:40:05.5489 2024  6  HTTP Connection from 192.168.10.101 Port 32866 with fd: 8
  Fri Apr  5 00:40:11.1051 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:40:12.1059 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:40:13.1070 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:40:14.8169 2024  6  HTTP Client 192.168.10.101:32866 Connection closed with fd: 8
  Fri Apr  5 00:40:15.5585 2024  6  HTTP Connection from 192.168.10.101 Port 17395 with fd: 8
  Fri Apr  5 00:40:32.0032 2024  6  RTSP Connection from 192.168.10.101 Port 8290 with fd: 9
  Fri Apr  5 00:40:32.0033 2024  6  RTSP Client 192.168.10.101:8290 Connection closed with fd: 9
  Fri Apr  5 00:40:32.1130 2024  6  RTSP Connection from 192.168.10.101 Port 8292 with fd: 9
  Fri Apr  5 00:40:32.1131 2024  6  RTSP Client 192.168.10.101:8292 Connection closed with fd: 9
  Fri Apr  5 00:40:32.1177 2024  6  RTSP Connection from 192.168.10.101 Port 8293 with fd: 9
  Fri Apr  5 00:40:32.1177 2024  7  RTSP Stream data from client None with IP 192.168.10.101 on Port 8293: OPTIONS rtsp://192.168.10.129:554/ RTSP/1.0
  Fri Apr  5 00:40:32.1177 2024  7  CSeq: 1
  Fri Apr  5 00:40:32.1177 2024  7  <CRLF>
  Fri Apr  5 00:40:32.1178 2024  7  Send reply in 0 ms
  Fri Apr  5 00:40:32.1178 2024  7  RTSP/1.0 200 OK
  Fri Apr  5 00:40:32.1178 2024  7  Server: satpi/1.6.2.199~g5b13ad1
  Fri Apr  5 00:40:32.1178 2024  7  CSeq: 1
  Fri Apr  5 00:40:32.1178 2024  7  Public: OPTIONS, DESCRIBE, SETUP, PLAY, TEARDOWN
  Fri Apr  5 00:40:32.1178 2024  7  <CRLF>
  Fri Apr  5 00:40:32.1392 2024  7  RTSP Stream data from client None with IP 192.168.10.101 on Port 8293: SETUP rtsp://192.168.10.129:554/?src=3&freq=11669&msys=dvbs2&plts=on&fec=35&pol=h&ro=0.35&sr=30000&mtype=qpsk RTSP/1.0
  Fri Apr  5 00:40:32.1392 2024  7  CSeq: 2
  Fri Apr  5 00:40:32.1392 2024  7  Transport: RTP/AVP;unicast;client_port=52034-52035
  Fri Apr  5 00:40:32.1392 2024  7  <CRLF>
  Fri Apr  5 00:40:32.1394 2024  6  Found FrondtendID: x (fe=x)  StreamID: x  SessionID: 0275405126  New Session: true
  Fri Apr  5 00:40:32.1396 2024  7  Frontend: 1, Found Streaming type: RTP/AVP
  Fri Apr  5 00:40:32.1397 2024  6  Frontend: 1, StreamClient with SessionID 0275405126 for dvbs2
  Fri Apr  5 00:40:32.1397 2024  6  Frontend: 1, Parsing transport parameters...
  Fri Apr  5 00:40:32.1398 2024  6  Frontend: 1, New frequency requested (0.0000 -> 11669.0000), clearing old channel data...
  Fri Apr  5 00:40:32.1400 2024  7  Frontend: 1, Parsing transport parameters (Finished)
  Fri Apr  5 00:40:32.1401 2024  6  Frontend: 1, Updating frontend...
  Fri Apr  5 00:40:32.1402 2024  6  Frontend: 1, Closing all active PID filters...
  Fri Apr  5 00:40:32.1906 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Read/Write with fd: 10 (0 ms)
  Fri Apr  5 00:40:32.1907 2024  6  Frontend: 1, Start tuning process for DVB-S(2)...
  Fri Apr  5 00:40:32.1907 2024  6  Frontend: 1, Opened /dev/dvb/adapter0/frontend0 for Writing DiSEqC command with fd: 10
  Fri Apr  5 00:40:32.5110 2024  7  Frontend: 1, Using LNB with: Lof-High=10600 MHz  Lof-Low=9750 MHz  Lof-Switch=11700 MHz -> IF=1919 MHz
  Fri Apr  5 00:40:32.5111 2024  6  Frontend: 1, Sending DiSEqC: [0xE0] [0x10] [0x38] [0xFA] - DiSEqC Src: 2
  Fri Apr  5 00:40:32.6196 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Fri Apr  5 00:40:32.8282 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Fri Apr  5 00:40:33.0367 2024  6  Frontend: 1, Sending DiSEqC: Command Repeated
  Fri Apr  5 00:40:33.2705 2024  7  Frontend: 1, Set Properties: Frequency 1919000
  Fri Apr  5 00:40:33.2707 2024  6  Frontend: 1, Tuned, waiting on lock...
[... cut ...]
  Fri Apr  5 00:41:12.1124 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:41:13.1138 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:41:41.1172 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 00:41:42.1170 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
`

Barracuda09, 2024-04-04:
Hi @xxj737 ,

Is this log from client TransEdit?

With this request:
```
  Fri Apr  5 00:40:32.1392 2024  7  RTSP Stream data from client None with IP 192.168.10.101 on Port 8293: SETUP rtsp://192.168.10.129:554/?src=3&freq=11669&msys=dvbs2&plts=on&fec=35&pol=h&ro=0.35&sr=30000&mtype=qpsk RTSP/1.0
  Fri Apr  5 00:40:32.1392 2024  7  CSeq: 2
  Fri Apr  5 00:40:32.1392 2024  7  Transport: RTP/AVP;unicast;client_port=52034-52035
  Fri Apr  5 00:40:32.1392 2024  7  <CRLF>
```
Has this SETUP:
```
SETUP rtsp://192.168.10.129:554/?
            src=3&freq=11669&
            msys=dvbs2&plts=on&fec=35&
            pol=h&ro=0.35&
            sr=30000&mtype=qpsk RTSP/1.0
```

Here is no `isi=036&plsc=248232` so probably there will be no TS packets

xxj737, 2024-04-05:
Hi @Barracuda09 
yes,the client is TransEdit，How to set it up?   isi=036&plsc=248232  

  pids=all?

xxj737, 2024-04-05:
1,   Tvheadend sat>ip  service--- client is TransEdit
[image]
[image]

2.Satpi   service------ client is TransEdit

xxj737, 2024-04-05:
2.Satpi service------ client is TransEdit

[image]
[image]

SATPI  LOG  ---[log.txt](https://github.com/Barracuda09/SATPI/files/14883715/log.txt)

xxj737, 2024-04-05:
I think it's a problem with SATPI
I used DVB-C, normal program stream
It still doesn't seem to work

xxj737, 2024-04-05:
The TransEdit   only sends instructions----- Analyze

xxj737, 2024-04-05:
[TransEdit_4_2_1_1.zip](https://github.com/Barracuda09/SATPI/files/14883869/TransEdit_4_2_1_1.zip)

Barracuda09, 2024-04-05:
There is no Lock to frequency 267 MHz see around `Fri Apr  5 18:25:50.5808 2024  6` see below:
```
  Fri Apr  5 18:25:49.5191 2024  7  RTSP Stream data from client None with IP 192.168.10.101 on Port 37592: SETUP rtsp://192.168.10.129:554/?freq=267&msys=dvbc&sr=6875&specinv=0&bw=8&mtype=64qam RTSP/1.0
  Fri Apr  5 18:25:49.5191 2024  7  CSeq: 2
  Fri Apr  5 18:25:49.5191 2024  7  Transport: RTP/AVP;unicast;client_port=52042-52043
  Fri Apr  5 18:25:49.5191 2024  7  <CRLF>
  Fri Apr  5 18:25:49.5193 2024  6  Found FrondtendID: x (fe=x)  StreamID: x  SessionID: 0260524140  New Session: true
  Fri Apr  5 18:25:49.5196 2024  6  Frontend: 1, Not capable of handling msys=dvbc with freq=267.0000
  Fri Apr  5 18:25:49.5198 2024  6  Frontend: 2, Not capable of handling msys=dvbc with freq=267.0000
  Fri Apr  5 18:25:49.5200 2024  7  Frontend: 3, Found Streaming type: RTP/AVP
  Fri Apr  5 18:25:49.5200 2024  6  Frontend: 3, StreamClient with SessionID 0260524140 for dvbc
  Fri Apr  5 18:25:49.5201 2024  6  Frontend: 3, Parsing transport parameters...
  Fri Apr  5 18:25:49.5202 2024  6  Frontend: 3, New frequency requested (0.0000 -> 267.0000), clearing old channel data...
  Fri Apr  5 18:25:49.5204 2024  7  Frontend: 3, Parsing transport parameters (Finished)
  Fri Apr  5 18:25:49.5206 2024  6  Frontend: 3, Updating frontend...
  Fri Apr  5 18:25:49.5206 2024  6  Frontend: 3, Closing all active PID filters...
  Fri Apr  5 18:25:49.5710 2024  6  Frontend: 3, Opened /dev/dvb/adapter0/frontend2 for Read/Write with fd: 16 (0 ms)
  Fri Apr  5 18:25:49.5711 2024  6  Frontend: 3, Start tuning process for DVB-C(2)...
  Fri Apr  5 18:25:49.5711 2024  7  Frontend: 3, Set Properties: Frequency 267000000
  Fri Apr  5 18:25:50.2806 2024  6  Frontend: 3, Tuned, waiting on lock...
  Fri Apr  5 18:25:50.2925 2024  6  Root Device Client 192.168.10.119 [Unkown]: tries to discover the network, sending reply back
  Fri Apr  5 18:25:50.5808 2024  6  Frontend: 3, Not locked yet   (Timeout 1009 ms)...  <-#-#-#- NO LOCK to 267 MHz -#-#-
  Fri Apr  5 18:25:50.5809 2024  6  Frontend: 3, Updating frontend (Finished in 1060332 us)
  Fri Apr  5 18:25:50.5811 2024  6  Frontend: 3, RTP/UDP set network buffer size: 160 KBytes
  Fri Apr  5 18:25:50.5811 2024  6  Frontend: 3, Start RTP/UDP stream to 192.168.10.101:52042
  Fri Apr  5 18:25:50.5811 2024  6  Frontend: 3, Start RTCP/UDP stream to 192.168.10.101:52043
  Fri Apr  5 18:25:50.5814 2024  7  Frontend: 3, Start Reader and Monitor Thread
  Fri Apr  5 18:25:50.5815 2024  7  Send reply in 1062 ms
  Fri Apr  5 18:25:50.5815 2024  7  RTSP/1.0 200 OK
```
That `TS Analyzer` is only showing `Null Packets` is because the tuning failed (maybe because of wrong parameters?)
Could you try VLC again for this situation?

xxj737, 2024-04-05:
> There is no Lock to frequency 267 MHz see around `Fri Apr 5 18:25:50.5808 2024  6` see below:
>
> ```
>   Fri Apr  5 18:25:49.5191 2024  7  RTSP Stream data from client None with IP 192.168.10.101 on [...]

VLC is  OK!
Use this link
`rtsp://192.168.10.129:554/?fe=1&freq=267&sr=6875&msys=dvbc&mtype=64qam&fec=35&pids=0,1,16,17,18,20,32,34,3511,3512,3519
`
[image]
Only one program is playing normally, 
but with a frequency of 267MHz, it contains four channels

pids=0,1,16,17,18,20,32,34,3511,3512,3519----channels  A
pids=all------channels  A+B+C+D   ??

[log.txt](https://github.com/Barracuda09/SATPI/files/14887789/log.txt)

Barracuda09, 2024-04-05:
`pids=all` seems not to be supported (see around `Sat Apr  6 00:26:20.3437 2024`) :
```
  Sat Apr  6 00:26:20.3423 2024  7  RTSP Stream data from client None with IP 192.168.10.101 on Port 52865: PLAY rtsp://192.168.10.129:554/stream=102?pids=all RTSP/1.0
  Sat Apr  6 00:26:20.3423 2024  7  CSeq: 3
  Sat Apr  6 00:26:20.3423 2024  7  Session: 0283745491
  Sat Apr  6 00:26:20.3423 2024  7  <CRLF>
  Sat Apr  6 00:26:20.3424 2024  6  Found FrondtendID: 3 (fe=3)  StreamID: 102  SessionID: 0283745491
  Sat Apr  6 00:26:20.3425 2024  6  Frontend: 3, StreamClient with SessionID 0283745491
  Sat Apr  6 00:26:20.3425 2024  6  Frontend: 3, Parsing transport parameters...
  Sat Apr  6 00:26:20.3426 2024  7  Frontend: 3, Parsing PID parameter: pids=all
  Sat Apr  6 00:26:20.3428 2024  7  Frontend: 3, Parsing transport parameters (Finished)
  Sat Apr  6 00:26:20.3428 2024  6  Frontend: 3, Updating frontend...
  Sat Apr  6 00:26:20.3429 2024  6  Frontend: 3, Updating PID filters...
  Sat Apr  6 00:26:20.3430 2024  6  Frontend: 3, Opened /dev/dvb/adapter0/demux2 using fd: 12
  Sat Apr  6 00:26:20.3434 2024  6  Frontend: 3, Set DMX buffer size to 3145728 Bytes
  Sat Apr  6 00:26:20.3436 2024  6  Frontend: 3, Set DMX_SET_SOURCE with (Src: 2 - Offset: 0)
  Sat Apr  6 00:26:20.3437 2024  3  Frontend: 3, Failed to set DMX_SET_PES_FILTER for PID: 8192: Operation not supported (code 95)
  Sat Apr  6 00:26:20.3438 2024  6  Frontend: 3, Updating frontend (Finished in 886 us)
  Sat Apr  6 00:26:20.3438 2024  7  Send reply in 1 ms
  Sat Apr  6 00:26:20.3438 2024  7  RTSP/1.0 200 OK
  Sat Apr  6 00:26:20.3438 2024  7  Server: satpi/1.6.2.199~g5b13ad1
  Sat Apr  6 00:26:20.3438 2024  7  RTP-Info: url=rtsp://192.168.10.129/stream=102
  Sat Apr  6 00:26:20.3438 2024  7  CSeq: 3
  Sat Apr  6 00:26:20.3438 2024  7  Session: 0283745491
  Sat Apr  6 00:26:20.3438 2024  7  Range: npt=0.000-
  Sat Apr  6 00:26:20.3438 2024  7  <CRLF>
```

To see other channel B,C and D, you need to specifies for each channel the correct pids.

xxj737, 2024-04-06:
343

xxj737, 2024-04-06:
Tvheadend sat>ip service--- client is TransEdit 4.2

[image]

The instruction from TransEdit contains-- pids=all

How to fix the SATPI support directive ---pids=all?

xxj737, 2024-04-06:
Tvheadend  log

[log.txt](https://github.com/Barracuda09/SATPI/files/14892151/log.txt)

Barracuda09, 2024-04-06:
> The instruction from TransEdit contains-- pids=all
> 
> How to fix the SATPI support directive ---pids=all?

SatPI does try to set `pids=all`, but for some reason this error shows up in the log:
```
  Sat Apr  6 00:26:20.3437 2024  3  Frontend: 3, Failed to set DMX_SET_PES_FILTER for PID: 8192: Operation not supported (code 95)

```
And this gives me the impression that this is not supported by the driver of the tuner. Or at least the way SatPI is setting up the `pids=all` with `DMX_SET_PES_FILTER`.

xxj737, 2024-04-07:
Failed to set DMX_SET_PES_FILTER for PID: 8192: Operation not supported (code 95)

----Is this an internal program issue with SATPI?
Can you correct this error？
Enable SATPI support   pids=all

xxj737, 2024-04-07:
DMX_SET_PES_FILTER 

`https://linuxtv.org/downloads/v4l-dvb-apis/userspace-api/dvb/headers.html`

`#define [DMX_START](https://linuxtv.org/downloads/v4l-dvb-apis/userspace-api/dvb/dmx-start.html#dmx-start)                _IO('o', 41)
#define [DMX_STOP](https://linuxtv.org/downloads/v4l-dvb-apis/userspace-api/dvb/dmx-stop.html#dmx-stop)                 _IO('o', 42)
#define [DMX_SET_FILTER](https://linuxtv.org/downloads/v4l-dvb-apis/userspace-api/dvb/dmx-set-filter.html#dmx-set-filter)           _IOW('o', 43, struct dmx_sct_filter_params)
#define [DMX_SET_PES_FILTER](https://linuxtv.org/downloads/v4l-dvb-apis/userspace-api/dvb/dmx-set-pes-filter.html#dmx-set-pes-filter)       _IOW('o', 44, struct dmx_pes_filter_params)
#define [DMX_SET_BUFFER_SIZE](https://linuxtv.org/downloads/v4l-dvb-apis/userspace-api/dvb/dmx-set-buffer-size.html#dmx-set-buffer-size)      _IO('o', 45)
#define [DMX_GET_PES_PIDS](https://linuxtv.org/downloads/v4l-dvb-apis/userspace-api/dvb/dmx-get-pes-pids.html#dmx-get-pes-pids)         _IOR('o', 47, __u16[5])
#define [DMX_GET_STC](https://linuxtv.org/downloads/v4l-dvb-apis/userspace-api/dvb/dmx-get-stc.html#dmx-get-stc)              _IOWR('o', 50, struct dmx_stc)
#define [DMX_ADD_PID](https://linuxtv.org/downloads/v4l-dvb-apis/userspace-api/dvb/dmx-add-pid.html#dmx-add-pid)              _IOW('o', 51, __u16)
#define [DMX_REMOVE_PID](https://linuxtv.org/downloads/v4l-dvb-apis/userspace-api/dvb/dmx-remove-pid.html#dmx-remove-pid)           _IOW('o', 52, __u16)`

Barracuda09, 2024-04-08:
SatPI can handle `pids=all`

This `Failed to set DMX_SET_PES_FILTER for PID: 8192: Operation not supported (code 95)` is an Linux error probably from the tuner driver.

xxj737, 2024-04-08:
SATPI have DOCKER and NAS Installation Program?

I would like to try this feature
I have a  synology NAS---------x86  liunx  4.4.180 gcc7.5

Barracuda09, 2024-04-08:
Sorry, I have no experience with building Docker images.