Barracuda09/SATPI #222: Latest bin for Dinobot 4K Plus

PrvtN00b, 2025-01-17:
I tried to compile it by myself, but I'm totally illiterate cross compiling files, so anyone with a bin that includes OSCAM too I would be more than welcome.

Thanks!

PrvtN00b, 2025-01-17:
i just compiled it for debian 12, but I have no clue of how to do it for an enigma 2 STB

[image]

PrvtN00b, 2025-01-18:
Try today and ran it like this, with the same result, it won start on my Enigma 2 STB.

[image]

Barracuda09, 2025-01-18:
Hi @PrvtN00b 

Which CPU is in the Dinobot 4K?

Did you look at https://github.com/Barracuda09/SATPI/wiki/Cross-Compiling-SatPI

PrvtN00b, 2025-01-18:
> Hi [@PrvtN00b](https://github.com/PrvtN00b)
> 
> Which CPU is in the Dinobot 4K?
>
> Did you look at https://github.com/Barracuda09/SATPI/wiki/Cross-Compiling-SatPI

CPU:

[image]

Did you look at https://github.com/Barracuda09/SATPI/wiki/Cross-Compiling-SatPI -----> yes i did read it, but in my case instructions aren´t clear, even using chatgpt, manage to compile it for my debian PC, but still no go for this Enigma 2 STB.

Barracuda09, 2025-01-18:
So basically it is an ARM core, just like Vuplus etc.
 
So you could use this compiler, did you download an ARM compiler? this one for example:
 - https://github.com/Broadcom/stbgcc-8.3/releases
 
Did you unpack it in (home dir **~**) like:
- **~**/vu/opt/

Then run this on the command line, in the SatPI dir:
- source envARM.sh

Then run first this to see if it compiles and runs:
- make ENIGMA=yes

PrvtN00b, 2025-01-20:
this is how I did it.

on a Debian 12 PC

- apt install gcc-arm-linux-gnueabihf g++-arm-linux-gnueabihf

confirm compiler is install:

[image]

- nano /root/satpi/envARM.sh

`#!/bin/sh

# use:
#    source ./env.sh [gcc12]
#
# CFLAGS as used for Vu+ uno 4K

# ARM based Chips
export CPU_FLAGS="-march=armv7-a -mtune=cortex-a15 -mfpu=neon-vfpv4"

if [[ -n $1 && $1 = "gcc12" ]] ; then
  export CXXPREFIX=arm-linux-gnueabihf-
  export CXXSUFFIX=-12
  export CXX=$CXXPREFIX"g++"$CXXSUFFIX
elif [[ -n $1 && $1 = "gcc" ]] ; then
  export CXXPREFIX="arm-linux-gnueabihf-"
  export CXXSUFFIX=
  export CXX=$CXXPREFIX"g++"$CXXSUFFIX
else
  export CXXPREFIX="arm-linux-gnueabihf-"
  export CXXSUFFIX=
  export CXX=$CXXPREFIX"g++"$CXXSUFFIX
fi
echo Using: $CXX
$CXX --version`

from my OpenATV copy from /usr/lib/libdvbcsa.so.1.1.0 rename to libdvbcsa.so the paste to the debian PC

ran  source envARM.sh then ranmake LIBDVBCSA=yes

[satpi-OpenATV - OSCAM.zip](https://github.com/user-attachments/files/18471343/satpi-OpenATV.-.OSCAM.zip)