Barracuda09/SATPI #224: OpenATV SATPI - No Showing/Broadcasting IP

PrvtN00b, 2025-01-24:
Any solution for this?, the only way so far I can get it working is by clicking the restart button, it doesn't matter if I build a *.sh script to reboot it everytime the box boots up

[image]

PrvtN00b, 2025-01-24:
here is an example, after clicking restart, it now shows the IP and also tvheadend can see it now.

[image]

[image]

lordduckstein, 2025-02-17:
Do you use tvheadend in docker?

Then you have do add in environment  name= "RUN_OPTS"  and value = "--satip_xml http://192.168.50.92:8875/desc.xml"

So tvheadend know then were is satip

LG
Lordi

PrvtN00b, 2025-02-24:
> Do you use tvheadend in docker?
> 
> Then you have do add in environment name= "RUN_OPTS" and value = "--satip_xml http://192.168.50.92:8875/desc.xml"
> 
> So tvheadend know then were is satip
> 
> LG Lordi

no, I don´t use tvheadend in docker

Barracuda09, 2025-02-24:
Probably the eth0 is not initialized at the point SatPI is started.

lgm75, 2025-07-22:
Same problem here after a power outage or electrical shutdown...