Barracuda09/SATPI #230: Every day the same problem, only a restart helps

Jimbo888888, 2025-05-11:
After a certain period of non-use of SatPi and TVHeadend (usually the next day), SatPi always stops working. As soon as I try to watch a channel (free or encrypted), the tuner in the front-end overview turns blue, but not green, and I can't watch TV. Only restarting Enigma2 (OpenATV 7.5.1) solves the problem.

SatPi Version: 1.6.2.204~gfe180a5

I would like to attach a log file, but it is really difficult to capture everything from the log window.

Barracuda09, 2025-05-11:
You can send the log to syslog, see attached picture and enable:
- Log messages to syslog
- Log Debug messages

[image]

Jimbo888888, 2025-05-14:
It happened again. I previously enabled syslog, but where is the log saved?

Barracuda09, 2025-05-15:
For some reason the tuner locks but does not give data in this situation.

Could you give a log when it is working.

Jimbo888888, 2025-05-16:
Hello. I restarted the VU+ and shortly afterward successfully ran a stream for about a minute. Here's the log file.

[messages.zip](https://github.com/user-attachments/files/20243027/messages.zip)

Jimbo888888, 2025-05-18:
Today it happened again, tuners are only displayed blue and there is no signal. This time I only restarted Satpi. After that, the tuner is not even more than blue when I open a channel. Nothing happens at all.

Here is the log:

[messages.0.zip](https://github.com/user-attachments/files/20272577/messages.0.zip)
[messages.zip](https://github.com/user-attachments/files/20272578/messages.zip)

Barracuda09, 2025-05-24:
Did you have a look at this, how to setup TvHeadend:
https://github.com/Barracuda09/SATPI/wiki/Using-SatPI-as-frontend-for-Tvheadend

Pay special attention to **Maximum PIDs** make this between 15 - 20 max

Barracuda09, 2025-05-24:
And pay attention to this settings as well **Full mux RX mode supported**

lgm75, 2025-07-22:
Same problem here after a power outage or electrical shutdown...

LITUATUI, 2025-07-25:
I had the same problem with SatPI and MiniSatIP, found out the solution recently.

Check [here](https://github.com/catalinii/minisatip/discussions/1249).

Jimbo888888, 2025-07-26:
I solved the problem differently. I set the receiver to restart every morning at 4:00 a.m. and then go directly into standby mode. Since then, I've had no problems with inactive tuners. I have OpenATV (latest version) installed.

Barracuda09, 2025-07-27:
@lgm75 
That seems not a problem of SatPI, but from the image an init order

Barracuda09, 2025-07-28:
@LITUATUI Thanks for you sponsoring.

You could try to restart SatPI from the web GUI.

What is your setup?
- Much activity like EPG scan or mux scan

I know the drivers from Vu are not as stable as it should be. And not updated that often.

Barracuda09, 2025-07-28:
This log does not provide that much useful information.

LITUATUI, 2025-07-29:
> [@LITUATUI](https://github.com/LITUATUI) I understand, but it is a way to try to bring the FBC tuners back to live, by rewriting the settings.
> 
> This rewrite, is done by toggling this option.

I see.

But am I using this setting correctly? "FBC Linked" enabled from FE 2 to FE 8?

LITUATUI, 2025-07-29:
Turning the "FBC Linked" on and off didn't bring the tuners back.

This problem also happens with MiniSatIP, it only works correctly when "mode 0" is used.

@Barracuda09 just out of curiosity, do you have any Enigma2 receiver with FBC tuner for testing?

Barracuda09, 2025-07-29:
Yes I have, for dvb-c and dvb-s2x

LITUATUI, 2025-07-30:
Correct @Barracuda09 

My DVB-C FBC tuner is almost always being used by family and friends, so it's very easy to detect problems.

The default modes on SatPI and MiniSatIP only work for a few hours at best. However "mode 0" (dvrX device) on MiniSatIP is super stable, but doesn't allow proper scanning. Anyway, it's good enough for me now.

Barracuda09, 2025-07-30:
I use a very old Vu+ stock image for my DVB-C which works for me.
```
root@vuuno4k:/media/Flash-Disk/satpi# cat /proc/version 
Linux version 3.14.28-1.12 (vuplus@vuplus) (gcc version 4.9.2 (GCC) ) #1 SMP Tue Feb 23 02:19:10 KST 2021
root@vuuno4k:/media/Flash-Disk/satpi# 
```
I use demuxX device to read and setup pids.

My impression is that this is all some limitation of the FBC drivers or hardware. Especially when you push it to the limits.

Barracuda09, 2025-08-02:
@LITUATUI 

There is only one mode that uses demuxX device to read and setup pids.

I have it here running, without any problems.

Barracuda09, 2025-09-05:
@Jimbo888888 and @LITUATUI 

Here is an test version [1.6.2.205_Enigma.zip](https://github.com/user-attachments/files/22174308/1.6.2.205_Enigma.zip)

Could one of you try this?

Barracuda09, 2025-09-13:
Fixed with 614f39ab48b7e600e4a8e95f2023a09fbb157adc

If anyone has other issues, please make another issue

closed 2025-09-13 by Barracuda09

Jimbo888888, 2025-09-25:
My Version is from OpenATV 7.5.1 with latent Updates.